coolsnowwolf/lede #11275: 在线编译X86版本报错

sclsyin, 2023-06-05:
日志如下，请各位大佬帮忙看下，谢谢！

```
make[3] -C toolchain/gcc/final compile
 make[2] target/compile
 make[3] -C target/linux compile
    ERROR: target/linux failed to build.
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 1
WARNING: Makefile 'package/feeds/other/luci-app-mwan3helper/Makefile' has a dependency on 'pdnsd-alt', which does not exist
 make[1] world
 make[2] target/compile
 make[3] -C target/linux compile
    ERROR: target/linux failed to build.
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 1
WARNING: Makefile 'package/feeds/other/luci-app-mwan3helper/Makefile' has a dependency on 'pdnsd-alt', which does not exist
make[1]: Entering directory '/workdir/openwrt'
make[2]: Entering directory '/workdir/openwrt'
make[3]: Entering directory '/workdir/openwrt/target/linux'
make[4]: Entering directory '/workdir/openwrt/target/linux/x86'
rm -f /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.238/localversion
/workdir/openwrt/scripts/kconfig.pl  + + /workdir/openwrt/target/linux/generic/config-5.4 /workdir/openwrt/target/linux/x86/config-5.4 /workdir/openwrt/target/linux/x86/64/config-5.4 > /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.238/.config.target
awk '/^(#[[:space:]]+)?CONFIG_KERNEL/{sub("CONFIG_KERNEL_","CONFIG_");print}' /workdir/openwrt/.config >> /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.238/.config.target
echo "# CONFIG_KALLSYMS_EXTRA_PASS is not set" >> /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.238/.config.target
echo "# CONFIG_KALLSYMS_ALL is not set" >> /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.238/.config.target
echo "CONFIG_KALLSYMS_UNCOMPRESSED=y" >> /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.238/.config.target
/workdir/openwrt/scripts/package-metadata.pl kconfig /workdir/openwrt/tmp/.packageinfo /workdir/openwrt/.config 5.4 > /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.238/.config.override
/workdir/openwrt/scripts/kconfig.pl 'm+' '+' /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.238/.config.target /dev/null /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.238/.config.override > /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.238/.config.set
mv /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.238/.config.set /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.238/.config.old
grep -v INITRAMFS /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.238/.config.old > /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.238/.config.set
echo 'CONFIG_INITRAMFS_SOURCE=""' >> /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.238/.config.set
echo '# CONFIG_INITRAMFS_FORCE is not set' >> /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.238/.config.set
rm -rf /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/modules
cmp -s /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.238/.config.set /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.238/.config.prev || { cp /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.238/.config.set /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.238/.config; cp /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.238/.config.set /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.238/.config.prev; }
export MAKEFLAGS= ; [ -d /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.238/user_headers ] || make -C /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.238 KCFLAGS="-fmacro-prefix-map=/workdir/openwrt/build_dir/target-x86_64_musl=target-x86_64_musl -fno-caller-saves " HOSTCFLAGS="-O2 -I/workdir/openwrt/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="x86_64-openwrt-linux-musl-" ARCH="x86" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Mon May  8 09:32:46 2023" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/workdir/openwrt/staging_dir/host/lib" KBUILD_HOSTLDLIBS="-L/workdir/openwrt/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls=  KERNELRELEASE=5.4.238 CC="x86_64-openwrt-linux-musl-gcc" INSTALL_HDR_PATH=/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.238/user_headers headers_install
grep '=[ym]' /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.238/.config.set | LC_ALL=C sort | mkhash md5 > /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.238/.vermagic
touch /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.238/.configured
rm -f /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.238/vmlinux /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.238/System.map
make -C /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.238 KCFLAGS="-fmacro-prefix-map=/workdir/openwrt/build_dir/target-x86_64_musl=target-x86_64_musl -fno-caller-saves " HOSTCFLAGS="-O2 -I/workdir/openwrt/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="x86_64-openwrt-linux-musl-" ARCH="x86" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Mon May  8 09:32:46 2023" KBUILD_BUILD_VERSION="0" HOST_LOADLIBES="-L/workdir/openwrt/staging_dir/host/lib" KBUILD_HOSTLDLIBS="-L/workdir/openwrt/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls=  KERNELRELEASE=5.4.238 CC="x86_64-openwrt-linux-musl-gcc" modules
make[5]: Entering directory '/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.238'
  CALL    scripts/checksyscalls.sh
  CALL    scripts/atomic/check-atomics.sh
  DESCEND  objtool
  CC [M]  net/netfilter/nf_conntrack_proto_tcp.o
In file included from ./include/linux/linkage.h:7,
                 from ./include/linux/kernel.h:8,
                 from ./include/linux/list.h:9,
                 from ./include/linux/timer.h:5,
                 from net/netfilter/nf_conntrack_proto_tcp.c:9:
net/netfilter/nf_conntrack_proto_tcp.c:35:19: error: 'nf_ct_tcp_no_window_check' undeclared here (not in a function); did you mean 'nf_ct_tcp_seqadj_set'?
 EXPORT_SYMBOL_GPL(nf_ct_tcp_no_window_check);
                   ^~~~~~~~~~~~~~~~~~~~~~~~~
./include/linux/export.h:[108](https://github.com/sclsyin/OpenWrt-Lienol-X86-X64/actions/runs/5172775521/jobs/9317435843#step:11:109):16: note: in definition of macro '___export_symbol_common'
  extern typeof(sym) sym;      \
                ^~~
./include/linux/export.h:177:34: note: in expansion of macro '___EXPORT_SYMBOL'
 #define __EXPORT_SYMBOL(sym,sec) ___EXPORT_SYMBOL(sym,sec)
                                  ^~~~~~~~~~~~~~~~
./include/linux/export.h:188:33: note: in expansion of macro '__EXPORT_SYMBOL'
 #define EXPORT_SYMBOL_GPL(sym)  __EXPORT_SYMBOL(sym, "_gpl")
                                 ^~~~~~~~~~~~~~~
net/netfilter/nf_conntrack_proto_tcp.c:35:1: note: in expansion of macro 'EXPORT_SYMBOL_GPL'
 EXPORT_SYMBOL_GPL(nf_ct_tcp_no_window_check);
 ^~~~~~~~~~~~~~~~~
make[7]: *** [scripts/Makefile.build:262: net/netfilter/nf_conntrack_proto_tcp.o] Error 1
make[6]: *** [scripts/Makefile.build:497: net/netfilter] Error 2
make[5]: Leaving directory '/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.238'
make[5]: *** [Makefile:1746: net] Error 2
make[4]: Leaving directory '/workdir/openwrt/target/linux/x86'
make[4]: *** [Makefile:30: /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.238/.modules] Error 2
make[3]: Leaving directory '/workdir/openwrt/target/linux'
make[3]: *** [Makefile:11: compile] Error 2
time: target/linux/compile#6.00#2.08#7.88
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:25: target/linux/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [target/Makefile:18: /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:230: world] Error 2
Error: Process completed with exit code 2.
```

WYC-2020, 2023-06-05:
这玩意不应该吧，自己修改过？613那个patch 还在么？

ChienHoWee, 2023-06-08:
我本地编译树莓派4B也出现了这个问题。
 CC [M]  net/netfilter/nf_conntrack_proto.o
  CC [M]  net/netfilter/nf_conntrack_proto_generic.o
  CC [M]  net/netfilter/nf_conntrack_proto_tcp.o
In file included from ./include/linux/linkage.h:7,
                 from ./include/linux/kernel.h:8,
                 from ./include/linux/list.h:9,
                 from ./include/linux/timer.h:5,
                 from net/netfilter/nf_conntrack_proto_tcp.c:9:
net/netfilter/nf_conntrack_proto_tcp.c:35:19: error: 'nf_ct_tcp_no_window_check' undeclared here (not in a function); did you mean 'nf_ct_tcp_seqadj_set'?
 EXPORT_SYMBOL_GPL(nf_ct_tcp_no_window_check);
                   ^~~~~~~~~~~~~~~~~~~~~~~~~
./include/linux/export.h:108:16: note: in definition of macro '___export_symbol_common'
  extern typeof(sym) sym;      \
                ^~~
./include/linux/export.h:177:34: note: in expansion of macro '___EXPORT_SYMBOL'
 #define __EXPORT_SYMBOL(sym,sec) ___EXPORT_SYMBOL(sym,sec)
                                  ^~~~~~~~~~~~~~~~
./include/linux/export.h:188:33: note: in expansion of macro '__EXPORT_SYMBOL'
 #define EXPORT_SYMBOL_GPL(sym)  __EXPORT_SYMBOL(sym, "_gpl")
                                 ^~~~~~~~~~~~~~~
net/netfilter/nf_conntrack_proto_tcp.c:35:1: note: in expansion of macro 'EXPORT_SYMBOL_GPL'
 EXPORT_SYMBOL_GPL(nf_ct_tcp_no_window_check);
 ^~~~~~~~~~~~~~~~~
make[7]: *** [scripts/Makefile.build:262: net/netfilter/nf_conntrack_proto_tcp.o] Error 1
make[6]: *** [scripts/Makefile.build:497: net/netfilter] Error 2
make[5]: *** [Makefile:1749: net] Error 2

sclsyin, 2023-06-08:
613那个patch 是哪啥子？

yuos-bit, 2023-08-06:
https://github.com/yuos-bit/AutoBuild-OpenWrt/actions/runs/5775049327/job/15652629217
问题一样，，请问楼主咋解决的呢？所有的源码来自op官方，补丁来自lede大佬的

Ovear, 2023-08-10:
官方把613补丁给干掉了，手动revert下就可以了。

5.4内核可以用

```
# git revert cbe73ea33d027dbb4b2cf1eca947ae746119e7d2
```

相关commit: https://github.com/openwrt/openwrt/commit/cbe73ea33d027dbb4b2cf1eca947ae746119e7d2

yuos-bit, 2023-08-11:
感谢over大佬，我已同步至我的云编译中，供后来人参考: https://github.com/yuos-bit/AutoBuild-OpenWrt/commit/2c14643f3ce364c5b0d04687276893f5685b8861

sclsyin, 2023-09-02:
感谢，已成功编译。

closed 2026-09-05 by coolsnowwolf