coolsnowwolf/lede #11277: luci-app-dockerman启动报错

pretendtobefree, 2023-06-05:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ X] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：docker能工作，抓取镜像、建立容器、启动等都正常，但系统日志中有一组错误信息，不知道如何解决，望赐教，谢谢！

### (2) 路由器型号和固件版本

A：版本为Powered by LuCI Master (git-23.141.16773-28dd4b3) / OpenWrt R23.5.1

### (3) 详细日志

A：
Mon Jun  5 15:16:35 2023 daemon.err dockerd[13146]: time="2023-06-05T07:16:35Z" level=warning msg="containerd config version `1` has been deprecated and will be removed in containerd v2.0, please switch to version `2`, see https://github.com/containerd/containerd/blob/main/docs/PLUGINS.md#version-header"
Mon Jun  5 15:16:35 2023 daemon.err modprobe: failed to find a module named aufs
Mon Jun  5 15:16:35 2023 daemon.err dockerd[13146]: time="2023-06-05T07:16:35.855136071Z" level=warning msg="failed to load plugin io.containerd.snapshotter.v1.devmapper" error="devmapper not configured"
Mon Jun  5 15:16:35 2023 daemon.err dockerd[13146]: time="2023-06-05T07:16:35.855542157Z" level=warning msg="could not use snapshotter devmapper in metadata plugin" error="devmapper not configured"
Mon Jun  5 15:16:35 2023 daemon.err dockerd[13146]: time="2023-06-05T07:16:35.856650786Z" level=error msg="failed to initialize a tracing processor \"otlp\"" error="no OpenTelemetry endpoint: skip plugin"
Mon Jun  5 15:16:35 2023 daemon.err dockerd[13146]: time="2023-06-05T07:16:35.885534922Z" level=warning msg="Unable to find memory controller"
Mon Jun  5 15:16:35 2023 daemon.err dockerd[13146]: time="2023-06-05T07:16:35.885568965Z" level=warning msg="Unable to find cpu controller"
Mon Jun  5 15:16:35 2023 daemon.err dockerd[13146]: time="2023-06-05T07:16:35.885572068Z" level=warning msg="Unable to find io controller"
Mon Jun  5 15:16:35 2023 daemon.err dockerd[13146]: time="2023-06-05T07:16:35.885574277Z" level=warning msg="Unable to find cpuset controller"
Mon Jun  5 15:16:35 2023 daemon.err dockerd[13146]: time="2023-06-05T07:16:35.885576588Z" level=warning msg="Unable to find pids controller"
Mon Jun  5 15:16:35 2023 daemon.err dockerd[13146]: time="2023-06-05T07:16:35.892958220Z" level=warning msg="Failed to find ip6tables: exec: \"ip6tables\": executable file not found in $PATH"

kxdn3, 2023-06-05:
6.1里面偶尔会起不来

pretendtobefree, 2023-06-05:
> 缺少aufs 模块而已，这个需要给内核打补丁 加入这个模块，另外一个是你没有编译ipv6而已，选择加上就可以了

怎么选择aufs模块？新手确实不懂，谢谢指教！

WYC-2020, 2023-06-06:
https://github.com/sfjro/aufs-standalone/tree/aufs5.x-rcN
自己去集成，现在没时间搞。linux  从内核5还是4开始移除了aufs从内核中，说是不稳定和没人维护，但是我看这个是有人在维护，需要就自己移植，但是稳定性需要自己测试，因为现在没有aufs默认是使用的overlay2,内核官方也是建议用这个

> > 缺少aufs 模块而已，这个需要给内核打补丁 加入这个模块，另外一个是你没有编译ipv6而已，选择加上就可以了
> 
> 怎么选择aufs模块？新手确实不懂，谢谢指教！

WYC-2020, 2023-06-06:
升级到最新就没有了，因为最新的docker移除了aufs这个文件系统 
[image]
[image]

coolsnowwolf, 2023-06-07:
这个用不上了，可以忽略

pretendtobefree, 2023-06-08:
> 升级到最新就没有了，因为最新的docker移除了aufs这个文件系统 [image] [image]

升级好麻烦啊，修改了docker、dockerd、containerd、runc的Makefile文件后相应的ipk倒是生成了，但与一堆插件有冲突，编译干脆通不过了，折腾2天了。

WYC-2020, 2023-06-08:
> > 升级到最新就没有了，因为最新的docker移除了aufs这个文件系统 [image] [image]
> 
> 升级好麻烦啊，修改了docker、dockerd、containerd、runc的Makefile文件后相应的ipk倒是生成了，但与一堆插件有冲突，编译干脆通不过了，折腾2天了。

还好吧也就改改版本和hash我都升完了都用起了

pretendtobefree, 2023-06-09:
> > > 升级到最新就没有了，因为最新的docker移除了aufs这个文件系统 [image] [image]
> > 
> > 
> > 升级好麻烦啊，修改了docker、dockerd、containerd、runc的Makefile文件后相应的ipk倒是生成了，但与一堆插件有冲突，编译干脆通不过了，折腾2天了。
> 
> 还好吧也就改改版本和hash我都升完了都用起了

我修改了相关的版本，也修改了 001-libdevmapper_name_fix.patch 文件，但卡在Verify dependencies are the vendored version这个环节，错误信息如下：
Verify dependencies are the vendored version
( DEP_VER=$( grep --only-matching --perl-regexp '(?<=PKG_VERSION:=)(.*)' "../containerd/Makefile" ); VEN_VER=$( grep --only-matching --perl-regexp '(?<=_VERSION:=v)(.*)(?=})' "/mnt/data/lede/build_dir/target-x86_64_musl/dockerd-24.0.2/hack/dockerfile/install/containerd.installer" ); if [ "${VEN_VER}" != "${DEP_VER}" ]; then echo "ERROR: Expected 'PKG_VERSION:=${VEN_VER}' in '../containerd/Makefile', found 'PKG_VERSION:=${DEP_VER}'"; exit 1; fi )
( DEP_VER=$( grep --only-matching --perl-regexp '(?<=PKG_SOURCE_VERSION:=)(.*)' "../libnetwork/Makefile" ); VEN_VER=$( grep --only-matching --perl-regexp '(?<=_COMMIT:=)(.*)(?=})' "/mnt/data/lede/build_dir/target-x86_64_musl/dockerd-24.0.2/hack/dockerfile/install/proxy.installer" ); if [ "${VEN_VER}" != "${DEP_VER}" ]; then echo "ERROR: Expected 'PKG_SOURCE_VERSION:=${VEN_VER}' in '../libnetwork/Makefile', found 'PKG_SOURCE_VERSION:=${DEP_VER}'"; exit 1; fi )
grep: /mnt/data/lede/build_dir/target-x86_64_musl/dockerd-24.0.2/hack/dockerfile/install/proxy.installer: No such file or directory
ERROR: Expected 'PKG_SOURCE_VERSION:=' in '../libnetwork/Makefile', found 'PKG_SOURCE_VERSION:=f6ccccb1c082a432c2a5814aaedaca56af33d9ea'
make[3]: *** [Makefile:182: /mnt/data/lede/build_dir/target-x86_64_musl/dockerd-24.0.2/.prepared_08ff2e21201760a4bdfbb7bfd13de2d8_6664517399ebbbc92a37c5bb081b5c53] Error 1

我强制把这个检查略过，编译是通过了，但dockerman无法正常工作，系统日志中相关提示如下：
Sat Jun 10 06:24:26 2023 daemon.err dockerd[8992]: time="2023-06-09T22:24:26.235126200Z" level=warning msg="failed to load plugin io.containerd.snapshotter.v1.devmapper" error="devmapper not configured"
Sat Jun 10 06:24:26 2023 daemon.err dockerd[8992]: time="2023-06-09T22:24:26.235251600Z" level=warning msg="could not use snapshotter devmapper in metadata plugin" error="devmapper not configured"
Sat Jun 10 06:24:27 2023 daemon.err dockerd[8992]: time="2023-06-09T22:24:27.294478500Z" level=error msg="failed to mount overlay: invalid argument" storage-driver=overlay2
Sat Jun 10 06:24:27 2023 daemon.err dockerd[8992]: time="2023-06-09T22:24:27.294627000Z" level=error msg="exec: \"fuse-overlayfs\": executable file not found in $PATH" storage-driver=fuse-overlayfs

打开dockerman的“概览”，提示如下：
/usr/lib/lua/luci/dispatcher.lua:477: Failed to execute cbi dispatcher target for entry '/admin/docker/overview'.
The called action terminated with an exception:
/usr/lib/lua/luci/model/cbi/dockerman/overview.lua:127: bad argument #1 to 'ipairs' (table expected, got nil)
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:477: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:129: in function </usr/lib/lua/luci/dispatcher.lua:128>

这就是所谓“会者不难，难者不会”吧。

WYC-2020, 2023-06-12:
> 我修改了相关的版本，也修改了 001-libdevmapper_name_fix.patch 文件，但卡在Verify dependencies are the vendored version这个环节，错误信息如下： Verify dependencies are the vendored version ( DEP_VER=$( grep --only-matching [...]
[docker.zip](https://github.com/coolsnowwolf/lede/files/11717462/docker.zip)
拿去替换

WYC-2020, 2023-06-12:
> 我修改了相关的版本，也修改了 001-libdevmapper_name_fix.patch 文件，但卡在Verify dependencies are the vendored version这个环节，错误信息如下： Verify dependencies are the vendored version ( DEP_VER=$( grep --only-matching [...]

dockerman 用我仓库那个 我有修复这个luci 问题，dockerman作者那边我也说过怎么修，如果不想用我这边的 你去那边issue看下 我又说修复方法

closed 2023-06-12 by pretendtobefree