coolsnowwolf/lede #11282: Lede OpenWrt version.

butinhi, 2023-06-07:
Why I built Lede OpenWrt is SNAPSHOT? I did not see it before

[image]

WYC-2020, 2023-06-08:
Compiled by yourself? Check to see if default-settings is installed. This package will modify the version information. It is selected by default. I don’t know how you did it.
[image]

butinhi, 2023-06-08:
> Compiled by yourself? Check to see if default-settings is installed. This package will modify the version information. It is selected by default. I don’t know how you did it. [image]

Yes, I complied by myself. I changed default language to auto from zzz-default-settings and luci-base.

butinhi, 2023-06-08:
> Compiled by yourself? Check to see if default-settings is installed. This package will modify the version information. It is selected by default. I don’t know how you did it. [image]

I see it. I unchecked it from menu config. But if checked it some settings will be default when upgrade new fw.

closed 2023-06-10 by butinhi