coolsnowwolf/lede #11286: passwall2 运行大概一天左右 国外连接失败 重新应用一次 又恢复正常 报错查看于iptable有关

ilafz, 2023-06-08:
passwall2 运行大概一天左右 国外连接失败 重新应用一次 又恢复正常 报错查看于iptable有关]

安装了nft后重启passwall2报错 但是软件能运行:

Warning: iptables-legacy tables present, use iptables-legacy to see them
iptables v1.8.8 (nf_tables): Could not fetch rule set generation id: Invalid argument

卸载nft 重启passwall2报错 但是软件仍能运行：
Error relocating /usr/sbin/ipset: ipset_xlate_argv: symbol not found
Error relocating /usr/sbin/ipset: ipset_xlate_argv: symbol not found
Error relocating /usr/sbin/ipset: ipset_xlate_argv: symbol not found
Error relocating /usr/sbin/ipset: ipset_xlate_argv: symbol not found
Error relocating /usr/sbin/ipset: ipset_xlate_argv: symbol not found
Error relocating /usr/sbin/ipset: ipset_xlate_argv: symbol not found
iptables v1.8.8 (legacy): Set laniplist doesn't exist.

Try `iptables -h' or 'iptables --help' for more information.
iptables v1.8.8 (legacy): Set vpsiplist doesn't exist.

Try `iptables -h' or 'iptables --help' for more information.
iptables v1.8.8 (legacy): Set laniplist doesn't exist.

Try `iptables -h' or 'iptables --help' for more information.
iptables v1.8.8 (legacy): Set vpsiplist doesn't exist.

Try `iptables -h' or 'iptables --help' for more information.
iptables v1.8.8 (legacy): Set laniplist doesn't exist.

Try `iptables -h' or 'iptables --help' for more information.
iptables v1.8.8 (legacy): Set vpsiplist doesn't exist.

Try `iptables -h' or 'iptables --help' for more information.
iptables v1.8.8 (legacy): Set laniplist doesn't exist.

Try `iptables -h' or 'iptables --help' for more information.
iptables v1.8.8 (legacy): Set vpsiplist doesn't exist.

Try iptables -h' or 'iptables --help' for more information. iptables v1.8.8 (legacy): invalid port/service disable' specified
Try iptables -h' or 'iptables --help' for more information. iptables v1.8.8 (legacy): invalid port/service disable' specified
Try `iptables -h' or 'iptables --help' for more information.

是内核的问题吗

固件版本	OpenWrt R21.6.22
内核版本	5.10.46

iptable -L如下
：
Chain PREROUTING (policy ACCEPT 4672 packets, 848K bytes)
pkts bytes target prot opt in out source destination
5715 915K PSW2_REDIRECT all -- * * 0.0.0.0/0 0.0.0.0/0
20733 3261K prerouting_rule all -- * * 0.0.0.0/0 0.0.0.0/0 /* !fw3: Custom prerouting rule chain /
176 12002 PSW2 icmp -- * * 0.0.0.0/0 0.0.0.0/0
20494 3251K zone_lan_prerouting all -- br-lan * 0.0.0.0/0 0.0.0.0/0 / !fw3 /
0 0 zone_docker_prerouting all -- docker0 * 0.0.0.0/0 0.0.0.0/0 / !fw3 /
959 62727 REDIRECT udp -- * * 0.0.0.0/0 0.0.0.0/0 / DNSMASQ */ udp dpt:53 redir ports 53

Chain INPUT (policy ACCEPT 4312 packets, 786K bytes)
pkts bytes target prot opt in out source destination

Chain OUTPUT (policy ACCEPT 6564 packets, 432K bytes)
pkts bytes target prot opt in out source destination
1432 120K PSW2_OUTPUT icmp -- * * 0.0.0.0/0 0.0.0.0/0

Chain POSTROUTING (policy ACCEPT 9212 packets, 672K bytes)
pkts bytes target prot opt in out source destination
33771 2544K postrouting_rule all -- * * 0.0.0.0/0 0.0.0.0/0 /* !fw3: Custom postrouting rule chain /
9768 826K zone_lan_postrouting all -- * br-lan 0.0.0.0/0 0.0.0.0/0 / !fw3 /
2 120 zone_docker_postrouting all -- * docker0 0.0.0.0/0 0.0.0.0/0 / !fw3 */

Chain PSW2 (1 references)
pkts bytes target prot opt in out source destination
94 7896 RETURN all -- * * 0.0.0.0/0 192.168.99.9 /* 'WAN_IP_RETURN' /
0 0 REDIRECT icmp -- * * 0.0.0.0/0 198.18.0.0/16 / '默认' /
82 4106 REDIRECT icmp -- * * 0.0.0.0/0 0.0.0.0/0 / '默认' /
0 0 RETURN tcp -- * * 0.0.0.0/0 0.0.0.0/0 / '默认' */

Chain PSW2_OUTPUT (1 references)
pkts bytes target prot opt in out source destination
0 0 RETURN all -- * * 0.0.0.0/0 0.0.0.0/0 mark match 0xff
0 0 REDIRECT icmp -- * * 0.0.0.0/0 198.18.0.0/16
1432 120K REDIRECT icmp -- * * 0.0.0.0/0 0.0.0.0/0

Chain PSW2_REDIRECT (1 references)
pkts bytes target prot opt in out source destination

Chain postrouting_docker_rule (1 references)
pkts bytes target prot opt in out source destination

Chain postrouting_lan_rule (1 references)
pkts bytes target prot opt in out source destination

Chain postrouting_rule (1 references)
pkts bytes target prot opt in out source destination

Chain postrouting_wan_rule (1 references)
pkts bytes target prot opt in out source destination

Chain prerouting_docker_rule (1 references)
pkts bytes target prot opt in out source destination

Chain prerouting_lan_rule (1 references)
pkts bytes target prot opt in out source destination

Chain prerouting_rule (1 references)
pkts bytes target prot opt in out source destination

Chain prerouting_wan_rule (1 references)
pkts bytes target prot opt in out source destination

Chain zone_docker_postrouting (1 references)
pkts bytes target prot opt in out source destination
2 120 postrouting_docker_rule all -- * * 0.0.0.0/0 0.0.0.0/0 /* !fw3: Custom docker postrouting rule chain */

Chain zone_docker_prerouting (1 references)
pkts bytes target prot opt in out source destination
0 0 prerouting_docker_rule all -- * * 0.0.0.0/0 0.0.0.0/0 /* !fw3: Custom docker prerouting rule chain */

Chain zone_lan_postrouting (1 references)
pkts bytes target prot opt in out source destination
9768 826K postrouting_lan_rule all -- * * 0.0.0.0/0 0.0.0.0/0 /* !fw3: Custom lan postrouting rule chain */

Chain zone_lan_prerouting (1 references)
pkts bytes target prot opt in out source destination
20494 3251K prerouting_lan_rule all -- * * 0.0.0.0/0 0.0.0.0/0 /* !fw3: Custom lan prerouting rule chain */

Chain zone_wan_postrouting (0 references)
pkts bytes target prot opt in out source destination
0 0 postrouting_wan_rule all -- * * 0.0.0.0/0 0.0.0.0/0 /* !fw3: Custom wan postrouting rule chain */

Chain zone_wan_prerouting (0 references)
pkts bytes target prot opt in out source destination
0 0 prerouting_wan_rule all -- * * 0.0.0.0/0 0.0.0.0/0 /* !fw3: Custom wan prerouting rule chain */

ilafz, 2023-06-08:
是否可以使用这个https://downloads.openwrt.org/snapshots/targets/x86/64/packages/kernel_5.15.114-1-13310873601d73e3bc11e04652a8fb02_x86_64.ipk
直接更新内核呢

ilafz, 2023-06-08:
然后升级 kmod-ipt-ipset

WYC-2020, 2023-06-08:
别动不动就怀疑内核，你这玩意明显就和内核没啥关系，你用老版本的lede，dnsmasq版本都很旧，都不是最新，结果pw插件你用最新的，根本就适配不好，pw这玩意对masq版本有要求的，可以去pw那边看相关的issue

hcym, 2023-06-08:
dnsmasq从master复制过来最新最好用，版本是289，2203和大雕的都能用，2102不能用

ilafz, 2023-06-09:
我升级到了dnsmasq-full 2.89 passwall无法启动 显示缺少ipset dnsmasq-v 查看发现no ipset换回2.86版本正常了