coolsnowwolf/lede #11288: 最近编译的版本运行一段时间后就会出现国内网络无法访问的情况

tone88, 2023-06-09:
X86 ESIX下跑的OP，最近编译了好几个版本都遇到运行1天多就出现国内网站无法访问或者访问特慢的情况，这时间访问国外网站是正常的，SSRP里选的“使用WAN下发的DNS”，出现这问题的时候点一下SSRP设置的保存&应用按钮就会恢复，OP固件是自己编译的插件列表见下图，网络环境也比较单纯，没折腾其它DNS也没开去广告之类的东西，感觉应该是SSRP 的DNS问题，有其它同学遇过到此类问题吗，不知道是不是个列。
插件列表：
[image]
出现问题时的系统日志：
出问题时除了阿里DDNS的这个外网IP读取错误外，没有其它错误信息。
[image]

WYC-2020, 2023-06-09:
日志都没贴到关键点，你这个是阿里云dns更新失败的日志，按你这个描述十之八九和lede仓库没关系，在遇到问题，在路由器看下外网是否通的，这个问题你应该去隔壁反馈

tone88, 2023-06-09:
> 

因为在出现这个问题的时候日志里没有其它错误信息，只有这个阿里DDNS读不到外网IP的错误，但是这个时候访问国外网站是没有问题的。盲猜会不会是SSRP的DNS出了问题。

WYC-2020, 2023-06-09:
> > 
> 
> 因为在出现这个问题的时候日志里没有其它错误信息，只有这个阿里DDNS读不到外网IP的错误，但是这个时候访问国外网站是没有问题的。盲猜会不会是SSRP的DNS出了问题，准确说应该是SSRP的国内DNS这块出了问题导致解析不到域名？

那你就别用上面下发的dns 直接用
[image]
直接用
这几个试下呗

tone88, 2023-06-09:
> 

挨个试到119.29.29.29那里了，估计悬

WYC-2020, 2023-06-09:
> > 
> 
> 挨个试到119.29.29.29那里了，估计悬

那应该说不定又是最近更新了 什么库导致的，以前有因为更新了那个china什么出问题的，如果都不行 你可以试试回滚几个commit试试

tone88, 2023-06-09:
> 

> > > 
> > 
> > 
> > 挨个试到119.29.29.29那里了，估计悬
> 
> 那应该说不定又是最近更新了 什么库导致的，以前有因为更新了那个china什么出问题的，如果都不行 你可以试试回滚几个commit试试

chinadns-ng，最近更新了这个，我回滚编译试试

lwb1978, 2023-06-09:
我昨天新编译的固件里新增加了SSR-P并用它翻墙，也出现了楼主的情况，突然全屋断网，还以为是电信的问题，后来就问题排查到了SSR-P，就是它引起的，以前一直用“哈喽我得”翻墙没出现过类似问题。

lwb1978, 2023-06-09:
> > 
> 
> > > > 
> > > 
> > > 
> > > 挨个试到119.29.29.29那里了，估计悬
> > 
> > 
> > 那应该说不定又是最近更新了 什么库导致的，以前有因为更新了那个china什么出问题的，如果都不行 你可以试试回滚几个commit试试
> 
> chinadns-ng，最近更新了这个，我回滚编译试试

楼主试试不编译这个chinadns-ng看看，但是据说ssr不选这个插件会报错。

tone88, 2023-06-09:
> 我昨天新编译的固件里新增加了SSR-P并用它翻墙，也出现了楼主的情况，突然全屋断网，还以为是电信的问题，后来就问题排查到了SSR-P，就是它引起的，以前一直用“哈喽我得”翻墙没出现过类似问题。

看来不是个例，我刚刚回滚了一个旧版本的chinadns-ng   正在编译。
SSRP好像默认就选中了这个chinadns-ng  如果取消的话可能得换成其它DNS，本人小白估计是折腾不了，就先用回滚到旧版本的办法试试吧

lwb1978, 2023-06-09:
还有就是我出问题的时候我网络接口的pppoe的dns是手动设置114和8.8.8.8，出问题后我设置成使用isp通告的dns，目前12小时过去暂时还没发现断网。

e703, 2023-06-10:
我的系统：
固件版本 | OpenWrt R23.6.6 / LuCI Master (git-23.141.16773-28dd4b3)
内核版本 | 6.1.32
passwall 版本 【 4.66-5 】 
Passwall DNS模式 ChinaDNS-NG
代理模式为 GFW列表
ChinaDNS-NG 版本【 2023.06.01 】
geoip  geosite 更新正常到最新

故障时海外代理正常，passwall上点击百度连接显示正常，但是客户端 国内DNS不通，无法访问国内站点。

e703, 2023-06-10:
故障时候手动把 “网络”→“防火墙”→“常规设置”→“网络启用FullCone-NAT”的状态修改一下，然后保存。就能恢复一段时间。不只是什么情况？

lwb1978, 2023-06-10:
> 我的系统： 固件版本 | OpenWrt R23.6.6 / LuCI Master (git-23.141.16773-28dd4b3) 内核版本 | 6.1.32 passwall 版本 【 4.66-5 】 Passwall DNS模式 ChinaDNS-NG 代理模式为 GFW列表 ChinaDNS-NG 版本【 2023.06.01 】 geoip geosite 更新正常到最新
> 
> 故障时海外代理正常，passwall上点击百度连接显示正常，但是客户端 国内DNS不通，无法访问国内站点。

这么说有可能就是chinadns的问题

tone88, 2023-06-10:
> > 我的系统： 固件版本 | OpenWrt R23.6.6 / LuCI Master (git-23.141.16773-28dd4b3) 内核版本 | 6.1.32 passwall 版本 【 4.66-5 】 Passwall DNS模式 ChinaDNS-NG 代理模式为 GFW列表 ChinaDNS-NG 版本【 2023.06.01 】 geoip geosite 更新正常到最新
> > 故障时海外代理正常，passwall上点击百度连接显示正常，但是客户端 国内DNS不通，无法访问国内站点。
> 
> 这么说有可能就是chinadns的问题

回滚chinadns-ng到4.20-1版本，目前运行近2天暂时没出问题，持续观察中。

mitsukileung, 2023-06-11:
> 回滚chinadns-ng到4.20-1版本，目前运行近2天暂时没出问题，持续观察中。

如果能确定是 chinadns-ng 的问题可以直接修改makefile
PKG_VERSION:=2023.04.20
PKG_HASH:=e1f5197ecd96ebab4440fe069e0a06935df7ea07ac1ad0bab3f88bffe0055294

mitsukileung, 2023-06-15:
> 回滚chinadns-ng到4.20-1版本，目前运行近2天暂时没出问题，持续观察中。

换回4.20还会出现同样的情况吗？

tone88, 2023-06-15:
> 换回4.20还会出现同样的情况吗？

目前没有再出现问题。
有遇到相同问题的可以回滚chinadns-ng到4.20-1版本

mitsukileung, 2023-06-15:
> 目前没有再出现问题。 有遇到相同问题的可以回滚chinadns-ng到4.20-1版本

你是直接回滚到4.20版本的PR，还是只回滚了chinadns-ng 到4.20版本？

tone88, 2023-06-15:
> 你是直接回滚到4.20版本的PR，还是只回滚了chinadns-ng 到4.20版本？

只回滚了chinadns-ng的版本，直接改makefile里的版本和hash值就行了。
我是从最新的版本一个版本一个版本往前试的，直到4.20-1才没再出过问题

mitsukileung, 2023-06-15:
> 只回滚了chinadns-ng的版本，直接改makefile里的版本和hash值就行了。 我是从最新的版本一个版本一个版本往前试的，直到4.20-1才没再出过问题

好的，谢谢！

closed 2023-06-28 by coolsnowwolf