coolsnowwolf/lede #11294: 路由管理页面打不开显示 Status: 500 Internal Server Error

smdx, 2023-06-10:
各位大佬好，最近时不时遇到登录路由一直转圈圈打不开，页面标题显示：
`Status: 500 Internal Server Error /usr/lib/lua/luci/template.lua:97: Failed to execute template 'sysauth'. A runtime error occured: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'header'. A runtime error occured: /usr/lib/lua/luci/template.lua:97: Failed to execute template 'themes/bootstrap/header'. A runtime error occured: [string "/usr/lib/lua/luci/view/themes/bootstrap/hea..."]:150: attempt to index local 'boardinfo' (a nil value) stack traceback: [C]: in function 'error' /usr/lib/lua/luci/template.lua:97: in function </usr/lib/lua/luci/template.lua:85> (tail call): ? /usr/lib/lua/luci/dispatcher.lua:400: in function 'dispatch' /usr/lib/lua/luci/dispatcher.lua:129: in function </usr/lib/lua/luci/dispatcher.lua:128>`

用的是默认主题出现问题时网络正常就是管理页面打不开，请问各位大佬这要怎么解决？谢谢
PVE 7.1-7 LXC运行OpenWRT

WYC-2020, 2023-06-10:
boardinfo这个为空了，代码逻辑应该是没处理这个为空的异常

smdx, 2023-06-10:
> boardinfo这个为空了，代码逻辑应该是没处理这个为空的异常

感谢回复，请问这个问题是需要等待源码修复吗？还是有什么解决方法？谢谢。

WYC-2020, 2023-06-12:
> > boardinfo这个为空了，代码逻辑应该是没处理这个为空的异常
> 
> 感谢回复，请问这个问题是需要等待源码修复吗？还是有什么解决方法？谢谢。

要么分析boardinfo为什么为空，要么就修改代码增加这个检测进行跳过，我的建议是修改代码，然后分析这个为啥为空，并且为空又没什么bug

closed 2026-09-05 by coolsnowwolf