coolsnowwolf/lede #11295: wdr 7500 v3  8MB flash 只能编译出测试固件，本地和云编译都一样

Winward-King, 2023-06-10:
## 1.关于你要提交的问题
wdr 7500 v3  8MB 只能编译出测试固件
本地和云编译都一样

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
wdr 7500 v3  8MB 只能编译出测试固件
全部取消 只选ipv6 helper，和TCC网络加速 只能编译出
openwrt-ath79-generic-tplink_tl-wdr7500-v3-initramfs-kernel.bin
A：

### (2) 路由器型号和固件版本
wdr 7500 v3  8MB  flash
A：

### (3) 详细日志
wdr7500-v3-squashfs-factory.bin
[mktplinkfw] rootfs offset aligned to 0x2244004
[mktplinkfw] *** error: images are too big by 990656 bytes
[mktplinkfw] rootfs offset aligned to 0x2244004
[mktplinkfw] *** error: images are too big by 990656 bytes
A：

WYC-2020, 2023-06-10:
不是给你说了，镜像太大了，这就是硬路由的硬bug，自己精简包大小

Winward-King, 2023-06-10:
Winward
***@***.***

Winward-King, 2023-06-10:
除了 luci app 路径 下还有什么能精简的，可以指导一下吗，目前用的openwrt 22.03.5  。千兆内网可以跑满，千兆宽带只能跑300Mbps，电脑拨号950Mbps

mitsukileung, 2023-06-11:
这个要自己慢慢试才行，flash 太小了

WYC-2020, 2023-06-12:
> 除了 luci app 路径 下还有什么能精简的，可以指导一下吗，目前用的openwrt 22.03.5 。千兆内网可以跑满，千兆宽带只能跑300Mbps，电脑拨号950Mbps

删除掉一些没必要的默认库，驱动的话自己知道那些用不上的 就可以精简

kokutoukiritsugu, 2023-06-27:
这老路由器op官网固件固件就可以了...
开个flow_offloading加速
uci set firewall.@defaults[0].flow_offloading='1'

closed 2026-09-05 by coolsnowwolf