coolsnowwolf/lede #11300: 竞斗云 2.0 运行 block mount 命令报错

LIKE2000-ART, 2023-06-12:
## 详细叙述

### (1) 具体问题

运行 block mount 命令报错：block: No "mount.ubi" utility available，df -h 看不到 /mnt/mtdblock0 分区

### (2) 路由器型号和固件版本

主机型号 | 竞斗云 2.0 （P&W R619AC 128M）
-- | --
架构 | ARMv7 Processor rev 5 (v7l) x 4 (896MHz, 45.0°C 44.0°C )

固件版本 | OpenWrt Full R23.6.6 / LuCI Master (git-23.141.16773-28dd4b3)
-- | --
内核版本 | 5.4.245

### (3) 详细日志

root@OpenWrt:/# block mount
block: No "mount.ubi" utility available
block: mounting /dev/mtdblock0 (ubi) as /mnt/mtdblock0 failed (2) - No such file or directory
root@OpenWrt:/# opkg install block-mount 
Package block-mount (2023-01-22-1ea5855e-1) installed in root is up to date.
root@OpenWrt:/# df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/root                62.0M     62.0M         0 100% /rom
tmpfs                   247.9M     24.8M    223.1M  10% /tmp
/dev/ubi0_2              48.3M    756.0K     45.0M   2% /overlay
overlayfs:/overlay       48.3M    756.0K     45.0M   2% /
tmpfs                   512.0K         0    512.0K   0% /dev
/dev/mmcblk0p2           57.4G     12.0K     54.4G   0% /mnt/mmcblk0p2

aiamadeus, 2023-06-12:
没看懂你要干啥，你是要操作 mtd 还是 mmc

LIKE2000-ART, 2023-06-12:
> 

mtd

LIKE2000-ART, 2023-06-12:
> 没看懂你要干啥，你是要操作 mtd 还是 mmc

我想操作mtd分区，mmc是我的存储卡

aiamadeus, 2023-06-12:
mtd 分区都在 /dev 下

LIKE2000-ART, 2023-06-12:
> block 不是用来操作 mtd 分区的

好的！谢谢您