coolsnowwolf/lede #11302: 6.1.30+以后遇到一个ipv6相关的内核core，不知道怎么解

littoy, 2023-06-12:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：内核异常

### (2) 路由器型号和固件版本

A：intel 7505 x86_64 + pve kvm host

### (3) 详细日志

A：
Sun Jun 11 23:02:41 2023 daemon.notice netifd: Network device 'eth5' link is down
Sun Jun 11 23:02:41 2023 kern.info kernel: [39210.778735] igc 0000:05:00.0 eth5: NIC Link is Down
Sun Jun 11 23:02:41 2023 kern.info kernel: [39210.780515] br-switch: port 6(eth5) entered disabled state
Sun Jun 11 23:03:01 2023 kern.info kernel: [39231.230004] igc 0000:05:00.0 eth5: NIC Link is Up 100 Mbps Full Duplex, Flow Control: RX
Sun Jun 11 23:03:01 2023 kern.info kernel: [39231.232791] br-switch: port 6(eth5) entered blocking state
Sun Jun 11 23:03:01 2023 kern.info kernel: [39231.234056] br-switch: port 6(eth5) entered forwarding state
Sun Jun 11 23:03:01 2023 daemon.notice netifd: Network device 'eth5' link is up
Sun Jun 11 23:03:15 2023 daemon.notice netifd: Network device 'eth5' link is down
Sun Jun 11 23:03:15 2023 kern.info kernel: [39245.340970] igc 0000:05:00.0 eth5: NIC Link is Down
Sun Jun 11 23:03:15 2023 kern.info kernel: [39245.342422] br-switch: port 6(eth5) entered disabled state
Sun Jun 11 23:03:18 2023 kern.info kernel: [39247.621179] igc 0000:05:00.0 eth5: NIC Link is Up 100 Mbps Full Duplex, Flow Control: RX
Sun Jun 11 23:03:18 2023 kern.info kernel: [39247.623059] br-switch: port 6(eth5) entered blocking state
Sun Jun 11 23:03:18 2023 kern.info kernel: [39247.623981] br-switch: port 6(eth5) entered forwarding state
Sun Jun 11 23:03:18 2023 daemon.notice netifd: Network device 'eth5' link is up
Sun Jun 11 23:03:49 2023 daemon.notice netifd: Network device 'eth5' link is down
Sun Jun 11 23:03:49 2023 kern.info kernel: [39278.783265] igc 0000:05:00.0 eth5: NIC Link is Down
Sun Jun 11 23:03:49 2023 kern.info kernel: [39278.784302] br-switch: port 6(eth5) entered disabled state
Sun Jun 11 23:03:52 2023 kern.info kernel: [39282.107399] igc 0000:05:00.0 eth5: NIC Link is Up 2500 Mbps Full Duplex, Flow Control: RX/TX
Sun Jun 11 23:03:52 2023 kern.info kernel: [39282.109761] br-switch: port 6(eth5) entered blocking state
Sun Jun 11 23:03:52 2023 kern.info kernel: [39282.111063] br-switch: port 6(eth5) entered forwarding state
Sun Jun 11 23:03:52 2023 daemon.notice netifd: Network device 'eth5' link is up
Sun Jun 11 23:16:47 2023 daemon.notice netifd: Network device 'eth5' link is down
Sun Jun 11 23:16:47 2023 kern.info kernel: [40056.879874] igc 0000:05:00.0 eth5: NIC Link is Down
Sun Jun 11 23:16:47 2023 kern.info kernel: [40056.881702] br-switch: port 6(eth5) entered disabled state
Sun Jun 11 23:16:49 2023 kern.info kernel: [40059.088058] igc 0000:05:00.0 eth5: NIC Link is Up 10 Mbps Half Duplex, Flow Control: None
Sun Jun 11 23:16:49 2023 kern.info kernel: [40059.090302] br-switch: port 6(eth5) entered blocking state
Sun Jun 11 23:16:49 2023 kern.info kernel: [40059.091758] br-switch: port 6(eth5) entered forwarding state
Sun Jun 11 23:16:49 2023 daemon.notice netifd: Network device 'eth5' link is up
Sun Jun 11 23:16:50 2023 kern.info kernel: [40059.511731] igc 0000:05:00.0 eth5: NIC Link is Down
Sun Jun 11 23:16:50 2023 daemon.notice netifd: Network device 'eth5' link is down
Sun Jun 11 23:16:50 2023 kern.info kernel: [40060.099637] br-switch: port 6(eth5) entered disabled state
Sun Jun 11 23:16:54 2023 daemon.notice netifd: Network device 'eth5' link is up
Sun Jun 11 23:16:54 2023 kern.info kernel: [40063.523816] igc 0000:05:00.0 eth5: NIC Link is Up 100 Mbps Full Duplex, Flow Control: RX
Sun Jun 11 23:16:54 2023 kern.info kernel: [40063.524879] br-switch: port 6(eth5) entered blocking state
Sun Jun 11 23:16:54 2023 kern.info kernel: [40063.525354] br-switch: port 6(eth5) entered forwarding state
Sun Jun 11 23:17:18 2023 daemon.notice netifd: Network device 'eth5' link is down
Sun Jun 11 23:17:18 2023 kern.info kernel: [40087.994321] igc 0000:05:00.0 eth5: NIC Link is Down
Sun Jun 11 23:17:18 2023 kern.info kernel: [40087.995991] br-switch: port 6(eth5) entered disabled state
Sun Jun 11 23:17:20 2023 kern.info kernel: [40090.230494] igc 0000:05:00.0 eth5: NIC Link is Up 100 Mbps Full Duplex, Flow Control: RX
Sun Jun 11 23:17:20 2023 kern.info kernel: [40090.232615] br-switch: port 6(eth5) entered blocking state
Sun Jun 11 23:17:20 2023 kern.info kernel: [40090.233612] br-switch: port 6(eth5) entered forwarding state
Sun Jun 11 23:17:20 2023 daemon.notice netifd: Network device 'eth5' link is up
Sun Jun 11 23:17:23 2023 daemon.notice netifd: Network device 'eth5' link is down
Sun Jun 11 23:17:23 2023 kern.info kernel: [40093.174017] igc 0000:05:00.0 eth5: NIC Link is Down
Sun Jun 11 23:17:23 2023 kern.info kernel: [40093.174883] br-switch: port 6(eth5) entered disabled state
Sun Jun 11 23:17:27 2023 daemon.notice netifd: Network device 'eth5' link is up
Sun Jun 11 23:17:27 2023 kern.info kernel: [40096.574134] igc 0000:05:00.0 eth5: NIC Link is Up 2500 Mbps Full Duplex, Flow Control: RX/TX
Sun Jun 11 23:17:27 2023 kern.info kernel: [40096.575504] br-switch: port 6(eth5) entered blocking state
Sun Jun 11 23:17:27 2023 kern.info kernel: [40096.576213] br-switch: port 6(eth5) entered forwarding state
Sun Jun 11 23:21:46 2023 kern.info kernel: [40355.969080] igc 0000:05:00.0 eth5: NIC Link is Down
[... 32 lines cut ...]
Mon Jun 12 08:06:15 2023 kern.warn kernel: [71823.520448]  ? __warn+0x72/0xc0
Mon Jun 12 08:06:15 2023 kern.warn kernel: [71823.520839]  ? inet_sock_destruct+0x194/0x1c0
Mon Jun 12 08:06:15 2023 kern.warn kernel: [71823.521371]  ? report_bug+0xf1/0x140
Mon Jun 12 08:06:15 2023 kern.warn kernel: [71823.521737]  ? handle_bug+0x46/0x80
Mon Jun 12 08:06:15 2023 kern.warn kernel: [71823.522087]  ? exc_invalid_op+0x1d/0x70
Mon Jun 12 08:06:15 2023 kern.warn kernel: [71823.522467]  ? asm_exc_invalid_op+0x1f/0x30
Mon Jun 12 08:06:15 2023 kern.warn kernel: [71823.522851]  ? inet_sock_destruct+0x194/0x1c0
Mon Jun 12 08:06:15 2023 kern.warn kernel: [71823.523277]  inet6_sock_destruct+0x1a/0x20
Mon Jun 12 08:06:15 2023 kern.warn kernel: [71823.523651]  __sk_destruct+0x27/0x180
Mon Jun 12 08:06:15 2023 kern.warn kernel: [71823.523995]  rcu_core+0x28f/0x6a0
Mon Jun 12 08:06:15 2023 kern.warn kernel: [71823.524367]  rcu_core_si+0xd/0x10
Mon Jun 12 08:06:15 2023 kern.warn kernel: [71823.524701]  __do_softirq+0xc1/0x1ec
Mon Jun 12 08:06:15 2023 kern.warn kernel: [71823.525155]  run_ksoftirqd+0x28/0x40
Mon Jun 12 08:06:15 2023 kern.warn kernel: [71823.525517]  smpboot_thread_fn+0xdf/0x1d0
Mon Jun 12 08:06:15 2023 kern.warn kernel: [71823.525862]  kthread+0xe2/0x110
Mon Jun 12 08:06:15 2023 kern.warn kernel: [71823.526222]  ? sort_range+0x30/0x30
Mon Jun 12 08:06:15 2023 kern.warn kernel: [71823.526538]  ? kthread_complete_and_exit+0x20/0x20
Mon Jun 12 08:06:15 2023 kern.warn kernel: [71823.526893]  ret_from_fork+0x1f/0x30
Mon Jun 12 08:06:15 2023 kern.warn kernel: [71823.527210]  </TASK>
Mon Jun 12 08:06:15 2023 kern.warn kernel: [71823.527448] ---[ end trace 0000000000000000 ]---
Mon Jun 12 08:11:06 2023 authpriv.info dropbear[5153]: Child connection from 192.168.125.32:50767
Mon Jun 12 08:11:06 2023 authpriv.notice dropbear[5153]: Pubkey auth succeeded for 'root' with ssh-rsa key SHA256:37RoS1zkDfuYSTf+3aQTaROzT0I6Y68UsJ0ccGWnfio from 192.168.125.32:50767
Mon Jun 12 08:12:19 2023 daemon.notice ttyd[10753]: [2023/06/12 08:12:19:8845] N: rops_handle_POLLIN_netlink: DELADDR
Mon Jun 12 08:12:19 2023 daemon.notice ttyd[10753]: [2023/06/12 08:12:19:8871] N: rops_handle_POLLIN_netlink: DELADDR
Mon Jun 12 08:16:51 2023 authpriv.info dropbear[5153]: Exit (root) from <192.168.125.32:50767>: Disconnect received

littoy, 2023-06-15:
反馈kernel bug了

closed 2023-06-15 by littoy

KevinMX, 2024-01-26:
6.6 内核下此问题似乎已经解决。
继续观测中。