coolsnowwolf/lede #11303: 运行出错：Waiting for boot device PARTUUID=xxxx....，不能继续运行

happymem2023, 2023-06-12:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：刚开始学着编译openwrt-x86-64-generic-squashfs-combined-efi，几经周折，编译成功（未提示编译错误），用StarWind V2V Converter转为VHD镜像后，在windows server 2008 R2的Hyper-v虚拟机上运行，出现错误：Waiting for boot device PARTUUID=xxxx....，请教大佬，编译时哪里选项设置错了？
[image]

### (2) 路由器型号和固件版本

A：OpenWrt-X86-64软路由

### (3) 详细日志

A：以下是.config文件内容（“#”前我多加了一个'字符，否则会被加大显示）
CONFIG_TARGET_x86=y
CONFIG_TARGET_x86_64=y
CONFIG_TARGET_x86_64_DEVICE_generic=y
CONFIG_CGROUPFS_MOUNT_KERNEL_CGROUPS=y
CONFIG_DOCKER_CGROUP_OPTIONS=y
CONFIG_DOCKER_NET_MACVLAN=y
CONFIG_DOCKER_STO_EXT4=y
CONFIG_KERNEL_CGROUP_DEVICE=y
CONFIG_KERNEL_CGROUP_FREEZER=y
CONFIG_KERNEL_CGROUP_NET_PRIO=y
CONFIG_KERNEL_EXT4_FS_POSIX_ACL=y
CONFIG_KERNEL_EXT4_FS_SECURITY=y
CONFIG_KERNEL_FS_POSIX_ACL=y
CONFIG_KERNEL_NET_CLS_CGROUP=y
'# CONFIG_PACKAGE_UnblockNeteaseMusic-Go is not set
'# CONFIG_PACKAGE_adbyby is not set
CONFIG_PACKAGE_bash=y
CONFIG_PACKAGE_btrfs-progs=y
CONFIG_PACKAGE_cgroupfs-mount=y
'# CONFIG_PACKAGE_chinadns-ng is not set
CONFIG_PACKAGE_containerd=y
'# CONFIG_PACKAGE_coreutils-base64 is not set
CONFIG_PACKAGE_coreutils-nohup=y
'# CONFIG_PACKAGE_dns2socks is not set
'# CONFIG_PACKAGE_dns2tcp is not set
CONFIG_PACKAGE_docker=y
CONFIG_PACKAGE_dockerd=y
'# CONFIG_PACKAGE_etherwake is not set
'# CONFIG_PACKAGE_iptables-mod-ipsec is not set
CONFIG_PACKAGE_kmod-asn1-encoder=y
CONFIG_PACKAGE_kmod-br-netfilter=y
'# CONFIG_PACKAGE_kmod-crypto-deflate is not set
'# CONFIG_PACKAGE_kmod-crypto-des is not set
'# CONFIG_PACKAGE_kmod-crypto-echainiv is not set
'# CONFIG_PACKAGE_kmod-crypto-kpp is not set
'# CONFIG_PACKAGE_kmod-crypto-md5 is not set
CONFIG_PACKAGE_kmod-crypto-rng=y
CONFIG_PACKAGE_kmod-crypto-sha256=y
CONFIG_PACKAGE_kmod-crypto-sha512=y
CONFIG_PACKAGE_kmod-dax=y
CONFIG_PACKAGE_kmod-dm=y
CONFIG_PACKAGE_kmod-dummy=y
CONFIG_PACKAGE_kmod-fs-btrfs=y
CONFIG_PACKAGE_kmod-ikconfig=y
CONFIG_PACKAGE_kmod-inet-diag=y
'# CONFIG_PACKAGE_kmod-ipsec is not set
'# CONFIG_PACKAGE_kmod-ipt-ipsec is not set
'# CONFIG_PACKAGE_kmod-iptunnel6 is not set
CONFIG_PACKAGE_kmod-keys-encrypted=y
CONFIG_PACKAGE_kmod-keys-trusted=y
CONFIG_PACKAGE_kmod-lib-raid6=y
CONFIG_PACKAGE_kmod-lib-xor=y
CONFIG_PACKAGE_kmod-lib-zstd=y
CONFIG_PACKAGE_kmod-nf-ipvs=y
CONFIG_PACKAGE_kmod-oid-registry=y
CONFIG_PACKAGE_kmod-random-core=y
'# CONFIG_PACKAGE_kmod-shortcut-fe is not set
'# CONFIG_PACKAGE_kmod-shortcut-fe-cm is not set
'# CONFIG_PACKAGE_kmod-tcp-bbr is not set
CONFIG_PACKAGE_kmod-tpm=y
CONFIG_PACKAGE_kmod-veth=y
'# CONFIG_PACKAGE_kmod-wireguard is not set
CONFIG_PACKAGE_libcap-bin=y
CONFIG_PACKAGE_libcap-bin-capsh-shell="/bin/sh"
'# CONFIG_PACKAGE_libcares is not set
CONFIG_PACKAGE_libdevmapper=y
'# CONFIG_PACKAGE_libev is not set
CONFIG_PACKAGE_liblzma=y
CONFIG_PACKAGE_liblzo=y
'# CONFIG_PACKAGE_libmbedtls is not set
'# CONFIG_PACKAGE_libminiupnpc is not set
'# CONFIG_PACKAGE_libnatpmp is not set
CONFIG_PACKAGE_libnetwork=y
CONFIG_PACKAGE_libruby=y
CONFIG_PACKAGE_libseccomp=y
'# CONFIG_PACKAGE_libsodium is not set
'# CONFIG_PACKAGE_libudns is not set
CONFIG_PACKAGE_libyaml=y
'# CONFIG_PACKAGE_lua-neturl is not set
'# CONFIG_PACKAGE_luci-app-adbyby-plus is not set
CONFIG_PACKAGE_luci-app-adguardhome=y
'# CONFIG_PACKAGE_luci-app-adguardhome_INCLUDE_binary is not set
'# CONFIG_PACKAGE_luci-app-arpbind is not set
'# CONFIG_PACKAGE_luci-app-autoreboot is not set
'# CONFIG_PACKAGE_luci-app-bypass_INCLUDE_ShadowsocksR_Libev_Client is not set
'# CONFIG_PACKAGE_luci-app-bypass_INCLUDE_ShadowsocksR_Libev_Server is not set
'# CONFIG_PACKAGE_luci-app-bypass_INCLUDE_Shadowsocks_Libev_Client is not set
'# CONFIG_PACKAGE_luci-app-bypass_INCLUDE_Shadowsocks_Libev_Server is not set
'# CONFIG_PACKAGE_luci-app-bypass_INCLUDE_Simple_Obfs is not set
'# CONFIG_PACKAGE_luci-app-bypass_INCLUDE_Socks_Server is not set
'# CONFIG_PACKAGE_luci-app-bypass_INCLUDE_Trojan is not set
'# CONFIG_PACKAGE_luci-app-bypass_INCLUDE_Xray is not set
'# CONFIG_PACKAGE_luci-app-ddns is not set
CONFIG_PACKAGE_luci-app-dockerman=y
'# CONFIG_PACKAGE_luci-app-ipsec-vpnd is not set
CONFIG_PACKAGE_luci-app-mosdns=y
'# CONFIG_PACKAGE_luci-app-nlbwmon is not set
CONFIG_PACKAGE_luci-app-openclash=y
'# CONFIG_PACKAGE_luci-app-rclone_INCLUDE_rclone-ng is not set
'# CONFIG_PACKAGE_luci-app-rclone_INCLUDE_rclone-webui is not set
'# CONFIG_PACKAGE_luci-app-ssr-plus is not set
'# CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_ChinaDNS_NG is not set
'# CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_ShadowsocksR_Libev_Client is not set
'# CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_ShadowsocksR_Libev_Server is not set
'# CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_Shadowsocks_Simple_Obfs is not set
'# CONFIG_PACKAGE_luci-app-turboacc is not set
'# CONFIG_PACKAGE_luci-app-turboacc_INCLUDE_BBR_CCA is not set
'# CONFIG_PACKAGE_luci-app-turboacc_INCLUDE_PDNSD is not set
'# CONFIG_PACKAGE_luci-app-turboacc_INCLUDE_SHORTCUT_FE_CM is not set
'# CONFIG_PACKAGE_luci-app-unblockmusic is not set
'# CONFIG_PACKAGE_luci-app-unblockmusic_INCLUDE_UnblockNeteaseMusic_Go is not set
'# CONFIG_PACKAGE_luci-app-upnp is not set
'# CONFIG_PACKAGE_luci-app-vsftpd is not set
'# CONFIG_PACKAGE_luci-app-vssr_INCLUDE_Hysteria is not set
'# CONFIG_PACKAGE_luci-app-vssr_INCLUDE_ShadowsocksR_Libev_Server is not set
'# CONFIG_PACKAGE_luci-app-vssr_INCLUDE_Trojan is not set
'# CONFIG_PACKAGE_luci-app-vssr_INCLUDE_Xray is not set
'# CONFIG_PACKAGE_luci-app-vssr_INCLUDE_Xray_plugin is not set
'# CONFIG_PACKAGE_luci-app-wireguard is not set
'# CONFIG_PACKAGE_luci-app-wol is not set
'# CONFIG_PACKAGE_luci-app-xlnetacc is not set
'# CONFIG_PACKAGE_luci-app-zerotier is not set
CONFIG_PACKAGE_luci-compat=y
CONFIG_PACKAGE_luci-i18n-adguardhome-zh-cn=y
CONFIG_PACKAGE_luci-i18n-dockerman-zh-cn=y
CONFIG_PACKAGE_luci-i18n-mosdns-zh-cn=y
CONFIG_PACKAGE_luci-lib-docker=y
'# CONFIG_PACKAGE_luci-lib-ipkg is not set
'# CONFIG_PACKAGE_luci-proto-wireguard is not set
CONFIG_PACKAGE_luci-theme-tomato=y
'# CONFIG_PACKAGE_microsocks is not set
'# CONFIG_PACKAGE_miniupnpd is not set
CONFIG_PACKAGE_mosdns=y
CONFIG_PACKAGE_mount-utils=y
'# CONFIG_PACKAGE_nlbwmon is not set
'# CONFIG_PACKAGE_pdnsd-alt is not set
'# CONFIG_PACKAGE_resolveip is not set
CONFIG_PACKAGE_ruby=y
CONFIG_PACKAGE_ruby-bigdecimal=y
CONFIG_PACKAGE_ruby-date=y
CONFIG_PACKAGE_ruby-digest=y
CONFIG_PACKAGE_ruby-enc=y
CONFIG_PACKAGE_ruby-forwardable=y
CONFIG_PACKAGE_ruby-pstore=y
CONFIG_PACKAGE_ruby-psych=y
CONFIG_PACKAGE_ruby-stringio=y
CONFIG_PACKAGE_ruby-strscan=y
CONFIG_PACKAGE_ruby-yaml=y
CONFIG_PACKAGE_runc=y
'# CONFIG_PACKAGE_shadowsocks-libev-ss-local is not set
'# CONFIG_PACKAGE_shadowsocks-libev-ss-redir is not set
'# CONFIG_PACKAGE_shadowsocks-libev-ss-server is not set
'# CONFIG_PACKAGE_shadowsocksr-libev-ssr-check is not set
'# CONFIG_PACKAGE_shadowsocksr-libev-ssr-local is not set
'# CONFIG_PACKAGE_shadowsocksr-libev-ssr-redir is not set
'# CONFIG_PACKAGE_shadowsocksr-libev-ssr-server is not set
'# CONFIG_PACKAGE_simple-obfs is not set
'# CONFIG_PACKAGE_strongswan is not set
'# CONFIG_PACKAGE_tcping is not set
CONFIG_PACKAGE_tini=y
CONFIG_PACKAGE_unzip=y
CONFIG_PACKAGE_v2dat=y
CONFIG_PACKAGE_v2ray-geoip=y
CONFIG_PACKAGE_v2ray-geosite=y
'# CONFIG_PACKAGE_vsftpd-alt is not set
'# CONFIG_PACKAGE_wireguard-tools is not set
'# CONFIG_PACKAGE_wol is not set
'# CONFIG_PACKAGE_xray-core is not set
CONFIG_PACKAGE_xz=y
CONFIG_PACKAGE_xz-utils=y
'# CONFIG_PACKAGE_zerotier is not set
CONFIG_VHDX_IMAGES=y
'# CONFIG_VMDK_IMAGES is not set

closed 2023-07-06 by happymem2023

luoei, 2024-04-04:
哈喽，这个问题解决了吗？

likevkin, 2024-12-03:
PARTUUID=xxxx... 我理解是没有识别到这个系统启动盘。需要查一下你的uuid  然后再boot gure 里面修改一下你的uuid