coolsnowwolf/lede #11307: 编译几次一直报错。

xtkong, 2023-06-13:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译报错

### (2) 路由器型号和固件版本

A：Phicomm K3 ARMv7 

### (3) 详细日志

A：mkdir -p /home/xtkong/lede/staging_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/stamp

SHELL= flock /home/xtkong/lede/tmp/.root-copy.flock -c 'cp -fpR /home/xtkong/lede/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/packages/.pkgdir/kernel/. /home/xtkong/lede/staging_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/'
touch /home/xtkong/lede/staging_dir/target-arm_cortex-a9_musl_eabi/root-bcm53xx/stamp/.kernel_installed
ERROR: module '/home/xtkong/lede/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/linux-5.4.246/arch/arm/crypto/sha1-arm.ko' is missing.
modules/crypto.mk:917: recipe for target '/home/xtkong/lede/bin/targets/bcm53xx/generic/packages/kmod-crypto-sha1_5.4.246-1_arm_cortex-a9.ipk' failed
make[3]: *** [/home/xtkong/lede/bin/targets/bcm53xx/generic/packages/kmod-crypto-sha1_5.4.246-1_arm_cortex-a9.ipk] Error 1
make[3]: Leaving directory '/home/xtkong/lede/package/kernel/linux'
time: package/kernel/linux/compile#5.38#0.68#6.12
    ERROR: package/kernel/linux failed to build.
package/Makefile:114: recipe for target 'package/kernel/linux/compile' failed
make[2]: *** [package/kernel/linux/compile] Error 1
make[2]: Leaving directory '/home/xtkong/lede'
package/Makefile:110: recipe for target '/home/xtkong/lede/staging_dir/target-arm_cortex-a9_musl_eabi/stamp/.package_compile' failed
make[1]: *** [/home/xtkong/lede/staging_dir/target-arm_cortex-a9_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/xtkong/lede'
/home/xtkong/lede/include/toplevel.mk:229: recipe for target 'world' failed
make: *** [world] Error 2

qingshujun, 2023-06-17:
ERROR: module '/home/xtkong/lede/build_dir/target-arm_cortex-a9_musl_eabi/linux-bcm53xx_generic/linux-5.4.246/arch/arm/crypto/sha1-arm.ko' is missing.
我编译k3，最近也是出现这个错误

coolsnowwolf, 2026-09-05:
已对照当前 LEDE 与 OpenWrt 主线的 package/kernel/linux/modules/crypto.mk：两者均按 ARM 架构条件为 crypto-sha1 选择 arch/arm/crypto/sha1-arm.ko，代码没有该 issue 所述的固定 BCM53xx 路径错误。原日志来自 2023 年 Linux 5.4.246，缺少最终 .config、CONFIG_CRYPTO_SHA1_ARM 值和完整配置流程；5.4 也已不是当前默认内核。请在当前版本若仍复现时附目标内核版本、最终 .config、内核构建日志及 ls arch/arm/crypto 后重新提交。本旧 issue 关闭为过时配置/缺少复现信息。

closed 2026-09-05 by coolsnowwolf