coolsnowwolf/lede #11308: 近期cr660x ap无线速度上限40MB/s

liwenjie119, 2023-06-13:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：作为ap使用，不知道从哪个版本开始测试cr660x的无线速度暴跌，之前最高能跑到90MB/s，现在只能跑不到一半了，全新刷机也是有问题

### (2) 路由器型号和固件版本

A：cr660x 最近版本

### (3) 详细日志

A：

liwenjie119, 2023-06-13:
测试了一下删除config全清缓存编译基础固件，测试速度最大能到47-49mb/s，强了一点但还是没有完全跑起来，是不是最近版本的无线驱动不给力？

DoveKi, 2023-06-14:
换主线或者xwrt吧 这边的ramipis系早就被当玩具玩了
一会切5.4内核一会切5.10内核
一会开源驱动 一会闭源驱动

liwenjie119, 2023-06-14:
> xwrt

主线和xwrt的网速没问题吗？这样的话我就切换到主线去

DoveKi, 2023-06-14:
> > xwrt
> 
> 主线和xwrt的网速没问题吗？这样的话我就切换到主线去

现在ramips机型建议用@MeIsReallyBa大佬的源码https://github.com/padavanonly/immortalwrt/
这里的ramips hwnat驱动其实也是合并他的代码 后面还一直瞎改 三天两天来回切换内核版本和无线驱动 导致bug一堆

aiamadeus, 2023-06-15:
@DoveKi 爱用不用，你行你来维护

coolsnowwolf, 2023-06-17:
可以尝试切换 5.10 内核

No06, 2023-09-30:
5.10 内核速度也上不去，甚至比 5.4 差

### 5.4 LAN
[image]

### 5.10 LAN
[image]

### 5.4 WLAN
[image]

### 5.10 WLAN
[image]

WRMing1008, 2023-10-10:
看来并不是只有我acrh17无线有问题，一样只能跑一半网速。隔壁6.1内核的问题更严重，只能30-40%

closed 2026-09-05 by coolsnowwolf