coolsnowwolf/lede #11310: Unexpected DP dual mode adaptor ID 7f

pretendtobefree, 2023-06-14:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ X] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：系统启动后在系统日志中反复出现“i915 0000:00:02.0: [drm] *ERROR* Unexpected DP dual mode adaptor ID 7f”错误提示，不知道应该如何解决，望赐教，谢谢！

### (2) 路由器型号和固件版本
CPU: Intel(R) Core(TM) i5-10400 
固件版本 OpenWrt R23.6.6 / LuCI Master (git-23.163.54516-50f4225)

### (3) 详细日志
Wed Jun 14 09:13:34 2023 kern.info kernel: [   21.208885] [drm] amdgpu kernel modesetting enabled.
Wed Jun 14 09:13:34 2023 kern.info kernel: [   21.243035] i915 0000:00:02.0: [drm] Finished loading DMC firmware i915/kbl_dmc_ver1_04.bin (v1.4)
Wed Jun 14 09:13:34 2023 kern.info kernel: [   21.500179] i915 0000:00:02.0: [drm] failed to retrieve link info, disabling eDP
Wed Jun 14 09:13:34 2023 kern.info kernel: [   21.575416] i915 0000:00:02.0: [drm] GuC firmware i915/kbl_guc_70.1.1.bin version 70.1.1
Wed Jun 14 09:13:34 2023 kern.info kernel: [   21.576117] i915 0000:00:02.0: [drm] HuC firmware i915/kbl_huc_4.0.0.bin version 4.0.0
Wed Jun 14 09:13:34 2023 kern.info kernel: [   21.602391] i915 0000:00:02.0: [drm] HuC authenticated
Wed Jun 14 09:13:34 2023 kern.info kernel: [   21.602885] i915 0000:00:02.0: [drm] GuC submission disabled
Wed Jun 14 09:13:34 2023 kern.info kernel: [   21.603404] i915 0000:00:02.0: [drm] GuC SLPC disabled
Wed Jun 14 09:13:34 2023 kern.info kernel: [   21.605497] [drm] Initialized i915 1.6.0 20201103 for 0000:00:02.0 on minor 0
Wed Jun 14 09:13:34 2023 kern.info kernel: [   21.607298] ACPI: video: Video Device [GFX0] (multi-head: yes  rom: no  post: no)
Wed Jun 14 09:13:34 2023 kern.info kernel: [   21.610483] input: Video Bus as /devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/LNXVIDEO:00/input/input7
Wed Jun 14 09:13:34 2023 kern.err kernel: [   21.989487] i915 0000:00:02.0: [drm] *ERROR* Unexpected DP dual mode adaptor ID 7f
Wed Jun 14 09:13:34 2023 kern.info kernel: [   22.031699] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
Wed Jun 14 09:13:34 2023 kern.info kernel: [   22.379347] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
Wed Jun 14 09:13:34 2023 kern.err kernel: [   22.767674] i915 0000:00:02.0: [drm] *ERROR* Unexpected DP dual mode adaptor ID 7f
Wed Jun 14 09:13:34 2023 kern.info kernel: [   22.803043] i915 0000:00:02.0: [drm] Cannot find any crtc or sizes
….
…

Wed Jun 14 10:47:01 2023 kern.err kernel: [ 1403.959760] i915 0000:00:02.0: [drm] *ERROR* Unexpected DP dual mode adaptor ID 7f
Wed Jun 14 10:47:47 2023 kern.err kernel: [ 1450.039767] i915 0000:00:02.0: [drm] *ERROR* Unexpected DP dual mode adaptor ID 7f
Wed Jun 14 10:47:47 2023 kern.err kernel: [ 1450.473093] i915 0000:00:02.0: [drm] *ERROR* Unexpected DP dual mode adaptor ID 7f
Wed Jun 14 10:48:10 2023 kern.err kernel: [ 1473.072426] i915 0000:00:02.0: [drm] *ERROR* Unexpected DP dual mode adaptor ID 7f
Wed Jun 14 10:48:11 2023 kern.err kernel: [ 1473.506547] i915 0000:00:02.0: [drm] *ERROR* Unexpected DP dual mode adaptor ID 7f
Wed Jun 14 10:48:56 2023 kern.err kernel: [ 1519.139744] i915 0000:00:02.0: [drm] *ERROR* Unexpected DP dual mode adaptor ID 7f
Wed Jun 14 10:49:19 2023 kern.err kernel: [ 1542.192671] i915 0000:00:02.0: [drm] *ERROR* Unexpected DP dual mode adaptor ID 7f
Wed Jun 14 10:49:20 2023 kern.err kernel: [ 1542.609872] i915 0000:00:02.0: [drm] *ERROR* Unexpected DP dual mode adaptor ID 7f
Wed Jun 14 10:50:29 2023 kern.err kernel: [ 1611.708475] i915 0000:00:02.0: [drm] *ERROR* Unexpected DP dual mode adaptor ID 7f
Wed Jun 14 10:52:00 2023 kern.err kernel: [ 1703.462230] i915 0000:00:02.0: [drm] *ERROR* Unexpected DP dual mode adaptor ID 7f

G-Pereira, 2023-08-19:
I am also having this issue with:

Motherboard: AsRock N100DC-ITX
CPU: Intel N100 (onboard)
RAM: Kingston Fury 16GB
PSU: 19V 65W Barrel jack
HDD: 2x 2TB WD Red

daniel-rabe, 2024-02-25:
i see the same error spam in logs

ASROCK N100DC-ITX
RAM: Corsair Vengeance LPX 8GB
PSU: 90W Leike
HDD: 4TB WD Red, 4TB WD Blue, HP SSD EX900 M.2 250GB

coolsnowwolf, 2026-09-05:
该日志来自 2023 年 i5-10400（Comet Lake）在无有效 DP/转换器连接时的 i915 探测：`Unexpected DP dual mode adaptor ID 7f` 与 `failed to retrieve link info, disabling eDP`/`Cannot find any crtc or sizes` 一致，属于显示输出链路/适配器识别提示，不影响 x86 软路由网络功能。没有提供实际显示故障或当前版本复现信息，LEDE 无需修改驱动；建议不需要显示时禁用 i915 或忽略该日志。

closed 2026-09-05 by coolsnowwolf