coolsnowwolf/lede #11313: openssl: bump to 1.1.1u 后云编译报错 x86机型

xayane, 2023-06-16:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ X] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：
openssl: bump to 1.1.1u
增加了两个补丁文件：
package/libs/openssl/patches/200-x509-excessive-resource-use-verifying-policy-constra.patch
package/libs/openssl/patches/210-Ensure-that-EXFLAG_INVALID_POLICY-is-checked-even-in.patch
云编译过程中报错，说要修复该patch文件。
请教各位大佬，这个patch文件该如何修改。

### (2) 路由器型号和固件版本

A：x86 R23.6.6

### (3) 详细日志

A：
Applying ./patches/200-x509-excessive-resource-use-verifying-policy-constra.patch using plaintext: 
patching file crypto/x509v3/pcy_local.h
Hunk #1 FAILED at 111.
Hunk #2 FAILED at 159.
2 out of 2 hunks FAILED -- saving rejects to file crypto/x509v3/pcy_local.h.rej
patching file crypto/x509v3/pcy_node.c
Hunk #1 FAILED at 59.
Hunk #2 FAILED at 70.
Hunk #3 FAILED at 90.
Hunk #4 FAILED at 103.
4 out of 4 hunks FAILED -- saving rejects to file crypto/x509v3/pcy_node.c.rej
patching file crypto/x509v3/pcy_tree.c
Hunk #1 succeeded at 13 with fuzz 2.
Hunk #2 succeeded at 197 with fuzz 2 (offset 17 lines).
Hunk #3 FAILED at 199.
Hunk #4 FAILED at 258.
Hunk #5 FAILED at 268.
Hunk #6 FAILED at 287.
Hunk #7 FAILED at 295.
Hunk #8 FAILED at 326.
Hunk #9 FAILED at 388.
Hunk #10 FAILED at 570.
Hunk #11 FAILED at 597.
9 out of 11 hunks FAILED -- saving rejects to file crypto/x509v3/pcy_tree.c.rej
Patch failed!  Please fix ./patches/200-x509-excessive-resource-use-verifying-policy-constra.patch!

WYC-2020, 2023-06-16:
没遇到这个问题，一切正常

WYC-2020, 2023-06-16:
并且我记得更新1.1.1u是删除这两个补丁 何时变成了新增？

xayane, 2023-06-16:
> 

我再看看，是不是我操作错了，感谢。

closed 2023-06-16 by xayane