coolsnowwolf/lede #11320: vm-tool编译出错

free-diy, 2023-06-22:
### (2) 路由器型号和固件版本

A：x86

### (3) 详细日志
autoreconf: Leaving directory `.'
(cd /workdir/openwrt/build_dir/target-x86_64_musl/open-vm-tools-11.3.0-18090558/./; if [ -x ./configure ]; then find /workdir/openwrt/build_dir/target-x86_64_musl/open-vm-tools-11.3.0-18090558/ -name config.guess | xargs -r chmod u+w; find /workdir/openwrt/build_dir/target-x86_64_musl/open-vm-tools-11.3.0-18090558/ -name config.guess | xargs -r -n1 cp --remove-destination /workdir/openwrt/scripts/config.guess; find /workdir/openwrt/build_dir/target-x86_64_musl/open-vm-tools-11.3.0-18090558/ -name config.sub | xargs -r chmod u+w; find /workdir/openwrt/build_dir/target-x86_64_musl/open-vm-tools-11.3.0-18090558/ -name config.sub | xargs -r -n1 cp --remove-destination /workdir/openwrt/scripts/config.sub; AR="x86_64-openwrt-linux-musl-gcc-ar" AS="x86_64-openwrt-linux-musl-gcc -c -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/workdir/openwrt/build_dir/target-x86_64_musl/open-vm-tools-11.3.0-18090558=open-vm-tools-11.3.0-18090558 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  " LD=x86_64-openwrt-linux-musl-ld NM="x86_64-openwrt-linux-musl-gcc-nm" CC="x86_64-openwrt-linux-musl-gcc" GCC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" RANLIB="x86_64-openwrt-linux-musl-gcc-ranlib" STRIP=x86_64-openwrt-linux-musl-strip OBJCOPY=x86_64-openwrt-linux-musl-objcopy OBJDUMP=x86_64-openwrt-linux-musl-objdump SIZE=x86_64-openwrt-linux-musl-size CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/workdir/openwrt/build_dir/target-x86_64_musl/open-vm-tools-11.3.0-18090558=open-vm-tools-11.3.0-18090558 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro   " CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/workdir/openwrt/build_dir/target-x86_64_musl/open-vm-tools-11.3.0-18090558=open-vm-tools-11.3.0-18090558 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro   " CPPFLAGS="-I/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/include -I/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include/fortify -I/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include   " LDFLAGS="-L/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/lib -L/workdir/openwrt/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib -znow -zrelro   -liconv "   ./configure --target=x86_64-openwrt-linux --host=x86_64-openwrt-linux --build=x86_64-pc-linux-gnu --program-prefix="" --program-suffix="" --prefix=/usr --exec-prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --libexecdir=/usr/lib --sysconfdir=/etc --datadir=/usr/share --localstatedir=/var --mandir=/usr/man --infodir=/usr/info --disable-nls  --without-icu --disable-multimon --disable-docs --disable-tests --without-gtkmm --without-gtkmm3 --without-xerces --without-pam --disable-grabbitmqproxy --disable-vgauth --disable-deploypkg --without-root-privileges --without-kernel-modules --without-dnet --with-tirpc --without-x --without-gtk2 --without-gtk3 --without-xerces --enable-resolutionkms=no ; fi; )
configure: WARNING: unrecognized options: --disable-nls, --disable-grabbitmqproxy
configure: loading site script /workdir/openwrt/include/site/x86_64
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-openwrt-linux-gnu
checking for x86_64-openwrt-linux-gcc... x86_64-openwrt-linux-musl-gcc
checking whether the C compiler works... no
configure: error: in `/workdir/openwrt/build_dir/target-x86_64_musl/open-vm-tools-11.3.0-18090558':
configure: error: C compiler cannot create executables
See `config.log' for more details
make[3]: *** [Makefile:150: /workdir/openwrt/build_dir/target-x86_64_musl/open-vm-tools-11.3.0-18090558/.configured_a17fb5ef857664f03cd0ce37cc5ea591] Error 77
make[3]: Leaving directory '/workdir/openwrt/feeds/packages/utils/open-vm-tools'
time: package/feeds/packages/open-vm-tools/compile#12.10#0.43#13.43
    ERROR: package/feeds/packages/open-vm-tools failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/open-vm-tools/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [package/Makefile:110: /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2
Error: Process completed with exit code 2.
A：

xpc0938, 2023-06-23:
同上

LIUBANGJIAN, 2023-06-23:
相同报错

LIUBANGJIAN, 2023-06-25:
L大佛系了

LFFAnd, 2023-06-25:
相同错误 有没有大佬帮忙解决解决

xibaolai, 2023-06-25:
make clean

yoseg, 2023-06-25:
+1 希望得到解决

LIUBANGJIAN, 2023-06-26:
自己添加官方open-vm-tools的源吧

WYC-2020, 2023-06-26:
这玩意 l 大不是已经撤回了么 ，撤回了还有错误？

free-diy, 2023-06-26:
> 自己添加官方Open-VM-tools的源吧

小白不懂，大神给个源的链接啊

LIUBANGJIAN, 2023-06-26:
> > 自己添加官方Open-VM-tools的源吧
> 
> 小白不懂，大神给个源的链接啊

https://github.com/vmware/open-vm-tools.git

WYC-2020, 2023-06-26:
那有那么麻烦，直接看关闭的 有一个和这个类似，直接吧l的删除了，再用官方openwrt 的 Open-VM-tools放进去就行了，自动就升级到了12.2了。我看l已经回滚了所以这个问题应该是不存在的，完全可以关闭了，你编译不过 自己在update 一下 在编译就能过的

closed 2023-06-27 by coolsnowwolf