coolsnowwolf/lede #11325: 最新的commit  xt_FLOWOFFLOAD.ko 没有生成

littoy, 2023-06-26:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：x86_64

### (3) 详细日志

A：
commit:  [3e35e077602c4021da4ab0f1c5a993b324d8023d](https://github.com/coolsnowwolf/lede/commit/3e35e077602c4021da4ab0f1c5a993b324d8023d)
即使选择了Flow Offloading， xt_FLOWOFFLOAD.ko 也没有生成，暂时不清楚哪个依赖导致的。
可以确定的是最终生成的build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.35/.config 里没有NETFILTER_XT_TARGET_FLOWOFFLOAD这个配置。

GreatMichaelLee, 2023-06-26:
我编译完最新的版本后在x86上网都不怎么通，丢包，时延大，内网都是这样子。回退上个版本没事。

WYC-2020, 2023-06-26:
[image]
https://github.com/coolsnowwolf/lede/commit/398eece1d6c3bc6c849d30cac58f1c368f26f62d#diff-9273e5adce6306feebaaac593c989d7d8176c62d1e71a4737031219ccd3f0802 在这个提交之前 config里面定义了CONFIG_NF_TABLES的所以依赖是全的，这个提交之后吧这个改成未定义了，结果 offload 间接必须依赖这个，所以最终导致这个打不开，我参考其他内核吧patch更新了一下 你用这个patch替换 编译试下功能，功能正常就可以这样搞了，如果不行你可以吧config这个宏定义还原回来并且吧我给的那个650patch还原成l的原版patch,再编译就可以了，

commit c22e0ea3 (2023-06-26): generic: fixes 6.1 nf-flow support with fw3

Fixes: #11325
closed 2023-06-26 by aiamadeus