coolsnowwolf/lede #11327: threading-for-backlog-processi.patch function broken

GreatMichaelLee, 2023-06-26:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：[.../linux/generic/pending-6.1/760-net-core-add-optional-threading-for-backlog-processi.patch](https://github.com/coolsnowwolf/lede/commit/398eece1d6c3bc6c849d30cac58f1c368f26f62d#diff-b730c02d144162064befe886669c980c00477044ce2a97ec3ec266d8310acfaf) 这个功能在e7bd136138ee18022ff4331d36c7fb17e60f95a1这个commit之后broken了， 

如果启动时在rc.local里加入 echo 1 > /proc/sys/net/core/backend_threaded 则重启后软路由WAN/LAN端会大量丢包和出现延时，
在脚本里恢复echo 0 重启，故障消失。使能的1在这个commit之前是好的，正常使用。

### (2) 路由器型号和固件版本

A：X86 N5105, 固件为本仓库最新代码

### (3) 详细日志

A：无

WYC-2020, 2023-06-26:
那肯定好的了，那是我搞的嘛。你当小白鼠测的你忘记了？哈哈，这个版本没看

GreatMichaelLee, 2023-06-27:
> 那肯定好的了，那是我搞的嘛。你当小白鼠测的你忘记了？哈哈，这个版本没看

自己打屁股吧，哈哈

GreatMichaelLee, 2023-07-06:
老板们，有人能修下不？

GreatMichaelLee, 2023-07-28:
@coolsnowwolf @WYC-2020 大佬们能否修一下啊.....感谢

GreatMichaelLee, 2023-07-28:
也请一并看下这个issue, 虽然这个issue由passwall触发，但不确定是不是系统的问题，我为啥一直想要修这个bug，因为似乎启用了这个feature之后，就没这个现象了...... @WYC-2020 @coolsnowwolf 

https://github.com/xiaorouji/openwrt-passwall/issues/2677

GreatMichaelLee, 2023-09-28:
老板们，有没有人能修一下这个问题呢？

GreatMichaelLee, 2023-10-27:
还是没人修啊，有大佬关注一下么，只要用这个feature 必然高CPU和假死。 echo 1 > /proc/sys/net/core/backlog_threaded @WYC-2020 @coolsnowwolf 
最新的6.1.60 也patch了相关的patch,但根本不能用....

closed 2026-09-05 by coolsnowwolf