coolsnowwolf/lede #11328: Ubus failed to build

XiaoliChan, 2023-06-27:
问题：
- Ubus 编译失败

平台：
- x86_64

日志：
```
make[3]: Entering directory '/home/xiaoli/lede/package/system/ubus'
rm -f /home/xiaoli/lede/build_dir/target-x86_64_musl/ubus-2022-06-01-2bebf93c/.built
touch /home/xiaoli/lede/build_dir/target-x86_64_musl/ubus-2022-06-01-2bebf93c/.built_check
MAKEFLAGS="" /home/xiaoli/lede/staging_dir/host/bin/ninja  -j1 -C /home/xiaoli/lede/build_dir/target-x86_64_musl/ubus-2022-06-01-2bebf93c 
ninja: Entering directory `/home/xiaoli/lede/build_dir/target-x86_64_musl/ubus-2022-06-01-2bebf93c'
[1/22] Linking C shared library libubus.so.20220601
FAILED: libubus.so.20220601 
: && /home/xiaoli/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/bin/x86_64-openwrt-linux-musl-gcc -fPIC -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/xiaoli/lede/build_dir/target-x86_64_musl/ubus-2022-06-01-2bebf93c=ubus-2022-06-01-2bebf93c -Wformat -Werror=format-security -DPIC -fpic -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -I/home/xiaoli/lede/staging_dir/target-x86_64_musl/usr/include -flto -DNDEBUG  -L/home/xiaoli/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/lib -L/home/xiaoli/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib -DPIC -fpic -specs=/home/xiaoli/lede/include/hardened-ld-pie.specs -znow -zrelro -flto -Wl,-Bsymbolic-functions -shared -Wl,-soname,libubus.so.20220601 -o libubus.so.20220601 CMakeFiles/ubus.dir/libubus.c.o CMakeFiles/ubus.dir/libubus-io.c.o CMakeFiles/ubus.dir/libubus-obj.c.o CMakeFiles/ubus.dir/libubus-sub.c.o CMakeFiles/ubus.dir/libubus-req.c.o CMakeFiles/ubus.dir/libubus-acl.c.o  -lubox && :
lto1: internal compiler error: original not compressed with zstd
0x94e91b lto_uncompression_zstd
        /home/xiaoli/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0/gcc/lto-compress.c:163
0x94e91b lto_end_uncompression(lto_compression_stream*, lto_compression)
        /home/xiaoli/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0/gcc/lto-compress.c:410
0x94d140 lto_get_section_data(lto_file_decl_data*, lto_section_type, char const*, int, unsigned long*, bool)
        /home/xiaoli/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0/gcc/lto-section-in.c:168
0x6bc89c cgraph_node::get_untransformed_body()
        /home/xiaoli/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0/gcc/cgraph.c:3947
0x6c6d7d cgraph_node::expand()
        /home/xiaoli/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0/gcc/cgraphunit.c:1801
0x6c6d7d cgraph_node::expand()
        /home/xiaoli/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0/gcc/cgraphunit.c:1783
0x6c830e expand_all_functions
        /home/xiaoli/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0/gcc/cgraphunit.c:1994
0x6c830e symbol_table::compile()
        /home/xiaoli/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0/gcc/cgraphunit.c:2358
0x6336d9 lto_main()
        /home/xiaoli/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0/gcc/lto/lto.c:653
Please submit a full bug report,
with preprocessed source if appropriate.
Please include the complete backtrace with any bug report.
See <http://bugs.openwrt.org/> for instructions.
lto-wrapper: fatal error: /home/xiaoli/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/bin/x86_64-openwrt-linux-musl-gcc returned 1 exit status
compilation terminated.
/home/xiaoli/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib/gcc/x86_64-openwrt-linux-musl/11.3.0/../../../../x86_64-openwrt-linux-musl/bin/ld: error: lto-wrapper failed
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
make[3]: *** [Makefile:83: /home/xiaoli/lede/build_dir/target-x86_64_musl/ubus-2022-06-01-2bebf93c/.built] Error 1
make[3]: Leaving directory '/home/xiaoli/lede/package/system/ubus'
time: package/system/ubus/compile#0.50#0.21#4.81
    ERROR: package/system/ubus failed to build.
make[2]: *** [package/Makefile:116: package/system/ubus/compile] Error 1
make[2]: Leaving directory '/home/xiaoli/lede'
make[1]: *** [package/Makefile:110: /home/xiaoli/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/xiaoli/lede'
make: *** [/home/xiaoli/lede/include/toplevel.mk:231: world] Error 2
```

WYC-2020, 2023-06-27:
复现不了，我这边一切正常

XiaoliChan, 2023-06-27:
> 复现不了，我这边一切正常

config文件能来一份吗？我diff看看

XiaoliChan, 2023-06-30:
Toolchains 需要更新，手动更新完toolchains后，问题解决

closed 2023-06-30 by XiaoliChan