coolsnowwolf/lede #11329: 6.1.35内核编译错误

coldrook, 2023-06-27:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：内核编译出错，exit code 2，之前编译从未出现过类似错误

### (2) 路由器型号和固件版本

A：x86 软路由 x550t2网卡，固件版本git-23.163.54516-50f4225

### (3) 详细日志

A：
 make[3] -C target/linux compile
    ERROR: target/linux failed to build.
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 1
make[2]: Entering directory '/workdir/openwrt/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/workdir/openwrt/scripts/config'
make[1]: Entering directory '/workdir/openwrt'
make[2]: Entering directory '/workdir/openwrt'
make[3]: Entering directory '/workdir/openwrt/target/linux'
make[4]: Entering directory '/workdir/openwrt/target/linux/x86'
rm -f /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.35/localversion
/workdir/openwrt/scripts/kconfig.pl  + + /workdir/openwrt/target/linux/generic/config-6.1 /workdir/openwrt/target/linux/x86/config-6.1 /workdir/openwrt/target/linux/x86/64/config-6.1 > /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.35/.config.target
awk '/^(#[[:space:]]+)?CONFIG_KERNEL/{sub("CONFIG_KERNEL_","CONFIG_");print}' /workdir/openwrt/.config >> /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.35/.config.target
echo "# CONFIG_KALLSYMS_EXTRA_PASS is not set" >> /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.35/.config.target
echo "# CONFIG_KALLSYMS_ALL is not set" >> /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.35/.config.target
echo "CONFIG_KALLSYMS_UNCOMPRESSED=y" >> /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.35/.config.target
/workdir/openwrt/scripts/package-metadata.pl kconfig /workdir/openwrt/tmp/.packageinfo /workdir/openwrt/.config 6.1 > /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.35/.config.override
/workdir/openwrt/scripts/kconfig.pl 'm+' '+' /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.35/.config.target /dev/null /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.35/.config.override > /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.35/.config.set
mv /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.35/.config.set /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.35/.config.old
grep -v INITRAMFS /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.35/.config.old > /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.35/.config.set
echo 'CONFIG_INITRAMFS_SOURCE=""' >> /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.35/.config.set
echo '# CONFIG_INITRAMFS_FORCE is not set' >> /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.35/.config.set
rm -rf /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/modules
cmp -s /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.35/.config.set /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.35/.config.prev || { cp /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.35/.config.set /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.35/.config; cp /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.35/.config.set /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.35/.config.prev; }
export MAKEFLAGS= ; [ -d /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.35/user_headers ] || make -C /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.35 KCFLAGS="-fmacro-prefix-map=/workdir/openwrt/build_dir/target-x86_64_musl=target-x86_64_musl -fno-caller-saves " HOSTCFLAGS="-O2 -I/workdir/openwrt/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="x86_64-openwrt-linux-musl-" ARCH="x86" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Tue Jun 27 15:08:57 2023" KBUILD_BUILD_VERSION="0" KBUILD_HOSTLDFLAGS="-L/workdir/openwrt/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls=  KERNELRELEASE=6.1.35 CC="ccache_cc"  INSTALL_HDR_PATH=/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.35/user_headers headers_install
grep '=[ym]' /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.35/.config.set | LC_ALL=C sort | /workdir/openwrt/staging_dir/host/bin/mkhash md5 > /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.35/.vermagic
touch /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.35/.configured
rm -f /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.35/vmlinux /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.35/System.map
make -C /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.35 KCFLAGS="-fmacro-prefix-map=/workdir/openwrt/build_dir/target-x86_64_musl=target-x86_64_musl -fno-caller-saves " HOSTCFLAGS="-O2 -I/workdir/openwrt/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="x86_64-openwrt-linux-musl-" ARCH="x86" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Tue Jun 27 15:08:57 2023" KBUILD_BUILD_VERSION="0" KBUILD_HOSTLDFLAGS="-L/workdir/openwrt/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls=  KERNELRELEASE=6.1.35 CC="ccache_cc" olddefconfig
make[5]: Entering directory '/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.35'
#
# No change to .config
#
make[5]: Leaving directory '/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.35'
make -C /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.35 KCFLAGS="-fmacro-prefix-map=/workdir/openwrt/build_dir/target-x86_64_musl=target-x86_64_musl -fno-caller-saves " HOSTCFLAGS="-O2 -I/workdir/openwrt/staging_dir/host/include  -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="x86_64-openwrt-linux-musl-" ARCH="x86" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Tue Jun 27 15:08:57 2023" KBUILD_BUILD_VERSION="0" KBUILD_HOSTLDFLAGS="-L/workdir/openwrt/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls=  KERNELRELEASE=6.1.35 CC="ccache_cc" bzImage modules
make[5]: Entering directory '/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.35'
  CALL    scripts/checksyscalls.sh
  DESCEND objtool
make[8]: *** No rule to make target 'drivers/ssb/fallback-sprom.o', needed by 'drivers/ssb/ssb.o'.  Stop.
make[7]: *** [scripts/Makefile.build:500: drivers/ssb] Error 2
make[6]: *** [scripts/Makefile.build:500: drivers] Error 2
make[5]: *** [Makefile:2012: .] Error 2
make[5]: Leaving directory '/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.35'
make[4]: Leaving directory '/workdir/openwrt/target/linux/x86'
make[4]: *** [Makefile:29: /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.35/.modules] Error 2
make[3]: Leaving directory '/workdir/openwrt/target/linux'
time: target/linux/compile#2.49#0.82#3.26
make[3]: *** [Makefile:11: compile] Error 2
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/compile] Error 1
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [target/Makefile:23: /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2
Error: Process completed with exit code 2.

coolsnowwolf, 2023-06-28:
缓存问题，make clean 一下就好了

coldrook, 2023-07-01:
自动编译取消缓存，make clean也没有什么效果，发生问题也是一致

WYC-2020, 2023-07-02:
那就贴单线程日志

coldrook, 2023-07-02:
看上去是driver/ssb编译错误了

2023-07-02T07:11:41.6562813Z   CC [M]  drivers/ssb/main.o
2023-07-02T07:11:41.9014038Z   CC [M]  drivers/net/ethernet/intel/ice/ice_fdir.o
2023-07-02T07:11:42.5648961Z make[8]: *** No rule to make target 'drivers/ssb/fallback-sprom.o', needed by 'drivers/ssb/ssb.o'.  Stop.
2023-07-02T07:11:42.5649787Z make[8]: *** Waiting for unfinished jobs....
2023-07-02T07:11:42.5654520Z   CC [M]  drivers/ssb/scan.o
2023-07-02T07:11:43.0709793Z   CC [M]  drivers/net/ethernet/intel/ice/ice_ethtool_fdir.o
2023-07-02T07:11:43.2618536Z make[7]: *** [scripts/Makefile.build:500: drivers/ssb] Error 2
2023-07-02T07:11:43.2618911Z make[7]: *** Waiting for unfinished jobs....

2023-07-02T07:14:52.2893659Z make[6]: *** [scripts/Makefile.build:500: drivers] Error 2
2023-07-02T07:14:52.2897898Z make[5]: Leaving directory '/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.35'
2023-07-02T07:14:52.2898520Z make[5]: *** [Makefile:2012: .] Error 2
2023-07-02T07:14:52.2899079Z make[4]: *** [Makefile:29: /workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.35/.modules] Error 2
2023-07-02T07:14:52.2906443Z make[3]: *** [Makefile:11: compile] Error 2
2023-07-02T07:14:52.2906924Z make[4]: Leaving directory '/workdir/openwrt/target/linux/x86'
2023-07-02T07:14:52.2907532Z make[3]: Leaving directory '/workdir/openwrt/target/linux'
2023-07-02T07:14:52.2908112Z time: target/linux/compile#3066.49#306.17#1718.01
2023-07-02T07:14:52.2909388Z     ERROR: target/linux failed to build.
2023-07-02T07:14:52.2925704Z make[2]: *** [target/Makefile:30: target/linux/compile] Error 1
2023-07-02T07:14:52.2926177Z make[2]: Leaving directory '/workdir/openwrt'
2023-07-02T07:14:52.2931160Z make[1]: Leaving directory '/workdir/openwrt'
2023-07-02T07:14:52.2934517Z make[1]: *** [target/Makefile:23: /workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.target_compile] Error 2
2023-07-02T07:14:52.2941908Z make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2
2023-07-02T07:14:52.3004279Z ##[error]Process completed with exit code 2.

xiangfeidexiaohuo, 2023-07-02:
> 看上去是driver/ssb编译错误了
>
> 2023-07-02T07:11:41.6562813Z CC [M] drivers/ssb/main.o 2023-07-02T07:11:41.9014038Z CC [M] drivers/net/ethernet/intel/ice/ice_fdir.o 2023-07-02T07:11:42.5648961Z make[8]: *** No [...]

同 x86编译出错的

WYC-2020, 2023-07-02:
新增的这个补丁有问题，要么是丢失了文件 要么就是这个补丁有问题，明明没有fallback-sprom.c这个文件 非要去 编译这个文件 自然出问题了，target\linux\generic\hack-6.1\810-bcma-ssb-fallback-sprom.patch 删除

WYC-2020, 2023-07-02:
看了下官方 是同步没同步全 有一个文件丢失了，自己从官方那边拷贝过来就可以了openwrt\target\linux\generic\files\drivers\ssb

WYC-2020, 2023-07-02:
openwrt\target\linux\generic\files\drivers\bcma  还有这里

1582130940, 2023-07-03:
# CONFIG_PACKAGE_kmod-bcma is not set
# CONFIG_PACKAGE_kmod-ssb is not set
要么disable掉bcma和ssb（关联kmod-b43，kmod-b44，也要disable）要么自己补

xiangfeidexiaohuo, 2023-07-03:
> openwrt\target\linux\generic\files\drivers\bcma 还有这里

确实 补了文件 能编译ok

closed 2023-07-05 by aiamadeus