coolsnowwolf/lede #11331: RTL8821CU驱动编译失败

LemoFire, 2023-06-29:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：RTL8821CU驱动编译失败

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：
```
make[3]: Entering directory '/home/shion/R23/lede/package/kernel/rtw88-usb'
rm -f /home/shion/R23/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-bcm27xx_bcm2709/rtw88-usb-2023-02-11-3af004dd/.built
touch /home/shion/R23/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-bcm27xx_bcm2709/rtw88-usb-2023-02-11-3af004dd/.built_check
make  -C "/home/shion/R23/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-bcm27xx_bcm2709/linux-5.15.116" KCFLAGS="-fmacro-prefix-map=/home/shion/R23/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi=target-arm_cortex-a7+neon-vfpv4_musl_eabi -fno-caller-saves " HOSTCFLAGS="-O2 -I/home/shion/R23/lede/staging_dir/host/include -I/home/shion/R23/lede/staging_dir/hostpkg/include -I/home/shion/R23/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/host/include -Wall -Wmissing-prototypes -Wstrict-prototypes" CROSS_COMPILE="arm-openwrt-linux-muslgnueabi-" ARCH="arm" KBUILD_HAVE_NLS=no KBUILD_BUILD_USER="" KBUILD_BUILD_HOST="" KBUILD_BUILD_TIMESTAMP="Wed Jun 28 15:38:20 2023" KBUILD_BUILD_VERSION="0" KBUILD_HOSTLDFLAGS="-L/home/shion/R23/lede/staging_dir/host/lib" CONFIG_SHELL="bash" V=''  cmd_syscalls= KBUILD_EXTRA_SYMBOLS="/home/shion/R23/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-bcm27xx_bcm2709/symvers/cryptodev-linux.symvers /home/shion/R23/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-bcm27xx_bcm2709/symvers/fullconenat.symvers /home/shion/R23/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-bcm27xx_bcm2709/symvers/mac80211.symvers" KERNELRELEASE=5.15.116 M="/home/shion/R23/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-bcm27xx_bcm2709/rtw88-usb-2023-02-11-3af004dd" NOSTDINC_FLAGS="-nostdinc  -isystem /home/shion/R23/lede/staging_dir/toolchain-arm_cortex-a7+neon-vfpv4_gcc-11.3.0_musl_eabi/lib/gcc/arm-openwrt-linux-muslgnueabi/11.3.0/include -I/home/shion/R23/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-bcm27xx_bcm2709/rtw88-usb-2023-02-11-3af004dd -I/home/shion/R23/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/include/mac80211-backport/uapi -I/home/shion/R23/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/include/mac80211-backport -I/home/shion/R23/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/include/mac80211/uapi -I/home/shion/R23/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/include/mac80211 -include backport/autoconf.h -include backport/backport.h -DBUILD_OPENWRT" modules
make[4]: Entering directory '/home/shion/R23/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-bcm27xx_bcm2709/linux-5.15.116'
  CC [M]  /home/shion/R23/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-bcm27xx_bcm2709/rtw88-usb-2023-02-11-3af004dd/main.o
/home/shion/R23/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-bcm27xx_bcm2709/rtw88-usb-2023-02-11-3af004dd/main.c: In function 'rtw_vif_watch_dog_iter':
/home/shion/R23/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-bcm27xx_bcm2709/rtw88-usb-2023-02-11-3af004dd/main.c:177:34: error: 'struct ieee80211_bss_conf' has no member named 'assoc'
  177 |                 if (vif->bss_conf.assoc)
      |                                  ^
/home/shion/R23/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-bcm27xx_bcm2709/rtw88-usb-2023-02-11-3af004dd/main.c: In function 'rtw_vif_assoc_changed':
/home/shion/R23/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-bcm27xx_bcm2709/rtw88-usb-2023-02-11-3af004dd/main.c:541:25: error: 'struct ieee80211_bss_conf' has no member named 'assoc'
  541 |         if (conf && conf->assoc) {
      |                         ^~
/home/shion/R23/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-bcm27xx_bcm2709/rtw88-usb-2023-02-11-3af004dd/main.c:542:35: error: 'struct ieee80211_bss_conf' has no member named 'aid'
  542 |                 rtwvif->aid = conf->aid;
      |                                   ^~
In file included from ./include/linux/byteorder/little_endian.h:5,
                 from ./arch/arm/include/uapi/asm/byteorder.h:22,
                 from ./include/asm-generic/bitops/le.h:7,
                 from ./arch/arm/include/asm/bitops.h:268,
                 from ./include/linux/bitops.h:33,
                 from /home/shion/R23/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/include/mac80211-backport/linux/bitops.h:3,
                 from ./include/linux/log2.h:12,
                 from ./include/asm-generic/div64.h:55,
                 from ./arch/arm/include/asm/div64.h:107,
                 from ./include/linux/math.h:5,
                 from /home/shion/R23/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/include/mac80211-backport/linux/math.h:7,
                 from ./include/linux/math64.h:6,
                 from /home/shion/R23/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/include/mac80211-backport/linux/math64.h:3,
                 from ./include/linux/time64.h:5,
                 from ./include/linux/restart_block.h:10,
                 from ./include/linux/thread_info.h:14,
                 from ./include/asm-generic/current.h:5,
                 from ./arch/arm/include/generated/asm/current.h:1,
                 from ./include/linux/sched.h:12,
                 from /home/shion/R23/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/include/mac80211-backport/linux/sched.h:4,
                 from ./include/linux/ratelimit.h:6,
                 from ./include/linux/dev_printk.h:16,
                 from ./include/linux/device.h:15,
                 from /home/shion/R23/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/include/mac80211-backport/linux/device.h:3,
                 from ./include/linux/devcoredump.h:8,
                 from /home/shion/R23/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/include/mac80211-backport/linux/devcoredump.h:3,
                 from /home/shion/R23/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-bcm27xx_bcm2709/rtw88-usb-2023-02-11-3af004dd/main.c:5:
/home/shion/R23/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-bcm27xx_bcm2709/rtw88-usb-2023-02-11-3af004dd/main.c: In function 'get_vht_ra_mask':
/home/shion/R23/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-bcm27xx_bcm2709/rtw88-usb-2023-02-11-3af004dd/main.c:959:38: error: 'struct ieee80211_sta' has no member named 'vht_cap'
  959 |         u16 mcs_map = le16_to_cpu(sta->vht_cap.vht_mcs.rx_mcs_map);
      |                                      ^~
./include/uapi/linux/byteorder/little_endian.h:37:51: note: in definition of macro '__le16_to_cpu'
[... 87 lines cut ...]
/home/shion/R23/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/include/mac80211/net/mac80211.h:6469:6: note: declared here
 6469 | void ieee80211_request_smps(struct ieee80211_vif *vif, unsigned int link_id,
      |      ^~~~~~~~~~~~~~~~~~~~~~
/home/shion/R23/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-bcm27xx_bcm2709/rtw88-usb-2023-02-11-3af004dd/main.c:1758:17: error: too few arguments to function 'ieee80211_request_smps'
 1758 |                 ieee80211_request_smps(vif, IEEE80211_SMPS_OFF);
      |                 ^~~~~~~~~~~~~~~~~~~~~~
In file included from /home/shion/R23/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-bcm27xx_bcm2709/rtw88-usb-2023-02-11-3af004dd/main.h:12,
                 from /home/shion/R23/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-bcm27xx_bcm2709/rtw88-usb-2023-02-11-3af004dd/main.c:7:
/home/shion/R23/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/usr/include/mac80211/net/mac80211.h:6469:6: note: declared here
 6469 | void ieee80211_request_smps(struct ieee80211_vif *vif, unsigned int link_id,
      |      ^~~~~~~~~~~~~~~~~~~~~~
make[5]: *** [scripts/Makefile.build:289: /home/shion/R23/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-bcm27xx_bcm2709/rtw88-usb-2023-02-11-3af004dd/main.o] Error 1
make[4]: *** [Makefile:1905: /home/shion/R23/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-bcm27xx_bcm2709/rtw88-usb-2023-02-11-3af004dd] Error 2
[... cut ...]
time: package/kernel/rtw88-usb/compile#0.96#0.40#1.34
    ERROR: package/kernel/rtw88-usb failed to build.
make[2]: *** [package/Makefile:116: package/kernel/rtw88-usb/compile] Error 1
make[2]: Leaving directory '/home/shion/R23/lede'
make[1]: *** [package/Makefile:110: /home/shion/R23/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/shion/R23/lede'
make: *** [/home/shion/R23/lede/include/toplevel.mk:231：world] 错误 2

```

WYC-2020, 2023-06-30:
明明错误是 rtw88-usb

WYC-2020, 2023-06-30:
[image]
[100-update-linux-version-macro.patch](https://github.com/coolsnowwolf/lede/files/11915141/100-update-linux-version-macro.patch)
按图片位置放

commit 9c0329ad (2023-07-04): rtw88-usb: bump to latest git HEAD

Fixes: #11331
closed 2023-07-04 by aiamadeus