coolsnowwolf/lede #11335: r2s ubuntu22.04 编译失败

tanbx, 2023-06-30:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述
按照readme操作，只选了r2s架构，没有增加任何选项，编译失败。

### (2) 路由器型号和固件版本
5.15.114

### (3) 详细日志
make[4]: Entering directory '/home/tan/lede/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r2s-rk3328/u-boot-2023.04'
  ENVC    include/generated/env.txt
  ENVP    include/generated/env.in
  ENVT    include/generated/environment.h
make -f ./scripts/Makefile.build obj=scripts/dtc
  PYMOD   rebuild
Traceback (most recent call last):
  File "/home/tan/lede/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r2s-rk3328/u-boot-2023.04/scripts/dtc/pylibfdt/setup.py", line 23, in <module>
    from setuptools import setup, Extension
ModuleNotFoundError: No module named 'setuptools'
make[6]: *** [scripts/dtc/pylibfdt/Makefile:33: rebuild] Error 1
make[5]: *** [scripts/Makefile.build:397: scripts/dtc/pylibfdt] Error 2
make[4]: *** [Makefile:2003: scripts_dtc] Error 2
make[4]: Leaving directory '/home/tan/lede/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r2s-rk3328/u-boot-2023.04'
make[3]: *** [Makefile:337: /home/tan/lede/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r2s-rk3328/u-boot-2023.04/.built] Error 2
make[3]: Leaving directory '/home/tan/lede/package/boot/uboot-rockchip'
time: package/boot/uboot-rockchip/nanopi-r2s-rk3328/compile#1.41#0.60#1.91
    ERROR: package/boot/uboot-rockchip failed to build (build variant: nanopi-r2s-rk3328).
make[2]: *** [package/Makefile:116: package/boot/uboot-rockchip/compile] Error 1
make[2]: Leaving directory '/home/tan/lede'
make[1]: *** [package/Makefile:110: /home/tan/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/tan/lede'
make: *** [/home/tan/lede/include/toplevel.mk:231: world] Error 2

eskimokk, 2023-06-30:
R4S， same error

scripts/kconfig/conf  --syncconfig Kconfig
.config:2296:warning: override: reassigning to symbol IDENT_STRING
  UPD     include/config.h
  CFG     u-boot.cfg
  GEN     include/autoconf.mk
  GEN     include/autoconf.mk.dep
  CFG     spl/u-boot.cfg
  GEN     spl/include/autoconf.mk
  CFG     tpl/u-boot.cfg
  GEN     tpl/include/autoconf.mk
  UPD     include/config/uboot.release
  UPD     include/generated/version_autogenerated.h
  UPD     include/generated/timestamp_autogenerated.h
  UPD     include/generated/dt.h
  ENVC    include/generated/env.txt
  ENVP    include/generated/env.in
  ENVT    include/generated/environment.h
  CC      lib/asm-offsets.s
  UPD     include/generated/generic-asm-offsets.h
  CC      arch/arm/lib/asm-offsets.s
  UPD     include/generated/asm-offsets.h
make -f ./scripts/Makefile.build obj=scripts/dtc
  SHIPPED scripts/dtc/pylibfdt/libfdt.i
  PYMOD   rebuild
Traceback (most recent call last):
  File "/home/kk/Codebase/lede.r/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r4s-rk3399/u-boot-2023.04/scripts/dtc/pylibfdt/setup.py", line 23, in <module>
    from setuptools import setup, Extension
ModuleNotFoundError: No module named 'setuptools'
make[6]: *** [scripts/dtc/pylibfdt/Makefile:33: rebuild] Error 1
make[5]: *** [scripts/Makefile.build:397: scripts/dtc/pylibfdt] Error 2
make[4]: *** [Makefile:2003: scripts_dtc] Error 2
make[4]: Leaving directory '/home/kk/Codebase/lede.r/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r4s-rk3399/u-boot-2023.04'
make[3]: *** [Makefile:337: /home/kk/Codebase/lede.r/build_dir/target-aarch64_generic_musl/u-boot-nanopi-r4s-rk3399/u-boot-2023.04/.built] Error 2
make[3]: Leaving directory '/home/kk/Codebase/lede.r/package/boot/uboot-rockchip'
time: package/boot/uboot-rockchip/nanopi-r4s-rk3399/compile#10.66#1.74#11.62
    ERROR: package/boot/uboot-rockchip failed to build (build variant: nanopi-r4s-rk3399).
make[2]: *** [package/Makefile:116: package/boot/uboot-rockchip/compile] Error 1
make[2]: Leaving directory '/home/kk/Codebase/lede.r'
make[1]: *** [package/Makefile:110: /home/kk/Codebase/lede.r/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/kk/Codebase/lede.r'
make: *** [/home/kk/Codebase/lede.r/include/toplevel.mk:231: world] Error 2

QYKing233, 2023-06-30:
sudo apt-get install python3-setuptools

tanbx, 2023-06-30:
安装了python3-setuptools错误消失。有时间再增加写项目进去编译试试。谢谢~

> sudo apt-get install python3-setuptools

closed 2023-06-30 by tanbx