coolsnowwolf/lede #11344: Luci界面实时速率不能显示

Rhuua, 2023-07-03:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：
我在使用基于nft-qos的实时速率显示时发现显示不正常，如下：
[image]
使用浏览器开发者模式发现是String.format没有定义：
[image]
本人对js仅一知半解，不知如何修复？

### (2) 路由器型号和固件版本

A：x86

### (3) 详细日志

A：
<html>
<body>
<div tabindex="-1" class="console-message-wrapper console-error-level" style="box-sizing: border-box; min-width: 0px; min-height: 0px; display: flex; border-top-width: ; border-top-style: ; border-top-color: var(--override-error-border-color); border-bottom: 1px solid transparent; --console-color-black: #000; --console-color-red: rgb(237 78 76); --console-color-green: rgb(1 200 1); --console-color-yellow: rgb(210 192 87); --console-color-blue: rgb(39 116 240); --console-color-magenta: rgb(161 66 244); --console-color-cyan: rgb(18 181 203); --console-color-gray: rgb(207 208 208); --console-color-darkgray: rgb(137 137 137); --console-color-lightred: rgb(242 139 130); --console-color-lightgreen: rgb(161 247 181); --console-color-lightyellow: rgb(221 251 85); --console-color-lightblue: rgb(102 157 246); --console-color-ightmagenta: #f5f; --console-color-lightcyan: rgb(132 240 255); --console-color-white: #fff; background-color: var(--override-console-error-background-color); --console-color-lightmagenta: rgb(214 112 214); color: rgb(255, 255, 255); font-family: consolas, &quot;lucida console&quot;, &quot;courier new&quot;, monospace; font-size: 12px; font-style: normal; font-variant-ligatures: normal; font-variant-caps: normal; font-weight: 400; letter-spacing: normal; orphans: 2; text-align: start; text-indent: 0px; text-transform: none; widows: 2; word-spacing: 0px; -webkit-text-stroke-width: 0px; white-space: normal; text-decoration-thickness: initial; text-decoration-style: initial; text-decoration-color: initial;"><div class="console-message" style="box-sizing: border-box; min-width: 0px; min-height: 17px; clear: right; position: relative; padding: 3px 22px 1px 0px; margin-left: 24px; flex: 1 1 auto; display: flex;"><div class="console-message-stack-trace-toggle" style="box-sizing: border-box; min-width: 0px; min-height: 0px; flex: 1 1 auto; display: flex; flex-direction: row; align-items: flex-start;"><div class="console-message-stack-trace-wrapper" aria-label="rate:85  Uncaught TypeError: String.format is not a function
    at bytes_label (rate:85:17)
    at print_table (rate:93:5)
    at rate:141:8
    at xhr.onreadystatechange (xhr.js?v=git-23.177.50322-72213cf:72:5) 已展开堆叠表" style="box-sizing: border-box; min-width: 0px; min-height: 0px; flex: 1 1 auto; display: flex; flex-direction: column; align-items: stretch;"><div aria-expanded="true" tabindex="-1" style="box-sizing: border-box; min-width: 0px; min-height: 0px; flex: 0 0 auto;"><span class="source-code" style="box-sizing: border-box; min-width: 0px; min-height: 0px; font-family: var(--source-code-font-family); white-space: pre-wrap; font-size: var(--source-code-font-size) !important; line-height: 1.2;"><span class="console-message-anchor" style="box-sizing: border-box; min-width: 0px; min-height: 0px; float: right; text-align: right; max-width: 100%; margin-left: 4px;"><span class="devtools-link" role="link" tabindex="-1" title="http://192.168.123.1/cgi-bin/luci/admin/status/realtime/rate:85" style="box-sizing: border-box; min-width: 0px; min-height: 0px; color: var(--color-text-secondary); text-decoration: underline; outline-offset: 2px; cursor: pointer; word-break: break-all;"><br class="Apple-interchange-newline">rate:85</span> <devtools-icon class="resource-links" title="在 Web 上搜索此消息" style="display: inline-block; white-space: nowrap; position: relative; box-sizing: border-box; min-width: 0px; min-height: 0px; margin-top: -1px; margin-bottom: -2px; cursor: pointer;"><div class="icon-basic" style="width: 16px; height: 16px; display: block; -webkit-mask-image: url(&quot;devtools://devtools/bundled/Images/edge-search-info.svg&quot;); -webkit-mask-position: center center; -webkit-mask-repeat: no-repeat; -webkit-mask-size: 99%; background-color: var(--icon-color, var(--color-link));"></div></devtools-icon> </span><span class="console-message-text" style="box-sizing: border-box; min-width: 0px; min-height: 0px; color: var(--override-error-text-color) !important;">Uncaught <span class="object-value-error source-code" style="box-sizing: border-box; min-width: 0px; min-height: 0px; font-family: var(--source-code-font-family); white-space: pre-wrap; font-size: var(--source-code-font-size) !important; line-height: 1.2;"><span style="box-sizing: border-box; min-width: 0px; min-height: 0px;">TypeError: String.format is not a function
<span style="box-sizing: border-box; min-width: 0px; min-height: 0px;">    at bytes_label (<span class="devtools-link" role="link" tabindex="-1" title="http://192.168.123.1/cgi-bin/luci/admin/status/realtime/rate:85:17" style="box-sizing: border-box; min-width: 0px; min-height: 0px; color: var(--override-console-link-color); text-decoration: underline; outline-offset: 2px; cursor: pointer; word-break: break-all;">rate:85:17</span>)
</span><span style="box-sizing: border-box; min-width: 0px; min-height: 0px;">    at print_table (<span class="devtools-link" role="link" tabindex="-1" title="http://192.168.123.1/cgi-bin/luci/admin/status/realtime/rate:93:5" style="box-sizing: border-box; min-width: 0px; min-height: 0px; color: var(--override-console-link-color); text-decoration: underline; outline-offset: 2px; cursor: pointer; word-break: break-all;">rate:93:5</span>)
</span><span style="box-sizing: border-box; min-width: 0px; min-height: 0px;">    at <span class="devtools-link" role="link" tabindex="-1" title="http://192.168.123.1/cgi-bin/luci/admin/status/realtime/rate:141:8" style="box-sizing: border-box; min-width: 0px; min-height: 0px; color: var(--override-console-link-color); text-decoration: underline; outline-offset: 2px; cursor: pointer; word-break: break-all;">rate:141:8</span>
</span><span style="box-sizing: border-box; min-width: 0px; min-height: 0px;">    at xhr.onreadystatechange (<span class="devtools-link" role="link" tabindex="-1" title="http://192.168.123.1/luci-static/resources/xhr.js?v=git-23.177.50322-72213cf:72:5" style="box-sizing: border-box; min-width: 0px; min-height: 0px; color: var(--override-console-link-color); text-decoration: underline; outline-offset: 2px; cursor: pointer; word-break: break-all;">xhr.js?v=git-23.177.50322-72213cf:72:5</span>)</span></span></span></span></span></div><div class="" role="group" style="box-sizing: border-box; min-width: 0px; min-height: 0px; flex: 0 0 auto;"><span class="monospace" style="display: inline-block; --monospace-font-size: 12px; --monospace-font-family: consolas, lucida console, courier new, monospace; --source-code-font-size: 12px; --source-code-font-family: consolas, lucida console, courier new, monospace; box-sizing: border-box; min-width: 0px; min-height: 0px; font-family: var(--monospace-font-family); font-size: var(--monospace-font-size) !important;">

  | bytes_label | @ | rate:85
-- | -- | -- | --
  | print_table | @ | rate:93
  | (匿名) | @ | rate:141
  | xhr.onreadystatechange | @ | xhr.js?v=git-23.177.50322-72213cf:72
  | XMLHttpRequest.send(异步) |   |  
  | XHR.get | @ | xhr.js?v=git-23.177.50322-72213cf:76
  | XHR._r | @ | xhr.js?v=git-23.177.50322-72213cf:184
  | XHR.run | @ | xhr.js?v=git-23.177.50322-72213cf:231
  | XHR.poll | @ | xhr.js?v=git-23.177.50322-72213cf:199
  | (匿名) | @ | rate:114
  | setTimeout(异步) |   |  
  | (匿名) | @ | rate:102

</span></div></div></div></div></div>
</body>
</html>

WYC-2020, 2023-07-04:
你可真是大聪明，官方的包 跑这里来问，害我对比大半天，l这边没你这些问题，自己跑官方去问 
[image]

Rhuua, 2023-07-04:
> 

那真是奇了怪了，我是前两天新clone了这边的仓库编译出来的，反而是你这边的界面。。确定是默认ui吗

WYC-2020, 2023-07-04:
你确定你克隆的是这个 仓库 https://github.com/coolsnowwolf/luci  你那个明显看那个列名称 就不是 l 这边的 ，我看过官网，那边的这个库的列才叫做实时下载速率

WYC-2020, 2023-07-04:
别想 用这边的lede 搭配官方的luci ，如果非要杂交 那就只能自己解决问题了

Rhuua, 2023-07-04:
> 

[image]
我克隆的是这个lede仓库，按我的理解，luci界面的仓库是看feed的配置吧，确实也是这边的仓库

WYC-2020, 2023-07-04:
额，你选了那个luci 我选的
[image]
咋我这个和你那个差别那么大，我标签都只有两个
[image]

WYC-2020, 2023-07-04:
看来我冤枉你了 对不住兄弟，我再状态哪里看来 也没问题呢,你说的是这里吧 
[image]

Rhuua, 2023-07-04:
[image]
我确认了一下软件包，都是有的。
确实是你后面发的这个界面看不到
[image]
我的插件界面压根没有实时速率显示，只有状态那里有

Rhuua, 2023-07-04:
> 额，你选了那个luci 我选的 [image] 咋我这个和你那个差别那么大，我标签都只有两个 [image]

我用的是bootstrap默认主题，不知你用的哪个主题？我装一下试试有没有改善

WYC-2020, 2023-07-04:
那我有时间看看吧 ，这个显示速率好像是有点问题 不显示

Rhuua, 2023-07-04:
好吧，我换成了argon主题，发现你一开始那个是限速设置界面，不是速率设置界面，又回到原点了。。
不得不说看腻了老主题，换成新主题感觉更像一个产品了。

WYC-2020, 2023-07-04:
稍微修复了一下 
[image]
[rate.zip](https://github.com/coolsnowwolf/lede/files/11945135/rate.zip)
替换掉原来的，这个原因应该是更新这个库的时候没适配现在的luci,我是不知道这个是从哪里扣过来的，修改了几个地方 现在应该没问题了

Rhuua, 2023-07-04:
我把这个文件替换到`/usr/lib/lua/luci/view/nft-qos/`后确实可以实时更新了，不过似乎还是有点问题：
[image]
这速率有点太夸张了
[image]
甚至还有这样的
我确定这不是正常的设备数和速率
而我在f12截取的速率数据是这样的
```
{"nftables":[{"rule":{"chain":"upload","table":"nft-qos-monitor","family":"inet","expr":[{"match":{"right":"192.168.123.202"}},{"counter":{"packets":"2","bytes":"672"}}],"handle":0}},{"rule":{"chain":"upload","table":"nft-qos-monitor","family":"inet","expr":[{"match":{"right":"192.168.123.202"}},{"counter":{"packets":"2","bytes":"672"}}],"handle":0}},{"rule":{"chain":"upload","table":"nft-qos-monitor","family":"inet","expr":[{"match":{"right":"192.168.123.212"}},{"counter":{"packets":"2960","bytes":"204178"}}],"handle":0}},{"rule":{"chain":"upload","table":"nft-qos-monitor","family":"inet","expr":[{"match":{"right":"192.168.123.211"}},{"counter":{"packets":"2322","bytes":"223791"}}],"handle":0}},{"rule":{"chain":"upload","table":"nft-qos-monitor","family":"inet","expr":[{"match":{"right":"192.168.123.211"}},{"counter":{"packets":"2322","bytes":"223791"}}],"handle":0}},{"rule":{"chain":"upload","table":"nft-qos-monitor","family":"inet","expr":[{"match":{"right":"192.168.123.212"}},{"counter":{"packets":"2960","bytes":"204178"}}],"handle":0}},{"rule":{"chain":"upload","table":"nft-qos-monitor","family":"inet","expr":[{"match":{"right":"192.168.123.211"}},{"counter":{"packets":"2322","bytes":"223791"}}],"handle":0}},{"rule":{"chain":"upload","table":"nft-qos-monitor","family":"inet","expr":[{"match":{"right":"192.168.123.200"}},{"counter":{"packets":"3129","bytes":"403479"}}],"handle":0}},{"rule":{"chain":"upload","table":"nft-qos-monitor","family":"inet","expr":[{"match":{"right":"192.168.123.200"}},{"counter":{"packets":"3129","bytes":"403479"}}],"handle":0}},{"rule":{"chain":"upload","table":"nft-qos-monitor","family":"inet","expr":[{"match":{"right":"192.168.123.213"}},{"counter":{"packets":"3060","bytes":"278019"}}],"handle":0}},{"rule":{"chain":"upload","table":"nft-qos-monitor","family":"inet","expr":[{"match":{"right":"192.168.123.213"}},{"counter":{"packets":"3060","bytes":"278019"}}],"handle":0}},{"rule":{"chain":"upload","table":"nft-qos-monitor","family":"inet","expr":[{"match":{"right":"192.168.123.213"}},{"counter":{"packets":"3060","bytes":"278019"}}],"handle":0}},{"rule":{"chain":"upload","table":"nft-qos-monitor","family":"inet","expr":[{"match":{"right":"192.168.123.213"}},{"counter":{"packets":"3060","bytes":"278019"}}],"handle":0}},{"rule":{"chain":"upload","table":"nft-qos-monitor","family":"inet","expr":[{"match":{"right":"192.168.123.213"}},{"counter":{"packets":"3058","bytes":"277935"}}],"handle":0}},{"rule":{"chain":"upload","table":"nft-qos-monitor","family":"inet","expr":[{"match":{"right":"192.168.123.158"}},{"counter":{"packets":"39347","bytes":"4389625"}}],"handle":0}},{"rule":{"chain":"upload","table":"nft-qos-monitor","family":"inet","expr":[{"match":{"right":"192.168.123.158"}},{"counter":{"packets":"38952","bytes":"4292059"}}],"handle":0}},{"rule":{"chain":"upload","table":"nft-qos-monitor","family":"inet","expr":[{"match":{"right":"192.168.123.213"}},{"counter":{"packets":"3018","bytes":"272079"}}],"handle":0}},{"rule":{"chain":"upload","table":"nft-qos-monitor","family":"inet","expr":[{"match":{"right":"192.168.123.200"}},{"counter":{"packets":"3089","bytes":"396740"}}],"handle":0}},{"rule":{"chain":"upload","table":"nft-qos-monitor","family":"inet","expr":[{"match":{"right":"192.168.123.211"}},{"counter":{"packets":"2306","bytes":"221510"}}],"handle":0}},{"rule":{"chain":"upload","table":"nft-qos-monitor","family":"inet","expr":[{"match":{"right":"192.168.123.212"}},{"counter":{"packets":"2951","bytes":"203576"}}],"handle":0}},{"rule":{"chain":"upload","table":"nft-qos-monitor","family":"inet","expr":[{"match":{"right":"192.168.123.202"}},{"counter":{"packets":"2","bytes":"672"}}],"handle":0}},{"rule":{"chain":"upload","table":"nft-qos-monitor","family":"inet","expr":[{"match":{"right":"192.168.123.150"}},{"counter":{"packets":"16801","bytes":"2276964"}}],"handle":0}},{"rule":{"chain":"upload","table":"nft-qos-monitor","family":"inet","expr":[{"match":{"right":"192.168.123.209"}},{"counter":{"packets":"3462","bytes":"373672"}}],"handle":0}},{"rule":{"chain":"upload","table":"nft-qos-monitor","family":"inet","expr":[{"match":{"right":"192.168.123.188"}},{"counter":{"packets":"4186774","bytes":"3576246794"}}],"handle":0}},{"rule":{"chain":"upload","table":"nft-qos-monitor","family":"inet","expr":[{"match":{"right":"192.168.123.138"}},{"counter":{"packets":"11893","bytes":"1840922"}}],"handle":0}},{"rule":{"chain":"upload","table":"nft-qos-monitor","family":"inet","expr":[{"match":{"right":"192.168.123.138"}},{"counter":{"packets":"9432","bytes":"1298422"}}],"handle":0}},{"rule":{"chain":"upload","table":"nft-qos-monitor","family":"inet","expr":[{"match":{"right":"192.168.123.188"}},{"counter":{"packets":"3555260","bytes":"3373783505"}}],"handle":0}},{"rule":{"chain":"upload","table":"nft-qos-monitor","family":"inet","expr":[{"match":{"right":"192.168.123.209"}},{"counter":{"packets":"2013","bytes":"216809"}}],"handle":0}},{"rule":{"chain":"upload","table":"nft-qos-monitor","family":"inet","expr":[{"match":{"right":"192.168.123.150"}},{"counter":{"packets":"12271","bytes":"1551305"}}],"handle":0}},{"rule":{"chain":"upload","table":"nft-qos-monitor","family":"inet","expr":[{"match":{"right":"192.168.123.158"}},{"counter":{"packets":"12542","bytes":"1432714"}}],"handle":0}},{"rule":{"chain":"upload","table":"nft-qos-monitor","family": [... cut ...] e":"nft-qos-monitor","family":"inet","expr":[{"match":{"right":"192.168.123.138"}},{"counter":{"packets":"0","bytes":"0"}}],"handle":0}},{"rule":{"chain":"download","table":"nft-qos-monitor","family":"inet","expr":[{"match":{"right":"192.168.123.168"}},{"counter":{"packets":"36","bytes":"4349"}}],"handle":0}},{"rule":{"chain":"download","table":"nft-qos-monitor","family":"inet","expr":[{"match":{"right":"192.168.123.214"}},{"counter":{"packets":"0","bytes":"0"}}],"handle":0}}]}
```
不知道又有哪里出问题了。。

WYC-2020, 2023-07-04:
那就不清楚了 这玩意 ip都很多重复 的，我理想应该是ip唯一才对，你怀疑不对 可以用nft list chain inet nft-qos-monitor download 或者nft list chain inet nft-qos-monitor upload 查看

closed 2026-09-05 by coolsnowwolf