coolsnowwolf/lede #11346: ZeroTier编译报错

LemoFire, 2023-07-04:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：R5C

### (3) 详细日志

A：
```
make[3]: Entering directory '/home/shion/lede/feeds/packages/net/zerotier'
rm -f /home/shion/lede/build_dir/target-aarch64_generic_musl/ZeroTierOne-1.10.6/.built
touch /home/shion/lede/build_dir/target-aarch64_generic_musl/ZeroTierOne-1.10.6/.built_check
CFLAGS="-Os -pipe -mcpu=generic -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/shion/lede/build_dir/target-aarch64_generic_musl/ZeroTierOne-1.10.6=ZeroTierOne-1.10.6 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -ffunction-sections -fdata-sections  -I/home/shion/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/usr/include -I/home/shion/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/include/fortify -I/home/shion/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/include " CXXFLAGS="-Os -pipe -mcpu=generic -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/shion/lede/build_dir/target-aarch64_generic_musl/ZeroTierOne-1.10.6=ZeroTierOne-1.10.6 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -ffunction-sections -fdata-sections  -I/home/shion/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/usr/include -I/home/shion/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/include/fortify -I/home/shion/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/include " LDFLAGS="-L/home/shion/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/usr/lib -L/home/shion/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/lib -znow -zrelro -Wl,--gc-sections,--as-needed " make  -C /home/shion/lede/build_dir/target-aarch64_generic_musl/ZeroTierOne-1.10.6/. AR="aarch64-openwrt-linux-musl-gcc-ar" AS="aarch64-openwrt-linux-musl-gcc -c -Os -pipe -mcpu=generic -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/shion/lede/build_dir/target-aarch64_generic_musl/ZeroTierOne-1.10.6=ZeroTierOne-1.10.6 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -ffunction-sections -fdata-sections" LD=aarch64-openwrt-linux-musl-ld NM="aarch64-openwrt-linux-musl-gcc-nm" CC="aarch64-openwrt-linux-musl-gcc" GCC="aarch64-openwrt-linux-musl-gcc" CXX="aarch64-openwrt-linux-musl-g++" RANLIB="aarch64-openwrt-linux-musl-gcc-ranlib" STRIP=aarch64-openwrt-linux-musl-strip OBJCOPY=aarch64-openwrt-linux-musl-objcopy OBJDUMP=aarch64-openwrt-linux-musl-objdump SIZE=aarch64-openwrt-linux-musl-size CROSS="aarch64-openwrt-linux-musl-" ARCH="aarch64" ZT_EMBEDDED=1 ZT_SSO_SUPPORTED=0 DEFS="" OSTYPE="Linux"  one;
make[4]: Entering directory '/home/shion/lede/build_dir/target-aarch64_generic_musl/ZeroTierOne-1.10.6'
aarch64-openwrt-linux-musl-g++ -Os -pipe -mcpu=generic -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/shion/lede/build_dir/target-aarch64_generic_musl/ZeroTierOne-1.10.6=ZeroTierOne-1.10.6 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -ffunction-sections -fdata-sections  -I/home/shion/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/usr/include -I/home/shion/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/include/fortify -I/home/shion/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/include  -Wall -Wno-deprecated -std=c++17 -pthread -Izeroidc/target -isystem ext -DNDEBUG -DZT_USE_MINIUPNPC -DZT_USE_SYSTEM_MINIUPNPC -DZT_USE_SYSTEM_NATPMP -DZT_NO_TYPE_PUNNING -DZT_ARCH_ARM_HAS_NEON -march=armv8-a+crypto -mtune=generic -mstrict-align -DZT_BUILD_PLATFORM=1 -DZT_BUILD_ARCHITECTURE=4 -DZT_SOFTWARE_UPDATE_DEFAULT="\"disable\"" -D_MT_ALLOCATOR_H -D_POOL_ALLOCATOR_H -D_EXTPTR_ALLOCATOR_H -D_DEBUG_ALLOCATOR_H -Wl,-z,relro,-z,now -o zerotier-one node/AES.o node/AES_aesni.o node/AES_armcrypto.o node/C25519.o node/Capability.o node/CertificateOfMembership.o node/CertificateOfOwnership.o node/Identity.o node/IncomingPacket.o node/InetAddress.o node/Membership.o node/Multicaster.o node/Network.o node/NetworkConfig.o node/Node.o node/OutboundMulticast.o node/Packet.o node/Path.o node/Peer.o node/Poly1305.o node/Revocation.o node/Salsa20.o node/SelfAwareness.o node/SHA512.o node/Switch.o node/Tag.o node/Topology.o node/Trace.o node/Utils.o node/Bond.o controller/EmbeddedNetworkController.o controller/DBMirrorSet.o controller/DB.o controller/FileDB.o controller/LFDB.o controller/PostgreSQL.o osdep/EthernetTap.o osdep/ManagedRoute.o osdep/Http.o osdep/OSUtils.o service/SoftwareUpdater.o service/OneService.o osdep/LinuxEthernetTap.o osdep/LinuxNetLink.o osdep/PortMapper.o ext/http-parser/http_parser.o one.o -lminiupnpc -lnatpmp
/home/shion/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/lib/gcc/aarch64-openwrt-linux-musl/11.3.0/../../../../aarch64-openwrt-linux-musl/bin/ld: /home/shion/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/lib/gcc/aarch64-openwrt-linux-musl/11.3.0/../../../../aarch64-openwrt-linux-musl/lib/crt1.o: in function `_start_c':
/home/shion/lede/build_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/musl-1.2.3/crt/crt1.c:18: undefined reference to `main'
/home/shion/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/lib/gcc/aarch64-openwrt-linux-musl/11.3.0/../../../../aarch64-openwrt-linux-musl/bin/ld: /home/shion/lede/build_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/musl-1.2.3/crt/crt1.c:18: undefined reference to `main'
/home/shion/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/lib/gcc/aarch64-openwrt-linux-musl/11.3.0/../../../../aarch64-openwrt-linux-musl/bin/ld: osdep/EthernetTap.o: in function `ZeroTier::EthernetTap::newInstance(char const*, char const*, ZeroTier::MAC const&, unsigned int, unsigned int, unsigned long, char const*, void (*)(void*, void*, unsigned long, ZeroTier::MAC const&, ZeroTier::MAC const&, unsigned int, unsigned int, void const*, unsigned int), void*)':
EthernetTap.cpp:(.text._ZN8ZeroTier11EthernetTap11newInstanceEPKcS2_RKNS_3MACEjjmS2_PFvPvS6_mS5_S5_jjPKvjES6_+0x70): undefined reference to `ZeroTier::LinuxEthernetTap::LinuxEthernetTap(char const*, ZeroTier::MAC const&, unsigned int, unsigned int, unsigned long, char const*, void (*)(void*, void*, unsigned long, ZeroTier::MAC const&, ZeroTier::MAC const&, unsigned int, unsigned int, void const*, unsigned int), void*)'
collect2: error: ld returned 1 exit status
make[4]: *** [make-linux.mk:360: zerotier-one] Error 1
make[4]: Leaving directory '/home/shion/lede/build_dir/target-aarch64_generic_musl/ZeroTierOne-1.10.6'
make[3]: *** [Makefile:84: /home/shion/lede/build_dir/target-aarch64_generic_musl/ZeroTierOne-1.10.6/.built] Error 2
make[3]: Leaving directory '/home/shion/lede/feeds/packages/net/zerotier'
time: package/feeds/packages/zerotier/compile#0.20#0.29#0.47
    ERROR: package/feeds/packages/zerotier failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/zerotier/compile] Error 1
make[2]: Leaving directory '/home/shion/lede'
make[1]: *** [package/Makefile:110: /home/shion/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/shion/lede'
make: *** [/home/shion/lede/include/toplevel.mk:231：world] 错误 2

```

WYC-2020, 2023-07-04:
那就不清楚 我这边一切正常，刚编译了试了下，你可以试试  make package/feeds/luci/luci-app-zerotier/{clean,compile} V=s,最好吧musl 再编译一次 然后再执行上面这个 你那个报错连main入口都没找打 不知道怎么玩出来的

LemoFire, 2023-07-04:
> 那就不清楚 我这边一切正常，刚编译了试了下，你可以试试 make package/feeds/luci/luci-app-zerotier/{clean,compile} V=s,最好吧musl 再编译一次 然后再执行上面这个 你那个报错连main入口都没找打 不知道怎么玩出来的

我单独执行你这条命令没有报错，但是一但勾选后make -j1 V=s就会报错

```
make[3]: Entering directory '/home/shion/lede/feeds/luci/applications/luci-app-webdav'
mkdir -p /home/shion/lede/staging_dir/target-aarch64_generic_musl/root-rockchip/stamp
SHELL= flock /home/shion/lede/tmp/.root-copy.flock -c 'cp -fpR /home/shion/lede/build_dir/target-aarch64_generic_musl/luci-app-webdav/.pkgdir/luci-app-webdav/. /home/shion/lede/staging_dir/target-aarch64_generic_musl/root-rockchip/'
touch /home/shion/lede/staging_dir/target-aarch64_generic_musl/root-rockchip/stamp/.luci-app-webdav_installed
echo "luci-app-webdav" >> /home/shion/lede/staging_dir/target-aarch64_generic_musl/pkginfo/luci-app-webdav.default.install
make[3]: Leaving directory '/home/shion/lede/feeds/luci/applications/luci-app-webdav'
time: package/feeds/luci/luci-app-webdav/compile#0.13#0.07#0.20
make[3]: Entering directory '/home/shion/lede/feeds/packages/net/etherwake'
mkdir -p /home/shion/lede/staging_dir/target-aarch64_generic_musl/root-rockchip/stamp
SHELL= flock /home/shion/lede/tmp/.root-copy.flock -c 'cp -fpR /home/shion/lede/build_dir/target-aarch64_generic_musl/etherwake-1.09.orig/.pkgdir/etherwake/. /home/shion/lede/staging_dir/target-aarch64_generic_musl/root-rockchip/'
touch /home/shion/lede/staging_dir/target-aarch64_generic_musl/root-rockchip/stamp/.etherwake_installed
echo "etherwake" >> /home/shion/lede/staging_dir/target-aarch64_generic_musl/pkginfo/etherwake.default.install
make[3]: Leaving directory '/home/shion/lede/feeds/packages/net/etherwake'
time: package/feeds/packages/etherwake/compile#0.13#0.07#0.20
make[3]: Entering directory '/home/shion/lede/package/lean/wol'
mkdir -p /home/shion/lede/dl
SHELL= flock /home/shion/lede/tmp/.wol-0.7.1.tar.gz.flock -c '  	/home/shion/lede/scripts/download.pl "/home/shion/lede/dl" "wol-0.7.1.tar.gz" "c2fa9d7e771134ac8c89d56b8197d4ca" "" "@SF/ahh"    '
mkdir -p /home/shion/lede/staging_dir/target-aarch64_generic_musl/root-rockchip/stamp
SHELL= flock /home/shion/lede/tmp/.root-copy.flock -c 'cp -fpR /home/shion/lede/build_dir/target-aarch64_generic_musl/wol-0.7.1/.pkgdir/wol/. /home/shion/lede/staging_dir/target-aarch64_generic_musl/root-rockchip/'
touch /home/shion/lede/staging_dir/target-aarch64_generic_musl/root-rockchip/stamp/.wol_installed
echo "wol" >> /home/shion/lede/staging_dir/target-aarch64_generic_musl/pkginfo/wol.default.install
make[3]: Leaving directory '/home/shion/lede/package/lean/wol'
time: package/lean/wol/compile#0.19#0.12#0.31
make[3]: Entering directory '/home/shion/lede/feeds/luci/applications/luci-app-wol'
mkdir -p /home/shion/lede/staging_dir/target-aarch64_generic_musl/root-rockchip/stamp
SHELL= flock /home/shion/lede/tmp/.root-copy.flock -c 'cp -fpR /home/shion/lede/build_dir/target-aarch64_generic_musl/luci-app-wol/.pkgdir/luci-app-wol/. /home/shion/lede/staging_dir/target-aarch64_generic_musl/root-rockchip/'
touch /home/shion/lede/staging_dir/target-aarch64_generic_musl/root-rockchip/stamp/.luci-app-wol_installed
echo "luci-app-wol" >> /home/shion/lede/staging_dir/target-aarch64_generic_musl/pkginfo/luci-app-wol.default.install
mkdir -p /home/shion/lede/staging_dir/target-aarch64_generic_musl/root-rockchip/stamp
SHELL= flock /home/shion/lede/tmp/.root-copy.flock -c 'cp -fpR /home/shion/lede/build_dir/target-aarch64_generic_musl/luci-app-wol/.pkgdir/luci-i18n-wol-zh-cn/. /home/shion/lede/staging_dir/target-aarch64_generic_musl/root-rockchip/'
touch /home/shion/lede/staging_dir/target-aarch64_generic_musl/root-rockchip/stamp/.luci-i18n-wol-zh-cn_installed
echo "luci-i18n-wol-zh-cn" >> /home/shion/lede/staging_dir/target-aarch64_generic_musl/pkginfo/luci-app-wol.default.install
make[3]: Leaving directory '/home/shion/lede/feeds/luci/applications/luci-app-wol'
time: package/feeds/luci/luci-app-wol/compile#0.88#0.95#1.72
make[3]: Entering directory '/home/shion/lede/feeds/packages/libs/libnatpmp'
mkdir -p /home/shion/lede/staging_dir/target-aarch64_generic_musl/root-rockchip/stamp
echo '1' | cmp -s - /home/shion/lede/staging_dir/target-aarch64_generic_musl/pkginfo/libnatpmp.version || { echo '1' > /home/shion/lede/staging_dir/target-aarch64_generic_musl/pkginfo/libnatpmp.version;  } 
SHELL= flock /home/shion/lede/tmp/.root-copy.flock -c 'cp -fpR /home/shion/lede/build_dir/target-aarch64_generic_musl/libnatpmp-20150609/.pkgdir/libnatpmp/. /home/shion/lede/staging_dir/target-aarch64_generic_musl/root-rockchip/'
touch /home/shion/lede/staging_dir/target-aarch64_generic_musl/root-rockchip/stamp/.libnatpmp_installed
echo "libnatpmp" >> /home/shion/lede/staging_dir/target-aarch64_generic_musl/pkginfo/libnatpmp.default.install
make[3]: Leaving directory '/home/shion/lede/feeds/packages/libs/libnatpmp'
time: package/feeds/packages/libnatpmp/compile#0.13#0.07#0.21
make[3]: Entering directory '/home/shion/lede/feeds/packages/net/miniupnpc'
mkdir -p /home/shion/lede/staging_dir/target-aarch64_generic_musl/root-rockchip/stamp
SHELL= flock /home/shion/lede/tmp/.root-copy.flock -c 'cp -fpR /home/shion/lede/build_dir/target-aarch64_generic_musl/miniupnpc-2.2.3/.pkgdir/libminiupnpc/. /home/shion/lede/staging_dir/target-aarch64_generic_musl/root-rockchip/'
touch /home/shion/lede/staging_dir/target-aarch64_generic_musl/root-rockchip/stamp/.libminiupnpc_installed
[... cut ...]
make[3]: Leaving directory '/home/shion/lede/feeds/packages/net/zerotier'
time: package/feeds/packages/zerotier/compile#0.21#0.25#0.44
    ERROR: package/feeds/packages/zerotier failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/zerotier/compile] Error 1
make[2]: Leaving directory '/home/shion/lede'
make[1]: *** [package/Makefile:110: /home/shion/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/shion/lede'
make: *** [/home/shion/lede/include/toplevel.mk:231：world] 错误 2
```

LemoFire, 2023-07-04:
> 那就不清楚 我这边一切正常，刚编译了试了下，你可以试试 make package/feeds/luci/luci-app-zerotier/{clean,compile} V=s,最好吧musl 再编译一次 然后再执行上面这个 你那个报错连main入口都没找打 不知道怎么玩出来的

在我执行完你上面那条命令成功后，执行make -j1 V=s后，再次执行你那条命令，也报错了
```
shion@shion-ubuntu:~/lede$ make package/feeds/luci/luci-app-zerotier/{clean,compile} V=s
make[2]: Entering directory '/home/shion/lede/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/home/shion/lede/scripts/config'
make[1]: Entering directory '/home/shion/lede'
make[2]: Entering directory '/home/shion/lede/feeds/luci/applications/luci-app-zerotier'
rm -rf /home/shion/lede/build_dir/target-aarch64_generic_musl/luci-app-zerotier
rm -f /home/shion/lede/staging_dir/target-aarch64_generic_musl/stamp/.luci-app-zerotier_installed
rm -f /home/shion/lede/staging_dir/target-aarch64_generic_musl/packages/luci-app-zerotier.list
make[2]: Leaving directory '/home/shion/lede/feeds/luci/applications/luci-app-zerotier'
time: package/feeds/luci/luci-app-zerotier/clean#0.09#0.04#0.14
make[1]: Leaving directory '/home/shion/lede'
make[2]: Entering directory '/home/shion/lede/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/home/shion/lede/scripts/config'
make[1]: Entering directory '/home/shion/lede'
make[2]: Entering directory '/home/shion/lede/feeds/luci/modules/luci-base'
. /home/shion/lede/include/shell.sh; bzcat /home/shion/lede/dl/LuaSrcDiet-0.12.1.tar.bz2 | tar -C /home/shion/lede/build_dir/hostpkg/LuaSrcDiet-0.12.1/.. -xf -
[ ! -d ./src/ ] || cp -fpR ./src/* /home/shion/lede/build_dir/hostpkg/LuaSrcDiet-0.12.1
touch /home/shion/lede/build_dir/hostpkg/LuaSrcDiet-0.12.1/.preparedeb5e1e1922bb5f59a7b7778aefe934e2_6664517399ebbbc92a37c5bb081b5c53
touch /home/shion/lede/build_dir/hostpkg/LuaSrcDiet-0.12.1/.configured
make -C src/ clean po2lmo
make[3]: Entering directory '/home/shion/lede/feeds/luci/modules/luci-base/src'
rm -f po2lmo parser.so version.lua *.o
cc    -c -o po2lmo.o po2lmo.c
cc    -c -o template_lmo.o template_lmo.c
cc  -o po2lmo po2lmo.o template_lmo.o
make[3]: Leaving directory '/home/shion/lede/feeds/luci/modules/luci-base/src'
make -C /home/shion/lede/build_dir/hostpkg/LuaSrcDiet-0.12.1 bin/LuaSrcDiet.lua
make[3]: Entering directory '/home/shion/lede/build_dir/hostpkg/LuaSrcDiet-0.12.1'
make[3]: Nothing to be done for 'bin/LuaSrcDiet.lua'.
make[3]: Leaving directory '/home/shion/lede/build_dir/hostpkg/LuaSrcDiet-0.12.1'
touch /home/shion/lede/build_dir/hostpkg/LuaSrcDiet-0.12.1/.built
install -d -m0755 /home/shion/lede/staging_dir/hostpkg/bin
install -m0755 src/po2lmo /home/shion/lede/staging_dir/hostpkg/bin/po2lmo
install -m0755 /home/shion/lede/build_dir/hostpkg/LuaSrcDiet-0.12.1/bin/LuaSrcDiet.lua /home/shion/lede/staging_dir/hostpkg/bin/LuaSrcDiet
mkdir -p /home/shion/lede/staging_dir/hostpkg/stamp
touch /home/shion/lede/build_dir/hostpkg/LuaSrcDiet-0.12.1/.built
touch /home/shion/lede/staging_dir/hostpkg/stamp/.luci-base_installed
make[2]: Leaving directory '/home/shion/lede/feeds/luci/modules/luci-base'
time: package/feeds/luci/luci-base/host-compile#1.02#1.12#2.00
make[2]: Entering directory '/home/shion/lede/package/libs/toolchain'
echo "libc" >> /home/shion/lede/staging_dir/target-aarch64_generic_musl/pkginfo/toolchain.default.install
echo "libgcc" >> /home/shion/lede/staging_dir/target-aarch64_generic_musl/pkginfo/toolchain.default.install
echo "libatomic" >> /home/shion/lede/staging_dir/target-aarch64_generic_musl/pkginfo/toolchain.default.install
echo "libstdcpp" >> /home/shion/lede/staging_dir/target-aarch64_generic_musl/pkginfo/toolchain.default.install
echo "libpthread" >> /home/shion/lede/staging_dir/target-aarch64_generic_musl/pkginfo/toolchain.default.install
echo "librt" >> /home/shion/lede/staging_dir/target-aarch64_generic_musl/pkginfo/toolchain.default.install
make[2]: Leaving directory '/home/shion/lede/package/libs/toolchain'
time: package/libs/toolchain/compile#0.12#0.07#0.18
make[2]: Entering directory '/home/shion/lede/feeds/packages/libs/libnatpmp'
echo "libnatpmp" >> /home/shion/lede/staging_dir/target-aarch64_generic_musl/pkginfo/libnatpmp.default.install
make[2]: Leaving directory '/home/shion/lede/feeds/packages/libs/libnatpmp'
time: package/feeds/packages/libnatpmp/compile#0.12#0.06#0.16
make[2]: Entering directory '/home/shion/lede/feeds/packages/net/miniupnpc'
echo "libminiupnpc" >> /home/shion/lede/staging_dir/target-aarch64_generic_musl/pkginfo/miniupnpc.default.install
make[2]: Leaving directory '/home/shion/lede/feeds/packages/net/miniupnpc'
time: package/feeds/packages/miniupnpc/compile#0.10#0.09#0.17
make[2]: Entering directory '/home/shion/lede/package/firmware/linux-firmware'
echo "mt7922bt-firmware" >> /home/shion/lede/staging_dir/target-aarch64_generic_musl/pkginfo/linux-firmware.default.install
echo "r8152-firmware" >> /home/shion/lede/staging_dir/target-aarch64_generic_musl/pkginfo/linux-firmware.default.install
make[2]: Leaving directory '/home/shion/lede/package/firmware/linux-firmware'
time: package/firmware/linux-firmware/compile#0.94#0.82#1.78
make[2]: Entering directory '/home/shion/lede/package/kernel/linux'
echo "kernel" >> /home/shion/lede/staging_dir/target-aarch64_generic_musl/pkginfo/linux.default.install
echo "kmod-dax" >> /home/shion/lede/staging_dir/target-aarch64_generic_musl/pkginfo/linux.default.install
[... 170 lines cut ...]
echo "ip-full" >> /home/shion/lede/staging_dir/target-aarch64_generic_musl/pkginfo/iproute2.ipfull.install
[... cut ...]
make[2]: *** [Makefile:84: /home/shion/lede/build_dir/target-aarch64_generic_musl/ZeroTierOne-1.10.6/.built] Error 2
make[2]: Leaving directory '/home/shion/lede/feeds/packages/net/zerotier'
time: package/feeds/packages/zerotier/compile#0.27#0.14#0.38
    ERROR: package/feeds/packages/zerotier failed to build.
make[1]: *** [package/Makefile:116: package/feeds/packages/zerotier/compile] Error 1
make[1]: Leaving directory '/home/shion/lede'
make: *** [/home/shion/lede/include/toplevel.mk:231：package/feeds/luci/luci-app-zerotier/compile] 错误 2
```

binge8, 2023-07-04:
一切正常，都删了全新环境重新编译试试吧