coolsnowwolf/lede #11347: X86使用6.1.35内核，mt7921就用不了了

ftebox, 2023-07-04:
反馈bug/问题模板，提建议请删除

## 1. X86使用6.1.35内核，mt7921就用不了了

Q：是否搜索了issue (使用 "x" 选择)
* [ x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：使用最新源码编译x86固件  编译完安装发现无线用不了了 报如下图的错误
[image]

### (2) 路由器型号和固件版本

A：X86_64  R86S

### (3) 详细日志

A：```log
Tue Jul  4 13:06:55 2023 kern.info kernel: [   13.435755] mt7921e 0000:03:00.0: enabling device (0000 -> 0002)
Tue Jul  4 13:06:55 2023 kern.info kernel: [   13.461839] mt7921e 0000:03:00.0: ASIC revision: 79610010
Tue Jul  4 13:06:55 2023 kern.warn kernel: [   13.469994] mt7921e: probe of 0000:03:00.0 failed with error -110
```

xiangfeidexiaohuo, 2023-07-04:
换一个可用的mt76 就成

ftebox, 2023-07-04:
> 换一个可用的mt76就成

7921和76有啥关系

xiangfeidexiaohuo, 2023-07-04:
> 

/package/kernel/mt76 ，有6.1能用的，替换下不就成了

WYC-2020, 2023-07-05:
这玩意probe失败 感觉像是什么依赖吧 ，没研究过

xiangfeidexiaohuo, 2023-07-07:
用这个mt76    7921 6.1内核能编译 https://github.com/immortalwrt/immortalwrt/tree/master/package/kernel/mt76
或者官方openwrt的也成

ftebox, 2023-07-07:
破案了 没重启，重启就好了

closed 2023-07-07 by ftebox