coolsnowwolf/lede #11349: SFTP和SSH问题

wuzhuang123, 2023-07-04:
节前SSH和SFTP还回光返照能用了几天，现在就又不行了，远程机器只剩个web能用。烦劳 @WYC-2020 出手弄一下吧

binge8, 2023-07-04:
https://github.com/coolsnowwolf/lede/issues/11204#issue-1708405001
这个问题已经提了，里面有解决方法。

wuzhuang123, 2023-07-04:
> [#11204 (comment)](https://github.com/coolsnowwolf/lede/issues/11204#issue-1708405001) 这个问题已经提了，里面有解决方法。

当然，那也是我提的。现在是替换zip后重新编译，不保留配置升级，开机后修改主机名修改默认密码再重启就又凉了。

wuzhuang123, 2023-07-04:
破案了，/etc/ssh/sshd_config文件中默认看到都是#PermitRootLogin prohibit-password以为被注释掉了就默认是yes(网上这么说的)一直没往这怀疑，后来手动改成PermitRootLogin yes就可以了。

binge8, 2023-07-04:
> 破案了，/etc/ssh/sshd_config文件中默认看到都是#PermitRootLogin prohibit-password以为被注释掉了就默认是yes(网上这么说的)一直没往这怀疑，后来手动改成PermitRootLogin yes就可以了。

所以，是不是可以修复了？

wuzhuang123, 2023-07-04:
> > 破案了，/etc/ssh/sshd_config文件中默认看到都是#PermitRootLogin prohibit-password以为被注释掉了就默认是yes(网上这么说的)一直没往这怀疑，后来手动改成PermitRootLogin yes就可以了。
> 
> 傻孩子，我一直用就没问题，所以下次在遇到多自己排查排查

我办公室的一台极路由4从没改过配置，这回升级直接就不可访问了，上次能用之后我也看了配置文件和这次一样也是注释掉的

WYC-2020, 2023-07-05:
> > > 破案了，/etc/ssh/sshd_config文件中默认看到都是#PermitRootLogin prohibit-password以为被注释掉了就默认是yes(网上这么说的)一直没往这怀疑，后来手动改成PermitRootLogin yes就可以了。
> > 
> > 
> > 傻孩子，我一直用就没问题，所以下次在遇到多自己排查排查
> 
> 我办公室的一台极路由4从没改过配置，这回升级直接就不可访问了，上次能用之后我也看了配置文件和这次一样也是注释掉的

这个不开 是不允许root登录的，要修复这个 自己改下默认配置啊 编译后就直接打开，我是吧这个集成到了luci界面上 所以不存在你们说的这个问题，并且你们要把启动的服务设置成开机自启动，别重启后 不启动了又说问题，这个我再另外一个里面已经说过了

binge8, 2023-07-05:
> 这个不开 是不允许root登录的，要修复这个 自己改下默认配置啊 编译后就直接打开，我是吧这个集成到了luci界面上 所以不存在你们说的这个问题，并且你们要把启动的服务设置成开机自启动，别重启后 不启动了又说问题，这个我再另外一个里面已经说过了

我又想学学怎么集成到luci

ChenSino, 2024-10-10:
> 破案了，/etc/ssh/sshd_config文件中默认看到都是#PermitRootLogin prohibit-password以为被注释掉了就默认是yes(网上这么说的)一直没往这怀疑，后来手动改成PermitRootLogin yes就可以了。

终于搞好了，跪谢大佬

coolsnowwolf, 2026-09-05:
评论已给出明确根因和解决方法：dropbear/OpenSSH 的 root 登录策略由 `/etc/ssh/sshd_config` 中的 `PermitRootLogin` 控制，默认策略并非允许密码登录；将其明确设为 `PermitRootLogin yes`（并确保 SSH 服务启用）即可恢复 SSH/SFTP。后续评论也确认已解决。该 issue 不需要 LEDE 源码修改，关闭为 completed。

closed 2026-09-05 by coolsnowwolf