coolsnowwolf/lede #11350: 怎么menuconfig里不给R4S这类设备了？

Travel2Here, 2023-07-04:
设备类型没有了~~啥情况？

[image]

Travel2Here, 2023-07-04:
> [image]
> 
> 

奇怪 我下午pull了一下就这样，缓存也清了…依然没有

Travel2Here, 2023-07-05:
没人解吗?
[image]

nuo12138, 2023-07-05:
> 没人解吗?

[image]
[image]
怎么可能没有，我3号才编译了固件，这个是新拉的代码

nuo12138, 2023-07-05:
找不到的话建议删了重新啦源码编译

WYC-2020, 2023-07-05:
是看不懂别人的答复么，都给你贴图说了 有，还在这里一直问，都不检查是否是自己的问题

Travel2Here, 2023-07-05:
> 找不到的话建议删了重新啦源码编译

神奇 删了重拉正常咯 ，可我啥也没干只是更新…

closed 2023-07-05 by Travel2Here