coolsnowwolf/lede #11351: fileassistant这个插件，编译后，不显示文件列表，无法上传

anjue39, 2023-07-04:
fileassistant这个插件，编译后，不显示文件列表，无法上传，内核为k3   5.4，k3每次升级内核都会有问题，每次都降内核，降了之后还是有问题

WYC-2020, 2023-07-04:
如果不能列举 就看下 
[image]
 它是如何枚举文件的 自己抠出来手工执行，看下是不是缺少什么指令，比如egrep

anjue39, 2023-07-04:
感谢

发自我的iPhone

 
明天抽空给你瞧瞧

WYC-2020, 2023-07-05:
无能为力我这边一切正常，
[image]
上传，删除 ，重命名，我都测试 了 一切正常

anjue39, 2023-07-05:
eb7eab0
已修复
你的这个插件是在系统下还是在服务菜单？
索性问一下，你用的lede哪个内核？5.15内核试过可以用吗？

发自我的iPhone

WYC-2020, 2023-07-05:
> eb7eab0 已修复 你的这个插件是在系统下还是在服务菜单？ 索性问一下，你用的lede哪个内核？5.15内核试过可以用吗？ 发自我的iPhone
> […](#)

你不是说5.4有问题嘛 自然是5.4了，再系统下，5.15应该也没问题没试

WYC-2020, 2023-07-05:
> > eb7eab0 已修复 你的这个插件是在系统下还是在服务菜单？ 索性问一下，你用的lede哪个内核？5.15内核试过可以用吗？ 发自我的iPhone
> > […](#)
> 
> 你不是说5.4有问题嘛 自然是5.4了，再系统下，5.15应该也没问题没试

5.15 也没问题，不过我都是编译x86 来测试的

closed 2026-09-05 by coolsnowwolf