coolsnowwolf/lede #11354: Linksys WRT32X 日志中网口频繁重启

alexyw, 2023-07-04:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：暂无影响，可能我只是单纯的上网

### (2) 路由器型号和固件版本

A：WRT32X，5月7日编译最新版本

### (3) 详细日志

A：部分日志如下
Wed Jul  5 00:05:56 2023 kern.info kernel: [10469.475941] br-lan: port 2(lan2) entered disabled state
Wed Jul  5 00:05:56 2023 daemon.notice netifd: Network device 'lan2' link is down
Wed Jul  5 00:05:58 2023 kern.info kernel: [10471.563182] mv88e6085 f1072004.mdio-mii:00 lan2: Link is Up - 10Mbps/Full - flow control off
Wed Jul  5 00:05:58 2023 kern.info kernel: [10471.571682] br-lan: port 2(lan2) entered blocking state
Wed Jul  5 00:05:58 2023 kern.info kernel: [10471.576940] br-lan: port 2(lan2) entered forwarding state
Wed Jul  5 00:05:58 2023 daemon.notice netifd: Network device 'lan2' link is up
Wed Jul  5 00:06:00 2023 kern.info kernel: [10473.101288] mv88e6085 f1072004.mdio-mii:00 lan2: Link is Down
Wed Jul  5 00:06:00 2023 kern.info kernel: [10473.107089] br-lan: port 2(lan2) entered disabled state
Wed Jul  5 00:06:00 2023 daemon.notice netifd: Network device 'lan2' link is down
Wed Jul  5 00:06:05 2023 kern.info kernel: [10478.273371] mv88e6085 f1072004.mdio-mii:00 lan2: Link is Up - 1Gbps/Full - flow control rx/tx
Wed Jul  5 00:06:05 2023 kern.info kernel: [10478.281988] br-lan: port 2(lan2) entered blocking state
Wed Jul  5 00:06:05 2023 kern.info kernel: [10478.287243] br-lan: port 2(lan2) entered forwarding state
Wed Jul  5 00:06:05 2023 daemon.notice netifd: Network device 'lan2' link is up
Wed Jul  5 00:06:07 2023 kern.info kernel: [10480.053739] mv88e6085 f1072004.mdio-mii:00 lan2: Link is Down
Wed Jul  5 00:06:07 2023 kern.info kernel: [10480.059548] br-lan: port 2(lan2) entered disabled state
Wed Jul  5 00:06:07 2023 daemon.notice netifd: Network device 'lan2' link is down
Wed Jul  5 00:06:10 2023 kern.info kernel: [10482.916989] mv88e6085 f1072004.mdio-mii:00 lan2: Link is Up - 1Gbps/Full - flow control rx/tx
Wed Jul  5 00:06:10 2023 kern.info kernel: [10482.925606] br-lan: port 2(lan2) entered blocking state
Wed Jul  5 00:06:10 2023 kern.info kernel: [10482.930856] br-lan: port 2(lan2) entered forwarding state
Wed Jul  5 00:06:10 2023 daemon.notice netifd: Network device 'lan2' link is up

此外，路由器我设置了vlan，以单线复用iptv，netconfig配置文件如下：
config device
	option name 'br-lan'
	option type 'bridge'
	list ports 'lan1'
	list ports 'lan2'
	list ports 'lan3'
	list ports 'lan4'
	list ports 'wan'

config bridge-vlan
	option device 'br-lan'
	option vlan '1'
	list ports 'lan1'
	list ports 'lan2'
	list ports 'lan3'

config bridge-vlan
	option device 'br-lan'
	option vlan '2'
	list ports 'wan'

config bridge-vlan
	option device 'br-lan'
	option vlan '45'
	list ports 'lan4'
	list ports 'wan:t'

config interface 'lan'
	option proto 'static'
	option device 'br-lan.1'
	option ifname 'br-lan.1'
	option ipaddr '192.168.1.1'
	option netmask '255.255.255.0'
	option ip6assign '60'

初步判断可能是交换机在5.10内核故障，参见https://github.com/openwrt/openwrt/issues/11077
如果是这样的话，也希望各位大神帮忙修复，毕竟这个固件太好用了，非常感谢

WYC-2020, 2023-07-05:
看了大半天也没看出 一个补丁，看到的就是切换内核到5.15哈哈

alexyw, 2023-07-05:
> 看了大半天也没看出 一个补丁，看到的就是切换内核到5.15哈哈

我看了下官方论坛，也有可能不是交换机的问题，因为2021年就有人反映了，https://forum.openwrt.org/t/internet-disconnections/113018，但解决方法也是指导降级到21.02或19.07，所以又不好排除是不是这个问题。

这个交换机问题看起来说大不大说小不小，但非常恶心，openwrt官方其中一个开发者直接在22.03版本代码中关掉了对使用这个芯片的路由器支持，说是bug很难，非常离谱！！！

现在只寄希望于咱们本土大神来解决bug了，毕竟老外这么多年吃老本也越来越衰了；要么就只能等大神帮我们把内核升到5.15，一劳永逸，现在5.10用起来安全漏洞也不少

alexyw, 2023-07-05:
> 同款路由器，用了好几年该源码库，每隔1-2个月编译一次固件更新，前几天刚编译了最新源码，查看了这几天的日志，没有发现该问题 默认5.10内核，没有启用vlan相关设置，关闭了无线，pppoe拨号

谢谢你帮我大概率确定是vlan导致的问题，就看啥时候升级内核版本了

WYC-2020, 2023-07-06:
> > 同款路由器，用了好几年该源码库，每隔1-2个月编译一次固件更新，前几天刚编译了最新源码，查看了这几天的日志，没有发现该问题 默认5.10内核，没有启用vlan相关设置，关闭了无线，pppoe拨号
> 
> 谢谢你帮我大概率确定是vlan导致的问题，就看啥时候升级内核版本了
这个你可以自己切啊 ，或者直接编译test内核就是5.15了，然后再试下是否正常

alexyw, 2023-07-06:
> > > 同款路由器，用了好几年该源码库，每隔1-2个月编译一次固件更新，前几天刚编译了最新源码，查看了这几天的日志，没有发现该问题 默认5.10内核，没有启用vlan相关设置，关闭了无线，pppoe拨号
> > 
> > 
> > 谢谢你帮我大概率确定是vlan导致的问题，就看啥时候升级内核版本了
> > 这个你可以自己切啊 ，或者直接编译test内核就是5.15了，然后再试下是否正常

那只怕是前门驱虎后门进狼了

alexyw, 2023-07-08:
又查了一下，海外的大神说要5.18的内核才能解决，国内的大神们能提早搞定不

closed 2026-09-05 by coolsnowwolf