coolsnowwolf/lede #11358: R5S出现未知断流，主路由查看接口变黄，能ping通主路由

Amoystyle, 2023-07-06:
# R5S出现未知断流，主路由查看接口变黄，能ping通主路由

## 拔插换了主路由端口恢复正常

### (2) 路由器型号和固件版本
```
FriendlyElec NanoPi R5S (CpuMark : 28997.924012 Scores)

OpenWrt R22.7.19 / LuCI openwrt-18.06-k5.4 branch (git-22.259.47290-a14e49c)
```

### (3) 详细日志
```
2023/7/6 23:07 OpenWrt-R5S - 系统日志 - LuCI
192.168.100.1/cgi-bin/luci/admin/status/syslog 1/4
Sat Jul 1 21:17:06 2023 daemon.notice netifd: bridge 'br-lan' link is up
Sat Jul 1 21:17:06 2023 daemon.notice netifd: Interface 'lan' has link connectivity
Sun Jul 2 00:09:59 2023 kern.info kernel: [1863158.370744] r8125: eth2: link down
Sun Jul 2 00:09:59 2023 daemon.notice netifd: Network device 'eth2' link is down
Sun Jul 2 00:09:59 2023 kern.info kernel: [1863158.383548] br-lan: port 2(eth2) entered disabled state
Sun Jul 2 00:10:00 2023 daemon.notice netifd: bridge 'br-lan' link is down
Sun Jul 2 00:10:00 2023 daemon.notice netifd: Interface 'lan' has link connectivity loss
Sun Jul 2 08:13:31 2023 daemon.notice netifd: wan (30701): udhcpc: sending renew to server 192.168.1.1
Sun Jul 2 08:13:31 2023 daemon.notice netifd: wan (30701): udhcpc: lease of 192.168.1.149 obtained from 192.168.1.1, lease time 86400
Sun Jul 2 09:21:06 2023 kern.info kernel: [1896225.766358] r8125: eth2: link up
Sun Jul 2 09:21:06 2023 kern.info kernel: [1896225.766749] br-lan: port 2(eth2) entered blocking state
Sun Jul 2 09:21:06 2023 kern.info kernel: [1896225.767525] br-lan: port 2(eth2) entered forwarding state
Sun Jul 2 09:21:06 2023 daemon.notice netifd: Network device 'eth2' link is up
Sun Jul 2 09:21:06 2023 daemon.notice netifd: bridge 'br-lan' link is up
Sun Jul 2 09:21:06 2023 daemon.notice netifd: Interface 'lan' has link connectivity
Sun Jul 2 09:21:21 2023 kern.info kernel: [1896240.690598] r8125: eth2: link down
Sun Jul 2 09:21:21 2023 daemon.notice netifd: Network device 'eth2' link is down
Sun Jul 2 09:21:21 2023 kern.info kernel: [1896240.703321] br-lan: port 2(eth2) entered disabled state
Sun Jul 2 09:21:22 2023 daemon.notice netifd: bridge 'br-lan' link is down
Sun Jul 2 09:21:22 2023 daemon.notice netifd: Interface 'lan' has link connectivity loss
Sun Jul 2 09:21:24 2023 kern.info kernel: [1896244.034870] r8125: eth2: link up
Sun Jul 2 09:21:24 2023 kern.info kernel: [1896244.035260] br-lan: port 2(eth2) entered blocking state
Sun Jul 2 09:21:24 2023 kern.info kernel: [1896244.036041] br-lan: port 2(eth2) entered forwarding state
Sun Jul 2 09:21:24 2023 daemon.notice netifd: Network device 'eth2' link is up
Sun Jul 2 09:21:24 2023 daemon.notice netifd: bridge 'br-lan' link is up
Sun Jul 2 09:21:24 2023 daemon.notice netifd: Interface 'lan' has link connectivity
Sun Jul 2 20:13:31 2023 daemon.notice netifd: wan (30701): udhcpc: sending renew to server 192.168.1.1
Sun Jul 2 20:13:31 2023 daemon.notice netifd: wan (30701): udhcpc: lease of 192.168.1.149 obtained from 192.168.1.1, lease time 86400
Sun Jul 2 23:17:01 2023 kern.info kernel: [1946382.500871] r8125: eth2: link down
Sun Jul 2 23:17:01 2023 daemon.notice netifd: Network device 'eth2' link is down
Sun Jul 2 23:17:01 2023 kern.info kernel: [1946382.513671] br-lan: port 2(eth2) entered disabled state
Sun Jul 2 23:17:02 2023 daemon.notice netifd: bridge 'br-lan' link is down
Sun Jul 2 23:17:02 2023 daemon.notice netifd: Interface 'lan' has link connectivity loss
Mon Jul 3 08:13:31 2023 daemon.notice netifd: wan (30701): udhcpc: sending renew to server 192.168.1.1
Mon Jul 3 08:13:31 2023 daemon.notice netifd: wan (30701): udhcpc: lease of 192.168.1.149 obtained from 192.168.1.1, lease time 86400
Mon Jul 3 20:13:31 2023 daemon.notice netifd: wan (30701): udhcpc: sending renew to server 192.168.1.1
Mon Jul 3 20:13:31 2023 daemon.notice netifd: wan (30701): udhcpc: lease of 192.168.1.149 obtained from 192.168.1.1, lease time 86400
Mon Jul 3 22:39:30 2023 kern.info kernel: [2030533.407509] r8125: eth2: link up
Mon Jul 3 22:39:30 2023 kern.info kernel: [2030533.407893] br-lan: port 2(eth2) entered blocking state Mon Jul 3 22:39:30 2023 kern.info kernel: [2030533.408682] br-lan: port 2(eth2) entered forwarding state Mon Jul 3 22:39:30 2023 daemon.notice netifd: Network device 'eth2' link is up
Mon Jul 3 22:39:30 2023 daemon.notice netifd: bridge 'br-lan' link is up
Mon Jul 3 22:39:30 2023 daemon.notice netifd: Interface 'lan' has link connectivity
Mon Jul 3 22:39:46 2023 kern.info kernel: [2030549.735791] r8125: eth2: link down
Mon Jul 3 22:39:46 2023 daemon.notice netifd: Network device 'eth2' link is down
Mon Jul 3 22:39:46 2023 kern.info kernel: [2030549.748484] br-lan: port 2(eth2) entered disabled state Mon Jul 3 22:39:47 2023 daemon.notice netifd: bridge 'br-lan' link is down
Mon Jul 3 22:39:47 2023 daemon.notice netifd: Interface 'lan' has link connectivity loss
Mon Jul 3 22:39:49 2023 kern.info kernel: [2030553.064063] r8125: eth2: link up
Mon Jul 3 22:39:49 2023 kern.info kernel: [2030553.064457] br-lan: port 2(eth2) entered blocking state Mon Jul 3 22:39:49 2023 kern.info kernel: [2030553.065233] br-lan: port 2(eth2) entered forwarding state Mon Jul 3 22:39:49 2023 daemon.notice netifd: Network device 'eth2' link is up
Mon Jul 3 22:39:49 2023 daemon.notice netifd: bridge 'br-lan' link is up
Mon Jul 3 22:39:49 2023 daemon.notice netifd: Interface 'lan' has link connectivity
Tue Jul 4 00:12:04 2023 kern.info kernel: [2036088.083926] r8125: eth2: link down
Tue Jul 4 00:12:04 2023 daemon.notice netifd: Network device 'eth2' link is down
Tue Jul 4 00:12:04 2023 kern.info kernel: [2036088.096717] br-lan: port 2(eth2) entered disabled state
Tue Jul 4 00:12:05 2023 daemon.notice netifd: bridge 'br-lan' link is down
Tue Jul 4 00:12:05 2023 daemon.notice netifd: Interface 'lan' has link connectivity loss
Tue Jul 4 08:13:31 2023 daemon.notice netifd: wan (30701): udhcpc: sending renew to server 192.168.1.1
[... 248 lines cut ...]
Thu Jul 6 23:03:01 2023 daemon.notice netifd: Interface 'wan' is disabled
Thu Jul 6 23:03:01 2023 daemon.notice netifd: Interface 'wan6' is disabled
Thu Jul 6 23:03:01 2023 kern.info kernel: [2291151.890760] RTL8211F Gigabit Ethernet stmmac-0:01: attached PHY driver [RTL8211F Gigabit Ethernet] (mii_bus:phy_addr=st
Thu Jul 6 23:03:01 2023 kern.info kernel: [2291151.902656] dwmac4: Master AXI performs any burst length
Thu Jul 6 23:03:01 2023 kern.info kernel: [2291151.903200] rk_gmac-dwmac fe2a0000.ethernet eth0: No Safety Features support found
Thu Jul 6 23:03:01 2023 kern.info kernel: [2291151.904412] rk_gmac-dwmac fe2a0000.ethernet eth0: IEEE 1588-2008 Advanced Timestamp supported
Thu Jul 6 23:03:01 2023 kern.info kernel: [2291151.906018] rk_gmac-dwmac fe2a0000.ethernet eth0: registered PTP clock
Thu Jul 6 23:03:01 2023 kern.info kernel: [2291151.907687] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
Thu Jul 6 23:03:01 2023 daemon.notice netifd: Interface 'wan' is enabled
Thu Jul 6 23:03:01 2023 daemon.notice netifd: Interface 'wan6' is enabled
Thu Jul 6 23:03:10 2023 kern.info kernel: [2291161.140471] rk_gmac-dwmac fe2a0000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
Thu Jul 6 23:03:10 2023 kern.info kernel: [2291161.141341] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
Thu Jul 6 23:03:10 2023 daemon.notice netifd: Network device 'eth0' link is up
Thu Jul 6 23:03:10 2023 daemon.notice netifd: Interface 'wan' has link connectivity
Thu Jul 6 23:03:10 2023 daemon.notice netifd: Interface 'wan' is setting up now
Thu Jul 6 23:03:10 2023 daemon.notice netifd: Interface 'wan6' has link connectivity
Thu Jul 6 23:03:10 2023 daemon.notice netifd: wan (15825): udhcpc: started, v1.35.0
Thu Jul 6 23:03:10 2023 daemon.notice netifd: wan (15825): udhcpc: broadcasting discover
Thu Jul 6 23:03:14 2023 daemon.notice netifd: wan (15825): udhcpc: broadcasting discover
Thu Jul 6 23:03:14 2023 daemon.notice netifd: wan (15825): udhcpc: broadcasting select for 192.168.1.149, server 192.168.1.1
Thu Jul 6 23:03:14 2023 daemon.notice netifd: wan (15825): udhcpc: lease of 192.168.1.149 obtained from 192.168.1.1, lease time 86400
Thu Jul 6 23:03:14 2023 daemon.notice netifd: Interface 'wan' is now up
Thu Jul 6 23:03:24 2023 user.notice firewall: Reloading firewall due to ifup of wan (eth0)
Thu Jul 6 23:03:24 2023 user.notice nlbwmon: Reloading nlbwmon due to ifup of wan (eth0)
```

nuo12138, 2023-07-13:
是不是开了sfe，我r4s开sfe会断流重启，换flow offloading就不断流了

Amoystyle, 2023-07-14:
> 是不是开了sfe，我r4s开sfe会断流重启，换flow offloading就不断流了

我试试先，上面的那种断流偶尔出现。

closed 2024-06-04 by Amoystyle