coolsnowwolf/lede #11363: 最新源码无法编译frpc

Yofo-me, 2023-07-08:
最新源码无法编译frp

```
平台： Ubuntu 20.04 LTS
目标平台： x64
```

日志：
```log

github.com/fatedier/frp/server/proxy
github.com/fatedier/frp/cmd/frpc/sub
github.com/fatedier/frp/server
github.com/fatedier/frp/cmd/frpc
github.com/fatedier/frp/cmd/frps
# github.com/fatedier/frp/cmd/frpc
/home/yoki/lede/staging_dir/hostpkg/lib/go-cross/pkg/tool/linux_amd64/link: running x86_64-openwrt-linux-musl-gcc failed: exit status 1
/home/yoki/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib/gcc/x86_64-openwrt-linux-musl/11.3.0/../../../../x86_64-openwrt-linux-musl/bin/ld: cannot find /home/yoki/lede/tmp/go-link-2260921937/go.o: Invalid argument
collect2: error: ld returned 1 exit status

# github.com/fatedier/frp/cmd/frps
/home/yoki/lede/staging_dir/hostpkg/lib/go-cross/pkg/tool/linux_amd64/link: running x86_64-openwrt-linux-musl-gcc failed: exit status 1
/home/yoki/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib/gcc/x86_64-openwrt-linux-musl/11.3.0/../../../../x86_64-openwrt-linux-musl/bin/ld: cannot find /home/yoki/lede/tmp/go-link-3196603506/go.o: Invalid argument
collect2: error: ld returned 1 exit status

make[3]: *** [Makefile:67: /home/yoki/lede/build_dir/target-x86_64_musl/frp-0.47.0/.built] Error 1
make[3]: Leaving directory '/home/yoki/lede/feeds/packages/net/frp'
time: package/feeds/packages/frp/compile#23.81#17.04#20.24
    ERROR: package/feeds/packages/frp failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/frp/compile] Error 1
make[2]: Leaving directory '/home/yoki/lede'
make[1]: *** [package/Makefile:110: /home/yoki/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/yoki/lede'
make: *** [/home/yoki/lede/include/toplevel.mk:231: world] Error 2
```

WYC-2020, 2023-07-25:
刚试了没问题，确定go 版本 适合l 仓库一致的？

yanxin152133, 2023-08-31:
[image]

mrchenxc168, 2023-10-09:
我也一样，没有选frpc的时候，顺利编译。添加frpc插件之后也是报错

yanxin152133, 2023-10-09:
> 我也一样，没有选frpc的时候，顺利编译。添加frpc插件之后也是报错
> 

试试低版本的

yanxin152133, 2023-10-13:
0.44版本的可以正常编译。
[image]

closed 2026-09-05 by coolsnowwolf