coolsnowwolf/lede #11365: 编译报错 UnblockNeteaseMusic

CussiusKon, 2023-07-08:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：第一次按照文档进行，配置上几乎没有做什么修改，编译成功，第二次编译添加了一些其他应用源，编译失败了，查过历史issues，有类似的但不一样。请帮我看一下是什么原因，谢谢

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：
```
touch /home/debian12/lede/build_dir/target-x86_64_musl/UnblockNeteaseMusic-0.2.13/.configured_5a4a454b6584916af2b82bcefc223b97
rm -f /home/debian12/lede/build_dir/target-x86_64_musl/UnblockNeteaseMusic-0.2.13/.built
touch /home/debian12/lede/build_dir/target-x86_64_musl/UnblockNeteaseMusic-0.2.13/.built_check
CONFIG_GOLANG_MOD_CACHE_WORLD_READABLE="" GO_BUILD_CACHE_DIR="/home/debian12/lede/tmp/go-build" GO_MOD_CACHE_DIR="/home/debian12/lede/dl/go-mod-cache" GO_MOD_ARGS="-modcacherw" GO_PKG="github.com/cnsilvan/UnblockNeteaseMusic" GO_INSTALL_EXTRA="" GO_INSTALL_ALL="" GO_SOURCE_ONLY="" GO_BUILD_PKG="github.com/cnsilvan/UnblockNeteaseMusic" GO_EXCLUDES="" GO_GO_GENERATE="" GO_INSTALL_BIN_PATH="/usr/bin" BUILD_DIR="/home/debian12/lede/build_dir/target-x86_64_musl/UnblockNeteaseMusic-0.2.13" GO_BUILD_DIR="/home/debian12/lede/build_dir/target-x86_64_musl/UnblockNeteaseMusic-0.2.13/.go_work/build" GO_BUILD_BIN_DIR="/home/debian12/lede/build_dir/target-x86_64_musl/UnblockNeteaseMusic-0.2.13/.go_work/build/bin" GO_BUILD_DEPENDS_PATH="/usr/share/gocode" GO_BUILD_DEPENDS_SRC="/home/debian12/lede/staging_dir/target-x86_64_musl/usr/share/gocode/src" GOOS="linux" GOARCH="amd64" GO386="" GOAMD64="v1" GOARM="" GOMIPS="" GOMIPS64="" GOPPC64="" CGO_ENABLED=1 CC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" CGO_CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/debian12/lede/build_dir/target-x86_64_musl/UnblockNeteaseMusic-0.2.13=UnblockNeteaseMusic-0.2.13 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CGO_CPPFLAGS="-I/home/debian12/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/include -I/home/debian12/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include/fortify -I/home/debian12/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/include" CGO_CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/debian12/lede/build_dir/target-x86_64_musl/UnblockNeteaseMusic-0.2.13=UnblockNeteaseMusic-0.2.13 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" CGO_LDFLAGS="-L/home/debian12/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/lib -L/home/debian12/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib -znow -zrelro" GOPATH="/home/debian12/lede/build_dir/target-x86_64_musl/UnblockNeteaseMusic-0.2.13/.go_work/build" GOCACHE="/home/debian12/lede/tmp/go-build" GOMODCACHE="/home/debian12/lede/dl/go-mod-cache" GOENV=off /usr/bin/env bash ../../lang/golang//golang-build.sh build -v -buildvcs=false -trimpath -ldflags "all=-buildid '1688660398' -linkmode external -extldflags '-L/home/debian12/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/lib -L/home/debian12/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib -Wl,-z,now -Wl,-z,relro'"        -ldflags "-s -w -X 'github.com/cnsilvan/UnblockNeteaseMusic/version.Version=0.2.13' -X 'github.com/cnsilvan/UnblockNeteaseMusic/version.BuildTime=2023-07-08 16:47:00' -X 'github.com/cnsilvan/UnblockNeteaseMusic/version.ExGoVersionInfo= '  -buildid '1688660398' -linkmode external -extldflags '-L/home/debian12/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/usr/lib -L/home/debian12/lede/staging_dir/toolchain-x86_64_gcc-11.3.0_musl/lib -Wl,-z,now -Wl,-z,relro'"  
Finding targets

Building targets
golang.org/x/text/width
# golang.org/x/text/width
../../../../../dl/go-mod-cache/golang.org/x/text@v0.3.3/width/width.go:100:7: undefined: elem
../../../../../dl/go-mod-cache/golang.org/x/text@v0.3.3/width/width.go:104:9: undefined: elem
../../../../../dl/go-mod-cache/golang.org/x/text@v0.3.3/width/transform.go:34:6: undefined: elem
../../../../../dl/go-mod-cache/golang.org/x/text@v0.3.3/width/transform.go:34:14: undefined: tagNeedsFold
../../../../../dl/go-mod-cache/golang.org/x/text@v0.3.3/width/transform.go:72:6: undefined: elem
../../../../../dl/go-mod-cache/golang.org/x/text@v0.3.3/width/transform.go:72:14: undefined: tagNeedsFold
../../../../../dl/go-mod-cache/golang.org/x/text@v0.3.3/width/transform.go:116:11: undefined: elem
../../../../../dl/go-mod-cache/golang.org/x/text@v0.3.3/width/transform.go:155:11: undefined: elem
../../../../../dl/go-mod-cache/golang.org/x/text@v0.3.3/width/transform.go:196:11: undefined: elem
../../../../../dl/go-mod-cache/golang.org/x/text@v0.3.3/width/transform.go:218:11: undefined: elem
../../../../../dl/go-mod-cache/golang.org/x/text@v0.3.3/width/transform.go:218:11: too many errors

make[3]: *** [Makefile:86: /home/debian12/lede/build_dir/target-x86_64_musl/UnblockNeteaseMusic-0.2.13/.built] Error 1
make[3]: Leaving directory '/home/debian12/lede/feeds/packages/multimedia/UnblockNeteaseMusic-Go'
time: package/feeds/packages/UnblockNeteaseMusic-Go/compile#0.59#0.85#2.07
    ERROR: package/feeds/packages/UnblockNeteaseMusic-Go failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/UnblockNeteaseMusic-Go/compile] Error 1
make[2]: Leaving directory '/home/debian12/lede'
make[1]: *** [package/Makefile:110: /home/debian12/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/debian12/lede'
make: *** [/home/debian12/lede/include/toplevel.mk:231：world] 错误 2
```

lxg2001, 2023-07-08:
这是网络问题，执行以下命令，然后再编译：
echo "export GO111MODULE=on" >> ~/.profile
echo "export GOPROXY=https://goproxy.cn" >> ~/.profile
source ~/.profile

CussiusKon, 2023-07-08:
> 这是网络问题，执行以下命令，再编译： `echo "export GO111MODULE=on" >> ~/.profile` `echo "export GOPROXY=https://goproxy.cn" >> ~/.profile` `source ~/.profile`

报错了
```
# golang.org/x/text/width
../../../../../dl/go-mod-cache/golang.org/x/text@v0.3.3/width/width.go:100:7: undefined: elem
../../../../../dl/go-mod-cache/golang.org/x/text@v0.3.3/width/width.go:104:9: undefined: elem
../../../../../dl/go-mod-cache/golang.org/x/text@v0.3.3/width/transform.go:34:6: undefined: elem
../../../../../dl/go-mod-cache/golang.org/x/text@v0.3.3/width/transform.go:34:14: undefined: tagNeedsFold
../../../../../dl/go-mod-cache/golang.org/x/text@v0.3.3/width/transform.go:72:6: undefined: elem
../../../../../dl/go-mod-cache/golang.org/x/text@v0.3.3/width/transform.go:72:14: undefined: tagNeedsFold
../../../../../dl/go-mod-cache/golang.org/x/text@v0.3.3/width/transform.go:116:11: undefined: elem
../../../../../dl/go-mod-cache/golang.org/x/text@v0.3.3/width/transform.go:155:11: undefined: elem
../../../../../dl/go-mod-cache/golang.org/x/text@v0.3.3/width/transform.go:196:11: undefined: elem
../../../../../dl/go-mod-cache/golang.org/x/text@v0.3.3/width/transform.go:218:11: undefined: elem
../../../../../dl/go-mod-cache/golang.org/x/text@v0.3.3/width/transform.go:218:11: too many errors
os/exec
regexp/syntax
regexp
github.com/cnsilvan/UnblockNeteaseMusic/version
github.com/cnsilvan/UnblockNeteaseMusic/cache
github.com/cnsilvan/UnblockNeteaseMusic/pool
html
github.com/cnsilvan/UnblockNeteaseMusic/processor/crypto
github.com/mitchellh/mapstructure
net/http/httptrace
os/signal
net/http
github.com/cnsilvan/UnblockNeteaseMusic/cookiestxt

make[3]: *** [Makefile:86: /home/debian12/lede/build_dir/target-x86_64_musl/UnblockNeteaseMusic-0.2.13/.built] Error 1
make[3]: Leaving directory '/home/debian12/lede/feeds/packages/multimedia/UnblockNeteaseMusic-Go'
time: package/feeds/packages/UnblockNeteaseMusic-Go/compile#24.45#4.75#17.95
    ERROR: package/feeds/packages/UnblockNeteaseMusic-Go failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/UnblockNeteaseMusic-Go/compile] Error 1
make[2]: Leaving directory '/home/debian12/lede'
make[1]: *** [package/Makefile:110: /home/debian12/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/debian12/lede'
make: *** [/home/debian12/lede/include/toplevel.mk:231：world] 错误 2
```

lxg2001, 2023-07-08:
你把go-mod-cache删了再试试，再不行的话，我也不知道了

CussiusKon, 2023-07-31:
谢谢~
网络问题影响真的很大，不稳定很难编译成功

closed 2023-07-31 by CussiusKon