coolsnowwolf/lede #11367: R2S/R4S编译报错kernel-headers failed to build

fxxccp, 2023-07-08:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：R2S/R4S

### (2) 路由器型号和固件版本

A：5.15

### (3) 详细日志

 make[3] -C tools/firmware-utils compile
 make[2] toolchain/compile
 make[3] -C toolchain/binutils compile
 make[3] -C toolchain/gdb compile
 make[3] -C toolchain/fortify-headers compile
 make[3] -C toolchain/gcc/initial compile
 make[3] -C toolchain/kernel-headers compile
 make[3] -C toolchain/musl compile
    ERROR: toolchain/kernel-headers failed to build.
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 1
make[2]: Entering directory '/workdir/openwrt/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/workdir/openwrt/scripts/config'
 make[1] world
 make[2] toolchain/compile
 make[3] -C toolchain/gdb compile
 make[3] -C toolchain/binutils compile
 make[3] -C toolchain/gcc/initial compile
 make[3] -C toolchain/kernel-headers compile
    ERROR: toolchain/kernel-headers failed to build.
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 1
make[2]: Entering directory '/workdir/openwrt/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/workdir/openwrt/scripts/config'
make[1]: Entering directory '/workdir/openwrt'
make[2]: Entering directory '/workdir/openwrt'
+ mkdir -p /workdir/openwrt/staging_dir/target-aarch64_generic_musl
+ cd /workdir/openwrt/staging_dir/target-aarch64_generic_musl
+ mkdir -p bin lib stamp usr/include usr/lib
mkdir -p /workdir/openwrt/build_dir/target-aarch64_generic_musl/stamp
touch /workdir/openwrt/staging_dir/target-aarch64_generic_musl/.prepared
make[3]: Entering directory '/workdir/openwrt/toolchain/gdb'
make[3]: Nothing to be done for 'compile'.
make[3]: Leaving directory '/workdir/openwrt/toolchain/gdb'
time: toolchain/gdb/compile#0.00#0.00#0.01
make[3]: Entering directory '/workdir/openwrt/toolchain/binutils'
make[3]: Nothing to be done for 'compile'.
make[3]: Leaving directory '/workdir/openwrt/toolchain/binutils'
time: toolchain/binutils/compile#0.00#0.00#0.01
make[3]: Entering directory '/workdir/openwrt/toolchain/gcc/initial'
make[3]: Nothing to be done for 'compile'.
make[3]: Leaving directory '/workdir/openwrt/toolchain/gcc/initial'
time: toolchain/gcc/initial/compile#0.00#0.00#0.01
make[3]: Entering directory '/workdir/openwrt/toolchain/kernel-headers'
rm -rf /workdir/openwrt/build_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/linux-*
xzcat /workdir/openwrt/dl/linux-5.15.120.tar.xz | tar -C /workdir/openwrt/build_dir/toolchain-aarch64_generic_gcc-11.3.0_musl -xf -
cp -fpR "/workdir/openwrt/target/linux/generic/files"/. /workdir/openwrt/build_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/linux-5.15.120/
find /workdir/openwrt/build_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/linux-5.15.120/ -name \*.rej -or -name \*.orig | xargs -r rm -f
if [ -d /workdir/openwrt/target/linux/generic/patches ]; then echo "generic patches directory is present. please move your patches to the pending directory" ; exit 1; fi

Applying /workdir/openwrt/target/linux/generic/backport-5.15/020-v6.1-01-mm-x86-arm64-add-arch_has_hw_pte_young.patch using plaintext: 
[... 230 lines cut ...]
patching file drivers/phy/marvell/phy-mvebu-a3700-comphy.c

Applying /workdir/openwrt/target/linux/generic/backport-5.15/344-v5.18-02-phy-marvell-phy-mvebu-a3700-comphy-Add-native-kernel.patch using plaintext: 
patching file drivers/phy/marvell/phy-mvebu-a3700-comphy.c

Applying /workdir/openwrt/target/linux/generic/backport-5.15/345-v5.17-arm64-dts-marvell-armada-37xx-Add-xtal-clock-to-comp.patch using plaintext: 
patching file arch/arm64/boot/dts/marvell/armada-37xx.dtsi

Applying /workdir/openwrt/target/linux/generic/backport-5.15/346-v5.18-01-Revert-ata-ahci-mvebu-Make-SATA-PHY-optional-for-Arm.patch using plaintext: 
patching file drivers/ata/ahci.h
Hunk #1 FAILED at 240.
1 out of 1 hunk FAILED -- saving rejects to file drivers/ata/ahci.h.rej
make[3]: *** [Makefile:121: /workdir/openwrt/build_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/linux-5.15.120/.prepared] Error 1
patching file drivers/ata/ahci_mvebu.c
patching file drivers/ata/libahci_platform.c
Patch failed!  Please fix /workdir/openwrt/target/linux/generic/backport-5.15/346-v5.18-01-Revert-ata-ahci-mvebu-Make-SATA-PHY-optional-for-Arm.patch!
make[3]: Leaving directory '/workdir/openwrt/toolchain/kernel-headers'
time: toolchain/kernel-headers/compile#10.39#4.44#12.55
    ERROR: toolchain/kernel-headers failed to build.
make[2]: Leaving directory '/workdir/openwrt'
make[2]: *** [toolchain/Makefile:97: toolchain/kernel-headers/compile] Error 1
make[1]: Leaving directory '/workdir/openwrt'
make[1]: *** [toolchain/Makefile:93: /workdir/openwrt/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/stamp/.toolchain_compile] Error 2
make: *** [/workdir/openwrt/include/toplevel.mk:231: world] Error 2
Error: Process completed with exit code 2.

yike8, 2023-07-08:
好巧，我今天git pull之后，也是这个问题。
Applying /home/user/lede/target/linux/generic/backport-5.15/346-v5.18-01-Revert-ata-ahci-mvebu-Make-SATA-PHY-optional-for-Arm.patch using plaintext: 
patching file drivers/ata/ahci.h
Hunk #1 FAILED at 240.
1 out of 1 hunk FAILED -- saving rejects to file drivers/ata/ahci.h.rej
patching file drivers/ata/ahci_mvebu.c
patching file drivers/ata/libahci_platform.c
Patch failed!  Please fix /home/user/lede/target/linux/generic/backport-5.15/346-v5.18-01-Revert-ata-ahci-mvebu-Make-SATA-PHY-optional-for-Arm.patch!
make[3]: *** [Makefile:121: /home/user/lede/build_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/linux-5.15.120/.prepared] Error 1
make[3]: Leaving directory '/home/user/lede/toolchain/kernel-headers'
time: toolchain/kernel-headers/compile#1.87#15.29#15.59
    ERROR: toolchain/kernel-headers failed to build.
make[2]: *** [toolchain/Makefile:97: toolchain/kernel-headers/compile] Error 1
make[2]: Leaving directory '/home/user/lede'
make[1]: *** [toolchain/Makefile:93: /home/user/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/stamp/.toolchain_compile] Error 2
make[1]: Leaving directory '/home/user/lede'
make: *** [/home/user/lede/include/toplevel.mk:231: world] Error 2

chpos, 2023-07-08:
电犀牛r68s 4G版编译同样报错

./tools/install.sh -D -m 644 include/threads.h /home/cucy/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0            _musl//include/threads.h
./tools/install.sh -D -m 644 include/time.h /home/cucy/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_mu            sl//include/time.h
./tools/install.sh -D -m 644 include/uchar.h /home/cucy/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_m            usl//include/uchar.h
./tools/install.sh -D -m 644 include/ucontext.h /home/cucy/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.            0_musl//include/ucontext.h
./tools/install.sh -D -m 644 include/ulimit.h /home/cucy/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_            musl//include/ulimit.h
./tools/install.sh -D -m 644 include/unistd.h /home/cucy/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_            musl//include/unistd.h
./tools/install.sh -D -m 644 include/utime.h /home/cucy/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_m            usl//include/utime.h
./tools/install.sh -D -m 644 include/utmp.h /home/cucy/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_mu            sl//include/utmp.h
./tools/install.sh -D -m 644 include/utmpx.h /home/cucy/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_m            usl//include/utmpx.h
./tools/install.sh -D -m 644 include/values.h /home/cucy/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_            musl//include/values.h
./tools/install.sh -D -m 644 include/wait.h /home/cucy/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_mu            sl//include/wait.h
./tools/install.sh -D -m 644 include/wchar.h /home/cucy/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_m            usl//include/wchar.h
./tools/install.sh -D -m 644 include/wctype.h /home/cucy/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_            musl//include/wctype.h
./tools/install.sh -D -m 644 include/wordexp.h /home/cucy/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0            _musl//include/wordexp.h
./tools/install.sh -D -l libc.so /home/cucy/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl//lib/ld-            musl-aarch64.so.1 || true
make[4]: Leaving directory '/home/cucy/lede/build_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/musl-1.2.3'
cp -fpR ./include /home/cucy/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/
mkdir -p /home/cucy/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/stamp
touch /home/cucy/lede/build_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/musl-1.2.3/.built
touch /home/cucy/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/stamp/.musl_installed
make[3]: Leaving directory '/home/cucy/lede/toolchain/musl'
time: toolchain/musl/compile#75.85#16.93#13.42
make[2]: Leaving directory '/home/cucy/lede'
make[1]: *** [toolchain/Makefile:93: /home/cucy/lede/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/stam            p/.toolchain_compile] Error 2
make[1]: Leaving directory '/home/cucy/lede'
make: *** [/home/cucy/lede/include/toplevel.mk:231: world] Error 2

xiaotan8, 2023-07-08:
X86 同样报错

closed 2023-07-22 by aiamadeus