coolsnowwolf/lede #11368: 最新源码编译cloudflared错误

kjrzxu, 2023-07-08:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

* 没有类似的issue

## 2. 详细叙述

### (1)路由器型号和固件版本

hiwifi_hc5962

### (2) 云编译日志

2023-07-08T12:05:55.9352175Z mipsel-openwrt-linux-musl-gcc -Isrc/headers/ -I./src/headers/ -I../ -I./../ -DLTC_SOURCE -I../libtommath/ -I./../libtommath/ -Os -pipe -mno-branch-likely -mips32r2 -mtune=24kc -fno-caller-saves -fno-plt -fhonour-copts -msoft-float -fmacro-prefix-map=/home/runner/work/OpenWrt-CI/OpenWrt-CI/openwrt/build_dir/target-mipsel_24kc_musl/dropbear-2022.82=dropbear-2022.82 -Wformat -Werror=format-security -DPIC -fpic -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro -DARGTYPE=3 -ffunction-sections -fdata-sections -flto  -Wno-pointer-sign -fno-strict-overflow  -I./libtomcrypt/src/headers/ -DLOCALOPTIONS_H_EXISTS -I/home/runner/work/OpenWrt-CI/OpenWrt-CI/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/usr/include -I/home/runner/work/OpenWrt-CI/OpenWrt-CI/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/include/fortify -I/home/runner/work/OpenWrt-CI/OpenWrt-CI/openwrt/staging_dir/toolchain-mipsel_24kc_gcc-8.4.0_musl/include  -I. -I. -DDROPBEAR_SERVER -DDROPBEAR_CLIENT -DPROGRESS_METER -DDBMULTI_dropbear -DDBMULTI_dbclient -DDBMULTI_dropbearkey -DDBMULTI_scp -DDROPBEAR_MULTI -c src/modes/xts/xts_init.c -o src/modes/xts/xts_init.o
2023-07-08T12:05:55.9366966Z make[3]: Leaving directory '/home/runner/work/OpenWrt-CI/OpenWrt-CI/openwrt/feeds/packages/net/cloudflared'
2023-07-08T12:05:55.9371095Z make[3]: *** [Makefile:68: /home/runner/work/OpenWrt-CI/OpenWrt-CI/openwrt/build_dir/target-mipsel_24kc_musl/cloudflared-2023.2.2/.built] Error 1
2023-07-08T12:05:55.9371818Z time: package/feeds/packages/cloudflared/compile#50.87#8.29#59.28
2023-07-08T12:05:55.9386254Z     ERROR: package/feeds/packages/cloudflared failed to build.
2023-07-08T12:05:55.9394271Z make[2]: *** [package/Makefile:116: package/feeds/packages/cloudflared/compile] Error 1

WYC-2020, 2023-07-25:
这是这个版本不适配go1.20 升级即可 ，功能自测
[image]

kjrzxu, 2023-07-28:
确实是这个问题，更换版本后编译成功。

kjrzxu, 2023-07-28:
> 这是这个版本不适配go1.20 升级即可 ，功能自测
> 
> [image]
> 
> 

确实是这个问题，更换版本后编译成功

hzjnet, 2023-09-25:
go版本升级后对应应用软件源码是不是应该由源码开发者自动升级更新啊？ 编译人员根本搞不懂喔

closed 2026-09-05 by coolsnowwolf