coolsnowwolf/lede #11373: 编译r4se固件后bin/targets下居然没有一个固件文件bin下有的都是ipk

magi-Z, 2023-07-09:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述
编译r4se固件后bin/targets里没固件文件都是些ipk文件，固件设置时选的256和512
### (1) 具体问题

A：编译r4se固件后targets下没有固件文件bin里各层文件夹都是ipk，ipk都编译成功为什么没有固件文件？

### (2) 路由器型号和固件版本

A：r4se

### (3) 详细日志

A：以上日志都正常
gopkg.in/square/go-jose.v2
github.com/coreos/go-oidc
github.com/boombuler/barcode
github.com/boombuler/barcode/utils
github.com/boombuler/barcode/qr
github.com/pquerna/otp
github.com/pquerna/otp/internal
github.com/pquerna/otp/hotp
github.com/pquerna/otp/totp
github.com/alist-org/alist/v3/server/handles
github.com/alist-org/alist/v3/server/middlewares
github.com/alist-org/alist/v3/server/webdav/internal/xml
github.com/alist-org/alist/v3/server/webdav
github.com/gin-contrib/cors
github.com/alist-org/alist/v3/server
github.com/spf13/pflag
github.com/spf13/cobra
github.com/alist-org/alist/v3/drivers/template
github.com/winfsp/cgofuse/fuse
# github.com/winfsp/cgofuse/fuse
../../../../../dl/go-mod-cache/github.com/winfsp/cgofuse@v1.5.0/fuse/host_cgo.go:116:10: fatal error: fuse.h: No such file or directory
  116 | #include <fuse.h>
      |          ^~~~~~~~
compilation terminated.
github.com/alist-org/alist/v3/pkg/chanio
github.com/alist-org/alist/v3/pkg/gowebdav/cmd/gowebdav
gorm.io/driver/sqlite
github.com/alist-org/alist/v3/internal/bootstrap
github.com/alist-org/alist/v3/cmd
github.com/alist-org/alist/v3

make[3]: *** [Makefile:76: /home/magi/lede/build_dir/target-aarch64_generic_musl/alist-3.20.1/.built] Error 1
make[3]: Leaving directory '/home/magi/lede/package/small-package/alist'
time: package/small-package/alist/compile#97.43#18.78#100.30
    ERROR: package/small-package/alist failed to build.
make[2]: *** [package/Makefile:116: package/small-package/alist/compile] Error 1
make[2]: Leaving directory '/home/magi/lede'
make[1]: *** [package/Makefile:110: /home/magi/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/magi/lede'
make: *** [/home/magi/lede/include/toplevel.mk:231：world] 错误 2

XXXstyle, 2023-07-09:
日志显示有错误，没有完成编译啊

magi-Z, 2023-07-09:
> 日志显示有错误，没有完成编译啊
能判断出什么原因总是出在哪？alist导致的吗取消这个编译呗！

lxg2001, 2023-07-09:
缺少依赖，安装libfuse-dev
`sudo apt install libfuse-dev`

magi-Z, 2023-07-09:
> 少文件，安装libfuse-dev `sudo apt install libfuse-dev`

安装依赖后仍然alist失败报错，清了配置重新取消alist配置试试！

closed 2023-07-24 by aiamadeus