coolsnowwolf/lede #11374: 最新源码无法编译qmi_wwan_f，qmi_wwan

Yofo-me, 2023-07-09:
from /home/yoki/lede/build_dir/target-x86_64_musl/linux-x86_64/qmi_wwan_f-1.0/qmi_wwan_f.c:13:
/home/yoki/lede/build_dir/target-x86_64_musl/linux-x86_64/qmi_wwan_f-1.0/qmi_wwan_f.c: In function 'qmap_register_device':
/home/yoki/lede/build_dir/target-x86_64_musl/linux-x86_64/qmi_wwan_f-1.0/qmi_wwan_f.c:1170:21: warning: passing argument 1 of '__builtin_memcpy' discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
 1170 |     memcpy (qmap_net->dev_addr, real_dev->dev_addr, ETH_ALEN);
      |             ~~~~~~~~^~~~~~~~~~
./include/linux/fortify-string.h:469:27: note: in definition of macro '__fortify_memcpy_chk'
  469 |         __underlying_##op(p, q, __fortify_size);                        \
      |                           ^
/home/yoki/lede/build_dir/target-x86_64_musl/linux-x86_64/qmi_wwan_f-1.0/qmi_wwan_f.c:1170:5: note: in expansion of macro 'memcpy'
 1170 |     memcpy (qmap_net->dev_addr, real_dev->dev_addr, ETH_ALEN);
      |     ^~~~~~
/home/yoki/lede/build_dir/target-x86_64_musl/linux-x86_64/qmi_wwan_f-1.0/qmi_wwan_f.c:1170:21: note: expected 'void *' but argument is of type 'const unsigned char *'
 1170 |     memcpy (qmap_net->dev_addr, real_dev->dev_addr, ETH_ALEN);
      |             ~~~~~~~~^~~~~~~~~~
./include/linux/fortify-string.h:469:27: note: in definition of macro '__fortify_memcpy_chk'
  469 |         __underlying_##op(p, q, __fortify_size);                        \
      |                           ^
/home/yoki/lede/build_dir/target-x86_64_musl/linux-x86_64/qmi_wwan_f-1.0/qmi_wwan_f.c:1170:5: note: in expansion of macro 'memcpy'
 1170 |     memcpy (qmap_net->dev_addr, real_dev->dev_addr, ETH_ALEN);
      |     ^~~~~~
/home/yoki/lede/build_dir/target-x86_64_musl/linux-x86_64/qmi_wwan_f-1.0/qmi_wwan_f.c: In function 'qmi_wwan_bind':
/home/yoki/lede/build_dir/target-x86_64_musl/linux-x86_64/qmi_wwan_f-1.0/qmi_wwan_f.c:1860:31: error: assignment of read-only location '*dev->net->dev_addr'
 1860 |         dev->net->dev_addr[0] |= 0x02;    /* set local assignment bit */
      |                               ^~
/home/yoki/lede/build_dir/target-x86_64_musl/linux-x86_64/qmi_wwan_f-1.0/qmi_wwan_f.c:1861:31: error: assignment of read-only location '*dev->net->dev_addr'
 1861 |         dev->net->dev_addr[0] &= 0xbf;    /* clear "IP" bit */
      |                               ^~
/home/yoki/lede/build_dir/target-x86_64_musl/linux-x86_64/qmi_wwan_f-1.0/qmi_wwan_f.c: In function 'rmnet_usb_bind':
/home/yoki/lede/build_dir/target-x86_64_musl/linux-x86_64/qmi_wwan_f-1.0/qmi_wwan_f.c:2072:5: warning: ISO C90 forbids mixed declarations and code [-Wdeclaration-after-statement]
 2072 |     int status = qmi_wwan_bind(dev, intf);
      |     ^~~
cc1: some warnings being treated as errors
make[5]: *** [scripts/Makefile.build:250: /home/yoki/lede/build_dir/target-x86_64_musl/linux-x86_64/qmi_wwan_f-1.0/qmi_wwan_f.o] Error 1
make[4]: *** [Makefile:2012: /home/yoki/lede/build_dir/target-x86_64_musl/linux-x86_64/qmi_wwan_f-1.0] Error 2
make[4]: Leaving directory '/home/yoki/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.35'
make[3]: *** [Makefile:50: /home/yoki/lede/build_dir/target-x86_64_musl/linux-x86_64/qmi_wwan_f-1.0/.built] Error 2
make[3]: Leaving directory '/home/yoki/lede/package/wwan/driver/fibocom_QMI_WWAN'
time: package/wwan/driver/fibocom_QMI_WWAN/compile#0.59#1.07#2.38
    ERROR: package/wwan/driver/fibocom_QMI_WWAN failed to build.
make[2]: *** [package/Makefile:116: package/wwan/driver/fibocom_QMI_WWAN/compile] Error 1
make[2]: Leaving directory '/home/yoki/lede'
make[1]: *** [package/Makefile:110: /home/yoki/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/yoki/lede'
make: *** [/home/yoki/lede/include/toplevel.mk:231: world] Error 2

WYC-2020, 2023-07-09:
[qmi_wwan_f.zip](https://github.com/coolsnowwolf/lede/files/11995269/qmi_wwan_f.zip)
替换/lede/package/wwan/driver/fibocom_QMI_WWAN/src

ChaingTsung, 2023-08-31:
ok

commit 5ae20da7 (2023-09-06): fibocom_qmi: fixes build error (#11516)

Fixed #11374
closed 2023-09-06 by aiamadeus