coolsnowwolf/lede #11375: 6.1.38编译失败

GreatMichaelLee, 2023-07-09:
ERROR: module '/home/builder/lede_x86/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.38/fs/smbfs_common/cifs_arc4.ko' is missing.

lxg2001, 2023-07-09:
昨天我刚解决，从6.1.36后，文件路径改了，把package\kernel\linux\modules\fs.mk，把“fs/cifs”换成“fs/smb/client”，“fs/smbfs_common” 换成“fs/smb/common”
`sed -i 's#fs/cifs#fs/smb/client#g' package/kernel/linux/modules/fs.mk`
`sed -i 's#fs/ksmbd#fs/smb/server#g' package/kernel/linux/modules/fs.mk`
`sed -i 's#fs/smbfs_common#fs/smb/common#g' package/kernel/linux/modules/fs.mk`

WYC-2020, 2023-07-09:
这种改发之适应6.1内核，其他内核不能这样 
目前这样改
[image]
有能力的 你们可以改成内核打补丁的方式，但是我懒不想打补丁

GreatMichaelLee, 2023-07-09:
大佬哪里懒，大佬动动小手我们就觉得你是世界上最勤奋的人:)

WYC-2020, 2023-07-10:
> 大佬哪里懒，大佬动动小手我们就觉得你是世界上最勤奋的人:)
> […](#)

少拍马屁，多做事，确实我懒 你们可以给内核打补丁改成一样的这样就用改这里

commit 92265caa (2023-07-10): kernel: modules: fs: adapt for kernel 6.1.36

Acknowledgment to john-tho for the changes to fs.mk to accommodate new paths
introduced in https://github.com/gregkh/linux/commit/29429a1f5871dbe54ee0da81bb12db8567f15379

Fixes: #11375
closed 2023-07-10 by aiamadeus