coolsnowwolf/lede #11376: R68S频繁断流，无法进入后台，只能拔电源重启

AlephZHAO, 2023-07-09:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：R68S频繁断流，无法进入路由器后台；只有重启后才恢复正常
使用的插件：动态DNS，AdguardHome，KMS服务器，UPNP
### (2) 路由器型号和固件版本

A：电犀牛R68S（瑞芯微RK3568）；固件`OpenWrt R23.2.14 / LuCI Master (git-23.070.53221-d1e7a12)`

### (3) 详细日志

A：问题发生时，系统日志：
```
Sun Jul  9 19:57:11 2023 daemon.info pppd[4560]: syncppp not active
Sun Jul  9 19:57:11 2023 daemon.info pppd[4560]: Remote message: Authentication failed!
Sun Jul  9 19:57:11 2023 daemon.err pppd[4560]: PAP authentication failed
Sun Jul  9 19:57:11 2023 daemon.notice pppd[4560]: Connection terminated.
Sun Jul  9 19:57:11 2023 daemon.info pppd[4560]: Exit.
Sun Jul  9 19:57:11 2023 daemon.notice netifd: Interface 'WAN' is now down
Sun Jul  9 19:57:11 2023 daemon.info avahi-daemon[3925]: Interface eth3.IPv6 no longer relevant for mDNS.
Sun Jul  9 19:57:11 2023 daemon.info avahi-daemon[3925]: Leaving mDNS multicast group on interface eth3.IPv6 with address fe80::220:18ff:fea8:8dab.
Sun Jul  9 19:57:11 2023 daemon.info avahi-daemon[3925]: Withdrawing address record for fe80::220:18ff:fea8:8dab on eth3.
Sun Jul  9 19:57:11 2023 daemon.notice netifd: Interface 'WAN' is disabled
Sun Jul  9 19:57:11 2023 kern.info kernel: [   33.324524] eth3: 0xffffffc00a800000, 00:20:18:a8:8d:ab, IRQ 25
Sun Jul  9 19:57:11 2023 daemon.notice netifd: Interface 'WAN' is enabled
Sun Jul  9 19:57:11 2023 daemon.notice netifd: Interface 'WAN' is setting up now
Sun Jul  9 19:57:11 2023 daemon.notice netifd: Network device 'eth3' link is down
Sun Jul  9 19:57:11 2023 daemon.notice netifd: Interface 'WAN' has link connectivity loss
Sun Jul  9 19:57:12 2023 daemon.notice netifd: Interface 'WAN' is now down
Sun Jul  9 19:57:12 2023 daemon.notice netifd: Interface 'WAN' is disabled
Sun Jul  9 19:57:12 2023 kern.info kernel: [   33.440102] eth3: 0xffffffc00a800000, 00:20:18:a8:8d:ab, IRQ 25
Sun Jul  9 19:57:12 2023 daemon.notice netifd: Interface 'WAN' is enabled
Sun Jul  9 19:57:12 2023 kern.info kernel: [   33.778698] vccio_sd: disabling
Sun Jul  9 19:57:12 2023 kern.info kernel: [   33.780703] vcc3v3_sd: disabling
Sun Jul  9 19:57:13 2023 daemon.notice procd: /etc/rc.d/S91wrtbwmon: Warning: using existing /tmp/usage.db
Sun Jul  9 19:57:13 2023 daemon.notice procd: /etc/rc.d/S91wrtbwmon: Warning: failed to detect WAN interface.
Sun Jul  9 19:57:13 2023 daemon.notice procd: /etc/rc.d/S91wrtbwmon: Warning: failed to detect WAN interface.
Sun Jul  9 19:57:14 2023 user.notice simple-adblock [6975]: WARNING: Some recommended packages are missing, install them by running:
Sun Jul  9 19:57:14 2023 user.notice simple-adblock [6975]: opkg update; opkg --force-overwrite install coreutils-sort;
Sun Jul  9 19:57:14 2023 daemon.notice netifd: Network device 'eth3' link is up
Sun Jul  9 19:57:14 2023 daemon.notice netifd: Interface 'WAN' has link connectivity
Sun Jul  9 19:57:14 2023 daemon.notice netifd: Interface 'WAN' is setting up now
Sun Jul  9 19:57:14 2023 kern.info kernel: [   36.293319] r8125: eth3: link up
Sun Jul  9 19:57:14 2023 kern.info kernel: [   36.293778] IPv6: ADDRCONF(NETDEV_CHANGE): eth3: link becomes ready
Sun Jul  9 19:57:14 2023 daemon.notice netifd: WAN (7077): uci: Entry not found
Sun Jul  9 19:57:14 2023 daemon.notice netifd: WAN (7077): sh: out of range
Sun Jul  9 19:57:14 2023 daemon.info pppd[7084]: Plugin pppoe.so loaded.
Sun Jul  9 19:57:14 2023 daemon.info pppd[7084]: PPPoE plugin from pppd 2.4.9
Sun Jul  9 19:57:15 2023 daemon.notice pppd[7084]: pppd 2.4.9 started by root, uid 0
Sun Jul  9 19:57:16 2023 daemon.info avahi-daemon[3925]: Joining mDNS multicast group on interface eth3.IPv6 with address fe80::220:18ff:fea8:8dab.
Sun Jul  9 19:57:16 2023 daemon.info avahi-daemon[3925]: New relevant interface eth3.IPv6 for mDNS.
Sun Jul  9 19:57:16 2023 daemon.info avahi-daemon[3925]: Registering new address record for fe80::220:18ff:fea8:8dab on eth3.*.
Sun Jul  9 19:57:17 2023 user.notice firewall: Reloading firewall due to ifup of lan (br-lan)
Sun Jul  9 19:57:20 2023 daemon.info pppd[7084]: PPP session is 61986
Sun Jul  9 19:57:20 2023 daemon.warn pppd[7084]: Connected to cc:1a:fa:e8:42:20 via interface eth3
Sun Jul  9 19:57:20 2023 kern.info kernel: [   41.499343] pppoe-WAN: renamed from ppp0
Sun Jul  9 19:57:20 2023 daemon.info pppd[7084]: Renamed interface ppp0 to pppoe-WAN
Sun Jul  9 19:57:20 2023 daemon.info pppd[7084]: Using interface pppoe-WAN
Sun Jul  9 19:57:20 2023 daemon.notice pppd[7084]: Connect: pppoe-WAN <--> eth3
Sun Jul  9 19:57:20 2023 daemon.info pppd[7084]: syncppp not active
Sun Jul  9 19:57:20 2023 daemon.info pppd[7084]: Remote message: Authentication failed!
Sun Jul  9 19:57:20 2023 daemon.err pppd[7084]: PAP authentication failed
Sun Jul  9 19:57:20 2023 daemon.notice pppd[7084]: Connection terminated.
Sun Jul  9 19:57:20 2023 daemon.info pppd[7084]: Exit.
[... 163 lines cut ...]
Sun Jul  9 19:57:54 2023 daemon.info avahi-daemon[3925]: Joining mDNS multicast group on interface pppoe-WAN.IPv4 with address 218.81.184.29.
Sun Jul  9 19:57:54 2023 daemon.notice pppd[10915]: local  IP address 218.81.184.29
Sun Jul  9 19:57:54 2023 daemon.notice pppd[10915]: remote IP address 218.81.184.1
Sun Jul  9 19:57:54 2023 daemon.notice pppd[10915]: primary   DNS address 180.168.255.118
Sun Jul  9 19:57:54 2023 daemon.notice pppd[10915]: secondary DNS address 116.228.111.18
Sun Jul  9 19:57:54 2023 daemon.info avahi-daemon[3925]: New relevant interface pppoe-WAN.IPv4 for mDNS.
Sun Jul  9 19:57:54 2023 daemon.info avahi-daemon[3925]: Registering new address record for 218.81.184.29 on pppoe-WAN.IPv4.
Sun Jul  9 19:57:54 2023 daemon.info avahi-daemon[3925]: Joining mDNS multicast group on interface pppoe-WAN.IPv6 with address fe80::220:1808:90a8:8dab.
Sun Jul  9 19:57:54 2023 daemon.info avahi-daemon[3925]: New relevant interface pppoe-WAN.IPv6 for mDNS.
Sun Jul  9 19:57:54 2023 daemon.notice pppd[10915]: local  LL address fe80::0220:1808:90a8:8dab
Sun Jul  9 19:57:54 2023 daemon.info avahi-daemon[3925]: Registering new address record for fe80::220:1808:90a8:8dab on pppoe-WAN.*.
Sun Jul  9 19:57:54 2023 daemon.notice pppd[10915]: remote LL address fe80::ce1a:faff:fee8:4220
Sun Jul  9 19:57:54 2023 daemon.notice netifd: Network device 'pppoe-WAN' link is up
Sun Jul  9 19:57:54 2023 daemon.notice netifd: Interface 'WAN' is now up
Sun Jul  9 19:57:54 2023 daemon.notice netifd: Network alias 'pppoe-WAN' link is up
Sun Jul  9 19:57:54 2023 daemon.notice netifd: Interface 'WAN_6' is enabled
Sun Jul  9 19:57:54 2023 daemon.notice netifd: Interface 'WAN_6' has link connectivity
Sun Jul  9 19:57:54 2023 daemon.notice netifd: Interface 'WAN_6' is setting up now
Sun Jul  9 19:57:56 2023 user.notice root: starting ntpclient
Sun Jul  9 19:57:56 2023 user.notice nlbwmon: Reloading nlbwmon due to ifup of loopback (lo)
Sun Jul  9 20:00:55 2023 daemon.notice miniupnpd[8221]: shutting down MiniUPnPd
Sun Jul  9 20:00:55 2023 daemon.err miniupnpd[11384]: could not open lease file: /var/upnp.leases
Sun Jul  9 20:00:55 2023 daemon.notice miniupnpd[11384]: HTTP listening on port 5000
Sun Jul  9 20:00:55 2023 daemon.notice miniupnpd[11384]: Listening for NAT-PMP/PCP traffic on port 5351
```

AlephZHAO, 2023-07-09:
因字数限制，补一下内核日志：
```
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x412fd050]
[    0.000000] Linux version 6.1.18-flippy-82+ (root@vm25) (Ubuntu clang version 14.0.0-1ubuntu1, Ubuntu LLD 14.0.0) #50 SMP Sat Mar 11 22:59:20 CST 2023
[    0.000000] Machine model: FastRhino R68S
[    0.000000] earlycon: uart8250 at MMIO32 0x00000000fe660000 (options '')
[    0.000000] printk: bootconsole [uart8250] enabled
[    0.000000] efi: UEFI not found.
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000200000-0x00000000ffffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   [mem 0x0000000100000000-0x00000001ffffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000200000-0x00000000083fffff]
[    0.000000]   node   0: [mem 0x0000000009400000-0x00000000efffffff]
[    0.000000]   node   0: [mem 0x00000001f0000000-0x00000001ffffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000200000-0x00000001ffffffff]
[    0.000000] On node 0, zone DMA: 512 pages in unavailable ranges
[    0.000000] On node 0, zone DMA: 4096 pages in unavailable ranges
[    0.000000] cma: Reserved 256 MiB at 0x00000000e0000000
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: Trusted OS migration not required
[    0.000000] psci: SMC Calling Convention v1.2
[    0.000000] percpu: Embedded 19 pages/cpu s37992 r8192 d31640 u77824
[    0.000000] pcpu-alloc: s37992 r8192 d31640 u77824 alloc=19*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: GIC system register CPU interface
[    0.000000] CPU features: detected: Virtualization Host Extensions
[    0.000000] CPU features: detected: Qualcomm erratum 1009, or ARM erratum 1286807, 2441009
[    0.000000] CPU features: detected: ARM errata 1165522, 1319367, or 1530923
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 1027656
[    0.000000] Kernel command line: root=UUID=309f4eb1-60d5-43c2-9121-e2a192fbcb86 rootwait rootfstype=btrfs rootflags=compress=zstd:6 bootsplash.bootfile=bootsplash.armbian earlycon=uart8250,mmio32,0xfe660000 console=ttyS2,1500000 console=tty1 consoleblank=0 loglevel=7 usb-storage.quirks=0x2537:0x1066:u,0x2537:0x1068:u  net.ifnames=0 cgroup_enable=cpuset cgroup_memory=1 cgroup_enable=memory swapaccount=1
[    0.000000] The swapaccount= commandline option is deprecated. Please report your usecase to linux-mm@kvack.org if you depend on this functionality.
[    0.000000] Unknown kernel command line parameters "cgroup_enable=memory cgroup_memory=1", will be passed to user space.
[    0.000000] Dentry cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[    0.000000] Inode-cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: area num 4.
[    0.000000] software IO TLB: mapped [mem 0x00000000dc000000-0x00000000e0000000] (64MB)
[    0.000000] Memory: 3735424K/4175872K available (17024K kernel code, 2908K rwdata, 7728K rodata, 2240K init, 888K bss, 178304K reserved, 262144K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=8 to nr_cpu_ids=4.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GICv3: GIC: Using split EOI/Deactivate mode
[    0.000000] GICv3: 320 SPIs implemented
[    0.000000] GICv3: 0 Extended SPIs implemented
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GICv3: GICv3 features: 16 PPIs
[    0.000000] GICv3: CPU0: found redistributor 0 region 0:0x00000000fd460000
[    0.000000] ITS [mem 0xfd440000-0xfd45ffff]
[    0.000000] GIC: enabling workaround for ITS: Rockchip erratum 114514
[    0.000000] ITS@0x00000000fd440000: allocated 8192 Devices @2410000 (indirect, esz 8, psz 64K, shr 0)
[    0.000000] ITS@0x00000000fd440000: allocated 32768 Interrupt Collections @2420000 (flat, esz 2, psz 64K, shr 0)
[    0.000000] ITS: using cache flushing for cmd queue
[    0.000000] ITS ALLOCATE PROP WORKAROUND
[    0.000000] GICv3: using LPI property table @0x0000000002430000
[    0.000000] GIC: using cache flushing for LPI property table
[    0.000000] GICv3: CPU0: using allocated LPI pending table @0x0000000002440000
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] arch_timer: cp15 timer(s) running at 24.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x588fe9dc0, max_idle_ns: 440795202592 ns
[... 633 lines cut ...]
[   61.609633] IPv6: ADDRCONF(NETDEV_CHANGE): eth3: link becomes ready
[   66.807298] pppoe-WAN: renamed from ppp0
[   67.031423] eth3: 0xffffffc00a800000, 00:20:18:a8:8d:ab, IRQ 25
[   67.137385] eth3: 0xffffffc00a800000, 00:20:18:a8:8d:ab, IRQ 25
[   70.101310] r8125: eth3: link up
[... cut ...]
[   97.422901] br-lan: port 1(eth0) entered disabled state
[  100.496931] rk_gmac-dwmac fe2a0000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[  100.498083] br-lan: port 1(eth0) entered blocking state
[  100.498606] br-lan: port 1(eth0) entered forwarding state
[  182.385727] NFSD: Using /var/lib/nfs/v4recovery as the NFSv4 state recovery directory
[  182.386474] NFSD: Using legacy client tracking operations.
[  182.386971] NFSD: starting 10-second grace period (net f0000000)

```

AlephZHAO, 2023-07-09:
因字数限制，补全系统日志：
```
Sun Jul  9 20:00:56 2023 daemon.info avahi-daemon[3925]: Leaving mDNS multicast group on interface pppoe-WAN.IPv6 with address fe80::220:1808:90a8:8dab.
Sun Jul  9 20:00:56 2023 daemon.info avahi-daemon[3925]: Joining mDNS multicast group on interface pppoe-WAN.IPv6 with address 240e:38f:3303:514:220:1808:90a8:8dab.
Sun Jul  9 20:00:56 2023 daemon.info avahi-daemon[3925]: Registering new address record for 240e:38f:3303:514:220:1808:90a8:8dab on pppoe-WAN.*.
Sun Jul  9 20:00:56 2023 daemon.info avahi-daemon[3925]: Withdrawing address record for fe80::220:1808:90a8:8dab on pppoe-WAN.
Sun Jul  9 20:00:56 2023 daemon.info avahi-daemon[3925]: Leaving mDNS multicast group on interface br-lan.IPv6 with address fe80::220:18ff:fea8:8daa.
Sun Jul  9 20:00:56 2023 daemon.info avahi-daemon[3925]: Joining mDNS multicast group on interface br-lan.IPv6 with address 240e:388:2802:7400::1.
Sun Jul  9 20:00:56 2023 daemon.info avahi-daemon[3925]: Registering new address record for 240e:388:2802:7400::1 on br-lan.*.
Sun Jul  9 20:00:56 2023 daemon.info avahi-daemon[3925]: Withdrawing address record for fe80::220:18ff:fea8:8daa on br-lan.
Sun Jul  9 20:00:56 2023 daemon.notice netifd: Interface 'WAN_6' is now up
Sun Jul  9 20:01:02 2023 daemon.notice netifd: Network device 'eth0' link is up
Sun Jul  9 20:01:02 2023 kern.info kernel: [   85.137370] rk_gmac-dwmac fe2a0000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
Sun Jul  9 20:01:02 2023 kern.info kernel: [   85.138497] br-lan: port 1(eth0) entered blocking state
Sun Jul  9 20:01:02 2023 kern.info kernel: [   85.139020] br-lan: port 1(eth0) entered forwarding state
Sun Jul  9 20:01:04 2023 daemon.err collectd[4235]: Sleeping only 2s because the next interval is 148.552 seconds in the past!
Sun Jul  9 20:01:05 2023 daemon.err collectd[4235]: rrdtool plugin: rrd_update_r failed: /tmp/rrd/AlephWRT/cpu-1/cpu-idle.rrd: opening '/tmp/rrd/AlephWRT/cpu-1/cpu-idle.rrd': No such file or directory
Sun Jul  9 20:01:06 2023 user.notice firewall: Reloading firewall due to ifup of WAN (pppoe-WAN)
Sun Jul  9 20:01:07 2023 daemon.notice netifd: Network device 'eth0' link is down
Sun Jul  9 20:01:07 2023 kern.info kernel: [   90.253758] rk_gmac-dwmac fe2a0000.ethernet eth0: Link is Down
Sun Jul  9 20:01:07 2023 kern.info kernel: [   90.254456] br-lan: port 1(eth0) entered disabled state
Sun Jul  9 20:01:07 2023 user.notice nlbwmon: Reloading nlbwmon due to ifup of WAN (pppoe-WAN)
Sun Jul  9 20:01:08 2023 user.notice ddns-scripts[12730]: DNSPODV4: PID '12730' started at 2023-07-09 20:01
Sun Jul  9 20:01:10 2023 kern.info kernel: [   93.328344] rk_gmac-dwmac fe2a0000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
Sun Jul  9 20:01:10 2023 kern.info kernel: [   93.329241] br-lan: port 1(eth0) entered blocking state
Sun Jul  9 20:01:10 2023 daemon.notice netifd: Network device 'eth0' link is up
Sun Jul  9 20:01:10 2023 kern.info kernel: [   93.329784] br-lan: port 1(eth0) entered forwarding state
Sun Jul  9 20:01:11 2023 user.warn ddns-scripts[12730]: DNSPODV4: NO valid IP found
Sun Jul  9 20:01:14 2023 daemon.notice netifd: Network device 'eth0' link is down
Sun Jul  9 20:01:14 2023 kern.info kernel: [   97.421891] rk_gmac-dwmac fe2a0000.ethernet eth0: Link is Down
Sun Jul  9 20:01:14 2023 kern.info kernel: [   97.422901] br-lan: port 1(eth0) entered disabled state
Sun Jul  9 20:01:17 2023 daemon.notice netifd: Network device 'eth0' link is up
Sun Jul  9 20:01:17 2023 kern.info kernel: [  100.496931] rk_gmac-dwmac fe2a0000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
Sun Jul  9 20:01:17 2023 kern.info kernel: [  100.498083] br-lan: port 1(eth0) entered blocking state
Sun Jul  9 20:01:17 2023 kern.info kernel: [  100.498606] br-lan: port 1(eth0) entered forwarding state
Sun Jul  9 20:01:19 2023 user.notice firewall: Reloading firewall due to ifup of WAN_6 (pppoe-WAN)
Sun Jul  9 20:01:20 2023 user.notice nlbwmon: Reloading nlbwmon due to ifup of WAN_6 (pppoe-WAN)
Sun Jul  9 20:01:44 2023 user.warn ddns-scripts[8333]: DNSPODV6: NO valid IP found
Sun Jul  9 20:01:44 2023 user.warn ddns-scripts[8333]: DNSPODV6: Get registered/public IP for 'xzxz.space' failed - retry 2/0 in 60 seconds
Sun Jul  9 20:02:12 2023 user.notice simple-adblock [6975]: WARNING: Some recommended packages are missing, install them by running:
Sun Jul  9 20:02:12 2023 user.notice simple-adblock [6975]: opkg update; opkg --force-overwrite install coreutils-sort;
Sun Jul  9 20:02:12 2023 user.notice simple-adblock [6975]: simple-adblock is currently disabled.
Sun Jul  9 20:02:12 2023 user.notice simple-adblock [6975]: Run the following commands before starting service again:
Sun Jul  9 20:02:12 2023 user.notice simple-adblock [6975]: uci set simple-adblock.config.enabled='1'; uci commit simple-adblock;
Sun Jul  9 20:02:12 2023 user.notice vpnbypass [18218]: service monitoring interfaces: WAN
Sun Jul  9 20:02:13 2023 daemon.notice procd: /etc/rc.d/S95AdGuardHome: 文件系统: btrfs
Sun Jul  9 20:02:13 2023 daemon.notice procd: /etc/rc.d/S95AdGuardHome: AdGuardHome 服务已启用
Sun Jul  9 20:02:13 2023 daemon.notice procd: /etc/rc.d/S95done: irq name:eth0, irq:56, affinity: 1
Sun Jul  9 20:02:13 2023 daemon.notice procd: /etc/rc.d/S95done: irq name:eth1, irq:54, affinity: 2
Sun Jul  9 20:02:13 2023 daemon.notice procd: /etc/rc.d/S95done: irq name:eth2-0, irq:126, affinity: 4
Sun Jul  9 20:02:13 2023 daemon.notice procd: /etc/rc.d/S95done: irq name:eth2-16, irq:142, affinity: 2
[... cut ...]
Sun Jul  9 20:02:47 2023 cron.err crond[26061]: crond (busybox 1.36.0) started, log level 8
Sun Jul  9 20:02:48 2023 user.notice vpnbypass [26264]: service monitoring interfaces: WAN
Sun Jul  9 20:02:52 2023 daemon.notice procd: /etc/rc.d/S99zerotier: disabled in config
Sun Jul  9 20:02:53 2023 daemon.notice procd: /etc/rc.d/S99zzz_passwall: grep: warning: stray \ before /
Sun Jul  9 20:02:54 2023 daemon.notice procd: /etc/rc.d/S99zzz_passwall: grep: warning: stray \ before /
Sun Jul  9 20:02:54 2023 cron.err crond[27462]: crond (busybox 1.36.0) started, log level 8

```

AlephZHAO, 2023-08-10:
更换新的固件后，OpenWRT稳定运行2个多月，问题已解决！

closed 2023-08-10 by AlephZHAO

haidaowenshu, 2023-08-11:
> 更换新的固件后，OpenWRT稳定运行2个多月，问题已解决！

能推荐一下固件的版本号吗？

AlephZHAO, 2023-08-29:
> > 更换新的固件后，OpenWRT稳定运行2个多月，问题已解决！
> 
> 能推荐一下固件的版本号吗？

当然可以，目前我使用的是比较新的内核和系统
固件版本 `bleach mod 20230727 @ OpenWrt R23.7.7 / LuCI Master (git-23.198.59743-991daf5)`
内核版本 `6.1.41`
附上链接 `https://www.right.com.cn/forum/thread-8293464-1-1.html`
**（注意，这个是可适用于电犀牛R68S的固件，其他设备适配情况不详）**
这里有很多OpenWRT编译完成的固件可供下载 `https://openwrt.mpdn.fun:8443/`

haidaowenshu, 2023-08-29:
谢谢你的回复，万分感谢