coolsnowwolf/lede #11377: 编译r4se固件出错

magi-Z, 2023-07-09:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency libiptext for firewall
 * pkg_hash_check_unresolved: cannot find dependency libiptext6 for firewall
 * pkg_hash_fetch_best_installation_candidate: Packages for firewall found, but incompatible with the architectures configured
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for default-settings:
 * 	libiptext
 * 	libiptext6
 * opkg_install_cmd: Cannot install package default-settings.
make[2]: *** [package/Makefile:70: package/install] Error 255
make[2]: Leaving directory '/home/magi/lede'
make[1]: *** [package/Makefile:111: /home/magi/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/magi/lede'
make: *** [/home/magi/lede/include/toplevel.mk:231：world] 错误 2

magi-Z, 2023-07-10:
WARNING: Makefile 'package/small-package/firewall/Makefile' has a dependency on 'libiptext', which does not exist
WARNING: Makefile 'package/small-package/firewall/Makefile' has a dependency on 'libiptext6', which does not exist
WARNING: Makefile 'package/small-package/transmission-web-control/Makefile' has a dependency on 'transmission-daemon', which does not exist
tmp/.config-package.in:82909:error: recursive dependency detected!
这个依赖总是如何解决！不兼容r4se构架

WYC-2020, 2023-07-10:
谁的仓库找谁去

closed 2023-07-24 by aiamadeus