coolsnowwolf/lede #11378: Softether 5 IPV6服务器无法连接

31551, 2023-07-10:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
Softether 5 IPV6服务器无法连接
Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：电信宽带无IPV4公网地址，有IPV6地址。ddns正确获取了ipv6地址，但openVPN无法连接softether服务器。
softether服务器设置为tap模式。直接桥接到lan接口

### (2) 路由器型号和固件版本

A：X86路由器 OpenWrt R23.7.7 / LuCI Master (git-23.177.50322-72213cf)

### (3) 详细日志

A：
[Jul 10, 2023, 17:37:49] START CONNECTION

[Jul 10, 2023, 17:37:49] ----- OpenVPN Start -----
OpenVPN core 3.git::081bfebe ios arm64 64-bit

[Jul 10, 2023, 17:37:49] OpenVPN core 3.git::081bfebe ios arm64 64-bit

[Jul 10, 2023, 17:37:49] Frame=512/2048/512 mssfix-ctrl=1250

[Jul 10, 2023, 17:37:49] UNUSED OPTIONS
6 [resolv-retry] [infinite]
7 [nobind]
8 [persist-key]
9 [persist-tun]
11 [verb] [3]

[Jul 10, 2023, 17:37:49] EVENT: RESOLVE

[Jul 10, 2023, 17:37:50] Contacting [240e:3a1:4a43:40d0::1]:1194 via UDP

[Jul 10, 2023, 17:37:50] EVENT: WAIT

[Jul 10, 2023, 17:37:50] Connecting to [suzhou.xxxxxxxxx.cn]:1194 (240e:3a1:4a43:40d0::1) via UDPv6

[Jul 10, 2023, 17:37:59] Server poll timeout, trying next remote entry...

[Jul 10, 2023, 17:37:59] EVENT: RECONNECTING

[Jul 10, 2023, 17:37:59] EVENT: RESOLVE

[Jul 10, 2023, 17:37:59] Contacting [240e:3a1:4a43:40d0::1]:1194 via UDP

[Jul 10, 2023, 17:37:59] EVENT: WAIT

[Jul 10, 2023, 17:37:59] Connecting to [suzhou.xxxxxxxxx.cn]:1194 (240e:3a1:4a43:40d0::1) via UDPv6

[Jul 10, 2023, 17:38:09] Server poll timeout, trying next remote entry...

[Jul 10, 2023, 17:38:09] EVENT: RECONNECTING

[Jul 10, 2023, 17:38:09] EVENT: RESOLVE

[Jul 10, 2023, 17:38:09] Contacting [240e:3a1:4a43:40d0::1]:1194 via UDP

[Jul 10, 2023, 17:38:09] EVENT: WAIT

[Jul 10, 2023, 17:38:09] Connecting to [suzhou.xxxxxxxxx.cn]:1194 (240e:3a1:4a43:40d0::1) via UDPv6

[Jul 10, 2023, 17:38:19] Server poll timeout, trying next remote entry...

[Jul 10, 2023, 17:38:19] EVENT: RECONNECTING

[Jul 10, 2023, 17:38:19] EVENT: RESOLVE

[Jul 10, 2023, 17:38:19] Contacting [240e:3a1:4a43:40d0::1]:1194 via UDP

[Jul 10, 2023, 17:38:19] EVENT: WAIT

[Jul 10, 2023, 17:38:19] Connecting to [suzhou.xxxxxxxxx.cn]:1194 (240e:3a1:4a43:40d0::1) via UDPv6

[Jul 10, 2023, 17:38:29] Server poll timeout, trying next remote entry...

[Jul 10, 2023, 17:38:29] EVENT: RECONNECTING

[Jul 10, 2023, 17:38:29] EVENT: RESOLVE

[Jul 10, 2023, 17:38:29] Contacting [240e:3a1:4a43:40d0::1]:1194 via UDP

[Jul 10, 2023, 17:38:29] EVENT: WAIT

[Jul 10, 2023, 17:38:29] Connecting to [suzhou.xxxxxxxxx.cn]:1194 (240e:3a1:4a43:40d0::1) via UDPv6

[Jul 10, 2023, 17:38:39] Server poll timeout, trying next remote entry...

[Jul 10, 2023, 17:38:39] EVENT: RECONNECTING

[Jul 10, 2023, 17:38:39] EVENT: RESOLVE

[Jul 10, 2023, 17:38:39] Contacting [240e:3a1:4a43:40d0::1]:1194 via UDP

[Jul 10, 2023, 17:38:39] EVENT: WAIT

[Jul 10, 2023, 17:38:39] Connecting to [suzhou.xxxxxxxxx.cn]:1194 (240e:3a1:4a43:40d0::1) via UDPv6

[Jul 10, 2023, 17:38:49] EVENT: DISCONNECTED

[Jul 10, 2023, 17:38:49] EVENT: CORE_THREAD_DONE

[Jul 10, 2023, 17:38:49] EVENT: DISCONNECT_PENDING

[Jul 10, 2023, 17:38:49] Raw stats on disconnect:
  BYTES_OUT : 840
  PACKETS_OUT : 60
  N_RECONNECT : 5

[Jul 10, 2023, 17:38:49] Performance stats on disconnect:
  CPU usage (microseconds): 104965
  Network bytes per CPU second: 8002
  Tunnel bytes per CPU second: 0

WYC-2020, 2023-07-11:
这个好像前端时间 也有说这个有问题的，当时是升级或者降级就可以了，我是升级解决的 ，因为官方没有打包了 所以自己克隆过来 自己打包 然后修改升级到 自己这个版本就行了,看这个我以前的答复
https://github.com/coolsnowwolf/lede/issues/10520
这个是主张降级的
https://github.com/coolsnowwolf/lede/issues/10386

31551, 2023-07-11:
谢谢，我看到您说软件打包的问题啦。回去重新打包试试👍

31551, 2023-07-12:
我的版本为5.01.9674-1的。奇怪，怎么有的小伙伴编译的版本是5.02.5180-3？是自己打包的吧？
[image]

WYC-2020, 2023-07-12:
那是他们 没有回滚而已 
[image]
就是因为5.02有问题才回滚 5.01的，你已经是5.01了还有问题？

31551, 2023-07-12:
是的，IPV6服务器连不上。苏州电信不给公网IP，服务器只能用V6地址了，现在怎么都连不上。softether设置的tap模式，直接桥接到lan口下的。现在用北京的服务器做实验，服务器V4地址可以连并且客户端获取的V4地址就是内网的地址，V6服务器搞不上去。

WYC-2020, 2023-07-12:
> 是的，IPV6服务器连不上。苏州电信不给公网IP，服务器只能用V6地址了，现在怎么都连不上。softether设置的tap模式，直接桥接到lan口下的。现在用北京的服务器做实验，服务器V4地址可以连并且客户端获取的V4地址就是内网的地址，V6服务器搞不上去。

[softethervpn5.zip](https://github.com/coolsnowwolf/lede/files/12023241/softethervpn5.zip)
用我这个 直接替换 原来的 在编译一个 试试

31551, 2023-07-12:
好的 我试试如何 晚点反馈

31551, 2023-07-12:
编译没有成功😂

31551, 2023-07-12:
奇怪了，刚才突然就连上了。看来不是服务器软件的问题，应该是网络的问题。 此问题可以关闭了，哈哈。谢谢[WYC-2020](https://github.com/WYC-2020)的热心帮助。

[Jul 12, 2023, 17:48:49] PROTOCOL OPTIONS:
  cipher: AES-128-CBC
  digest: SHA1
  key-derivation: OpenVPN PRF
  compress: NONE
  peer ID: -1

[Jul 12, 2023, 17:48:49] EVENT: ASSIGN_IP

[Jul 12, 2023, 17:48:49] NIP: preparing TUN network settings

[Jul 12, 2023, 17:48:49] NIP: init TUN network settings with endpoint: 2408:8207:xxxx:1ea0::1

[Jul 12, 2023, 17:48:49] NIP: adding IPv4 address to network settings 10.0.0.221/255.255.255.0

[Jul 12, 2023, 17:48:49] NIP: adding (included) IPv4 route 10.0.0.0/24

[Jul 12, 2023, 17:48:49] NIP: redirecting all IPv4 traffic to TUN interface

[Jul 12, 2023, 17:48:49] NIP: adding match domain lan

[Jul 12, 2023, 17:48:49] NIP: adding DNS 180.76.76.76

[Jul 12, 2023, 17:48:49] NIP: adding DNS 223.5.5.5

[Jul 12, 2023, 17:48:49] Connected via NetworkExtensionTUN

[Jul 12, 2023, 17:48:49] EVENT: CONNECTED charles-mobile@beijing.xxxx.cn:1194 (2408:8207:xxxx:1ea0::1) via /UDPv6 on NetworkExtensionTUN/10.0.0.221/ gw=[/]

closed 2023-07-12 by 31551

31551, 2023-07-12:
我想起来了，我是导出了softether服务器的配置到文件，然后编辑了导出的配置文件。
本来配置文件里只有 string ListenIP 0.0.0.0 这一项，我在这行下面给它加了一行：string ListenIP ::，重启服务后好像就可以了。