coolsnowwolf/lede #11387: last version X86-64 build with xr-usb-serial module fail

668168, 2023-07-16:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
last version X86-64 build with xr-usb-serial module fail

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
+ 参考如下guide [^1]增加4G 模块时,增加了xr-usb-serial 模块,导致build fail.
+ 增加前build pass, 移除后build pass

[^1]: https://oldwiki.archive.openwrt.org/doc/howtobuild/wireless-router-with-a-3g-dongle

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：X86-64

X86-64 R23.7.7 / LuCI Master (git-23.194.57997-26d16e6)

### (3) 详细日志

A：

CONFIG_PACKAGE_kmod-xr-usb-serial=y

```
/home/x/work/lede/build_dir/target-x86_64_musl/linux-x86_64/symvers/fullconenat.symvers /home/x/work/lede/build_dir/target-x86_64_musl/linux-x86_64/symvers/ksmbd.symvers /home/x/work/lede/build_dir/target-x86_64_musl/linux-x86_64/symvers/shortcut-fe.symvers" KERNELRELEASE=6.1.38 M="/home/x/work/lede/build_dir/target-x86_64_musl/linux-x86_64/xr-usb-serial" EXTRA_CFLAGS="" modules
make[4]: Entering directory '/home/x/work/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.38'
  CC [M]  /home/x/work/lede/build_dir/target-x86_64_musl/linux-x86_64/xr-usb-serial/xr_usb_serial_common.o
/home/x/work/lede/build_dir/target-x86_64_musl/linux-x86_64/xr-usb-serial/xr_usb_serial_common.c:2019:33: error: initialization of 'unsigned int (*)(struct tty_struct *)' from incompatible pointer type 'int (*)(struct tty_struct *)' [-Werror=incompatible-pointer-types]
 2019 |         .write_room =           xr_usb_serial_tty_write_room,
      |                                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/x/work/lede/build_dir/target-x86_64_musl/linux-x86_64/xr-usb-serial/xr_usb_serial_common.c:2019:33: note: (near initialization for 'xr_usb_serial_ops.write_room')
/home/x/work/lede/build_dir/target-x86_64_musl/linux-x86_64/xr-usb-serial/xr_usb_serial_common.c:2026:33: error: initialization of 'unsigned int (*)(struct tty_struct *)' from incompatible pointer type 'int (*)(struct tty_struct *)' [-Werror=incompatible-pointer-types]
 2026 |         .chars_in_buffer =      xr_usb_serial_tty_chars_in_buffer,
      |                                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/x/work/lede/build_dir/target-x86_64_musl/linux-x86_64/xr-usb-serial/xr_usb_serial_common.c:2026:33: note: (near initialization for 'xr_usb_serial_ops.chars_in_buffer')
/home/x/work/lede/build_dir/target-x86_64_musl/linux-x86_64/xr-usb-serial/xr_usb_serial_common.c:2028:33: error: initialization of 'void (*)(struct tty_struct *, const struct ktermios *)' from incompatible pointer type 'void (*)(struct tty_struct *, struct ktermios *)' [-Werror=incompatible-pointer-types]
 2028 |         .set_termios =          xr_usb_serial_tty_set_termios,
      |                                 ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/x/work/lede/build_dir/target-x86_64_musl/linux-x86_64/xr-usb-serial/xr_usb_serial_common.c:2028:33: note: (near initialization for 'xr_usb_serial_ops.set_termios')
/home/x/work/lede/build_dir/target-x86_64_musl/linux-x86_64/xr-usb-serial/xr_usb_serial_common.c: In function 'xr_usb_serial_init':
/home/x/work/lede/build_dir/target-x86_64_musl/linux-x86_64/xr-usb-serial/xr_usb_serial_common.c:2040:36: error: implicit declaration of function 'alloc_tty_driver' [-Werror=implicit-function-declaration]
 2040 |         xr_usb_serial_tty_driver = alloc_tty_driver(XR_USB_SERIAL_TTY_MINORS);
      |                                    ^~~~~~~~~~~~~~~~
/home/x/work/lede/build_dir/target-x86_64_musl/linux-x86_64/xr-usb-serial/xr_usb_serial_common.c:2040:34: warning: assignment to 'struct tty_driver *' from 'int' makes pointer from integer without a cast [-Wint-conversion]
 2040 |         xr_usb_serial_tty_driver = alloc_tty_driver(XR_USB_SERIAL_TTY_MINORS);
      |                                  ^
/home/x/work/lede/build_dir/target-x86_64_musl/linux-x86_64/xr-usb-serial/xr_usb_serial_common.c:2057:17: error: implicit declaration of function 'put_tty_driver' [-Werror=implicit-function-declaration]
 2057 |                 put_tty_driver(xr_usb_serial_tty_driver);
      |                 ^~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[5]: *** [scripts/Makefile.build:250: /home/x/work/lede/build_dir/target-x86_64_musl/linux-x86_64/xr-usb-serial/xr_usb_serial_common.o] Error 1
make[4]: *** [Makefile:2012: /home/x/work/lede/build_dir/target-x86_64_musl/linux-x86_64/xr-usb-serial] Error 2
make[4]: Leaving directory '/home/x/work/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.38'
make[3]: *** [Makefile:35: /home/x/work/lede/build_dir/target-x86_64_musl/linux-x86_64/xr-usb-serial/.built] Error 2
make[3]: Leaving directory '/home/x/work/lede/package/kernel/xr-usb-serial'
time: package/kernel/xr-usb-serial/compile#1.09#0.13#1.22
    ERROR: package/kernel/xr-usb-serial failed to build.
make[2]: *** [package/Makefile:116: package/kernel/xr-usb-serial/compile] Error 1
make[2]: Leaving directory '/home/x/work/lede'
make[1]: *** [package/Makefile:110: /home/x/work/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/x/work/lede'
make: *** [/home/x/work/lede/include/toplevel.mk:231: world] Error 2
```

WYC-2020, 2023-07-18:
不适配而已，等下给你适配一下

WYC-2020, 2023-07-18:
替换lede\package\kernel\xr-usb-serial\src\xr_usb_serial_common.c

[xr_usb_serial_common.zip](https://github.com/coolsnowwolf/lede/files/12075466/xr_usb_serial_common.zip)

closed 2023-07-20 by 668168