coolsnowwolf/lede #11388: 编译报错信息如下，哪位大神知道这是哪里的问题？

FL-Alan, 2023-07-16:
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.38-1-6e55f9417737728536435b7bcf69f33d) for kmod-button-hotplug
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-button-hotplug found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package kmod-button-hotplug.
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.38-1-6e55f9417737728536435b7bcf69f33d) for kmod-cryptodev
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-cryptodev found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package kmod-cryptodev.
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.38-1-6e55f9417737728536435b7bcf69f33d) for kmod-r8125
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-r8125 found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package kmod-r8125.
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.38-1-6e55f9417737728536435b7bcf69f33d) for kmod-usb-net-rtl8152-vendor
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-usb-net-rtl8152-vendor found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package kmod-usb-net-rtl8152-vendor.
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.38-1-6e55f9417737728536435b7bcf69f33d) for kmod-r8168
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-r8168 found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package kmod-r8168.
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.38-1-6e55f9417737728536435b7bcf69f33d) for kmod-shortcut-fe
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-shortcut-fe found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package kmod-shortcut-fe.
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.38-1-6e55f9417737728536435b7bcf69f33d) for kmod-shortcut-fe-cm
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-shortcut-fe-cm found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package kmod-shortcut-fe-cm.
 * pkg_hash_check_unresolved: cannot find dependency kernel (= 6.1.38-1-6e55f9417737728536435b7bcf69f33d) for kmod-ipt-fullconenat
 * pkg_hash_fetch_best_installation_candidate: Packages for kmod-ipt-fullconenat found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package kmod-ipt-fullconenat.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for ddns-scripts_aliyun:
 * 	kernel (= 6.1.38-1-6e55f9417737728536435b7bcf69f33d)
 * opkg_install_cmd: Cannot install package ddns-scripts_aliyun.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for default-settings:
 * 	kernel (= 6.1.38-1-6e55f9417737728536435b7bcf69f33d)
 * opkg_install_cmd: Cannot install package default-settings.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-turboacc:
 * 	kernel (= 6.1.38-1-6e55f9417737728536435b7bcf69f33d)
 * opkg_install_cmd: Cannot install package luci-app-turboacc.
make[2]: *** [package/Makefile:70: package/install] Error 255
make[2]: Leaving directory '/home/user/lede'
make[1]: *** [package/Makefile:111: /home/user/lede/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/user/lede'
make: *** [/home/user/lede/include/toplevel.mk:231：world] 错误 2

CYaNu, 2023-07-16:
选择的package被别的package包含了

FL-Alan, 2023-07-16:
> 选择的package被别的package包含了

谢谢！我试试删掉重新来一遍看看

WYC-2020, 2023-07-17:
明显内核不匹配

closed 2023-07-23 by aiamadeus