coolsnowwolf/lede #11390: xdr6088 每一到两天无线断网

water0012, 2023-07-16:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：每隔一到两天 无线客户端断网，表现如下：
1. 无法通过dhcp获取ip
2. 手动为客户端指定ip 无法ping通6088 lan ip
3. 有线连接的设备未断网，可以正常ping通 路由器ip。

### (2) 路由器型号和固件版本

A：xdr6088 OpenWrt R23.7.7 / LuCI Master (git-23.177.50322-72213cf)

### (3) 详细日志
系统日志
Sun Jul 16 16:21:07 2023 daemon.notice hostapd: wlan1-3: BEACON-REQ-TX-STATUS bc:d0:74:03:9d:4b 165 ack=1
Sun Jul 16 16:21:07 2023 daemon.notice hostapd: wlan1-3: BEACON-RESP-RX bc:d0:74:03:9d:4b 165 04
Sun Jul 16 16:21:07 2023 daemon.err hostapd: nl80211: kernel reports: integer out of range
Sun Jul 16 16:21:08 2023 daemon.err hostapd: nl80211: kernel reports: integer out of range
Sun Jul 16 16:21:08 2023 daemon.err hostapd: nl80211: kernel reports: integer out of range
Sun Jul 16 16:21:09 2023 daemon.err hostapd: nl80211: kernel reports: integer out of range
Sun Jul 16 16:21:09 2023 daemon.err hostapd: nl80211: kernel reports: integer out of range
Sun Jul 16 16:21:09 2023 daemon.notice hostapd: wlan0-1: BEACON-REQ-TX-STATUS 50:64:2b:f7:e0:c7 165 ack=1
Sun Jul 16 16:21:09 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Sun Jul 16 16:21:11 2023 daemon.notice hostapd: wlan1-2: BEACON-REQ-TX-STATUS a4:39:b3:4f:80:b1 157 ack=1
Sun Jul 16 16:21:11 2023 daemon.notice hostapd: wlan1-2: BEACON-RESP-RX a4:39:b3:4f:80:b1 157 00 8124cead15c600000000000000bfff02f9f85e5d7f015aae15c6
Sun Jul 16 16:21:13 2023 daemon.err hostapd: nl80211: kernel reports: integer out of range
Sun Jul 16 16:21:14 2023 daemon.err hostapd: nl80211: kernel reports: integer out of range
Sun Jul 16 16:21:14 2023 daemon.err hostapd: nl80211: kernel reports: integer out of range
Sun Jul 16 16:21:15 2023 daemon.err hostapd: nl80211: kernel reports: integer out of range
Sun Jul 16 16:21:15 2023 daemon.err hostapd: nl80211: kernel reports: integer out of range
Sun Jul 16 16:21:16 2023 daemon.notice hostapd: wlan1-1: BEACON-REQ-TX-STATUS ba:a0:8c:1a:09:fd 198 ack=1
Sun Jul 16 16:21:16 2023 daemon.notice hostapd: wlan1-1: BEACON-RESP-RX ba:a0:8c:1a:09:fd 198 04
Sun Jul 16 16:21:19 2023 daemon.err hostapd: nl80211: kernel reports: integer out of range
Sun Jul 16 16:21:19 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Sun Jul 16 16:21:20 2023 daemon.err hostapd: nl80211: kernel reports: integer out of range
Sun Jul 16 16:21:20 2023 daemon.err hostapd: nl80211: kernel reports: integer out of range
Sun Jul 16 16:21:21 2023 daemon.err hostapd: nl80211: kernel reports: integer out of range
Sun Jul 16 16:21:21 2023 daemon.err hostapd: nl80211: kernel reports: integer out of range
Sun Jul 16 16:21:25 2023 daemon.err hostapd: nl80211: kernel reports: integer out of range
Sun Jul 16 16:21:26 2023 daemon.err hostapd: nl80211: kernel reports: integer out of range
Sun Jul 16 16:21:26 2023 daemon.err hostapd: nl80211: kernel reports: integer out of range
Sun Jul 16 16:21:27 2023 daemon.err hostapd: nl80211: kernel reports: integer out of range
Sun Jul 16 16:21:27 2023 daemon.err hostapd: nl80211: kernel reports: integer out of range
Sun Jul 16 16:21:27 2023 daemon.notice hostapd: wlan1-3: BEACON-REQ-TX-STATUS bc:d0:74:03:9d:4b 166 ack=1
Sun Jul 16 16:21:27 2023 daemon.notice hostapd: wlan1-3: BEACON-RESP-RX bc:d0:74:03:9d:4b 166 04
Sun Jul 16 16:21:29 2023 daemon.notice hostapd: wlan0-1: BEACON-REQ-TX-STATUS 50:64:2b:f7:e0:c7 166 ack=1
Sun Jul 16 16:21:30 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Sun Jul 16 16:21:31 2023 daemon.notice hostapd: wlan1-2: BEACON-REQ-TX-STATUS a4:39:b3:4f:80:b1 158 ack=1
Sun Jul 16 16:21:31 2023 daemon.err hostapd: nl80211: kernel reports: integer out of range
Sun Jul 16 16:21:31 2023 daemon.notice hostapd: wlan1-2: BEACON-RESP-RX a4:39:b3:4f:80:b1 158 00 81247ced47c700000000000000beff02f9f85e5d7f01e0ed47c7
Sun Jul 16 16:21:32 2023 daemon.err hostapd: nl80211: kernel reports: integer out of range
Sun Jul 16 16:21:32 2023 daemon.err hostapd: nl80211: kernel reports: integer out of range
Sun Jul 16 16:21:33 2023 daemon.err hostapd: nl80211: kernel reports: integer out of range
Sun Jul 16 16:21:33 2023 daemon.err hostapd: nl80211: kernel reports: integer out of range
Sun Jul 16 16:21:36 2023 daemon.notice hostapd: wlan1-1: BEACON-REQ-TX-STATUS ba:a0:8c:1a:09:fd 199 ack=1
Sun Jul 16 16:21:36 2023 daemon.notice hostapd: wlan1-1: BEACON-RESP-RX ba:a0:8c:1a:09:fd 199 04
Sun Jul 16 16:21:37 2023 daemon.err hostapd: nl80211: kernel reports: integer out of range
Sun Jul 16 16:21:38 2023 daemon.err hostapd: nl80211: kernel reports: integer out of range
Sun Jul 16 16:21:38 2023 daemon.err hostapd: nl80211: kernel reports: integer out of range
Sun Jul 16 16:21:39 2023 daemon.err hostapd: nl80211: kernel reports: integer out of range
Sun Jul 16 16:21:39 2023 daemon.err hostapd: nl80211: kernel reports: integer out of range
Sun Jul 16 16:21:40 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
[... 25 lines cut ...]
Sun Jul 16 16:21:56 2023 daemon.notice hostapd: wlan1-1: BEACON-RESP-RX ba:a0:8c:1a:09:fd 201 04
Sun Jul 16 16:21:56 2023 daemon.err hostapd: nl80211: kernel reports: integer out of range
Sun Jul 16 16:21:56 2023 daemon.err hostapd: nl80211: kernel reports: integer out of range
Sun Jul 16 16:21:57 2023 daemon.err hostapd: nl80211: kernel reports: integer out of range
Sun Jul 16 16:21:57 2023 daemon.err hostapd: nl80211: kernel reports: integer out of range
Sun Jul 16 16:22:00 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Sun Jul 16 16:22:01 2023 daemon.err hostapd: nl80211: kernel reports: integer out of range
Sun Jul 16 16:22:02 2023 daemon.err hostapd: nl80211: kernel reports: integer out of range
Sun Jul 16 16:22:02 2023 daemon.err hostapd: nl80211: kernel reports: integer out of range
Sun Jul 16 16:22:03 2023 daemon.err hostapd: nl80211: kernel reports: integer out of range
Sun Jul 16 16:22:03 2023 daemon.err hostapd: nl80211: kernel reports: integer out of range
Sun Jul 16 16:22:07 2023 daemon.notice hostapd: wlan1-3: BEACON-REQ-TX-STATUS bc:d0:74:03:9d:4b 168 ack=1
Sun Jul 16 16:22:07 2023 daemon.notice hostapd: wlan1-3: BEACON-RESP-RX bc:d0:74:03:9d:4b 168 04
Sun Jul 16 16:22:07 2023 daemon.err hostapd: nl80211: kernel reports: integer out of range
Sun Jul 16 16:22:08 2023 daemon.err hostapd: nl80211: kernel reports: integer out of range
Sun Jul 16 16:22:08 2023 daemon.err hostapd: nl80211: kernel reports: integer out of range
Sun Jul 16 16:22:09 2023 daemon.err hostapd: nl80211: kernel reports: integer out of range
Sun Jul 16 16:22:09 2023 daemon.err hostapd: nl80211: kernel reports: integer out of range
Sun Jul 16 16:22:09 2023 daemon.notice hostapd: wlan0-1: BEACON-REQ-TX-STATUS 50:64:2b:f7:e0:c7 168 ack=1
Sun Jul 16 16:22:10 2023 daemon.info dawn: Client A4:39:B3:4F:80:B1: Kicking due to low active data transfer: RX rate 6.000000 below 6 limit
Sun Jul 16 16:22:10 2023 daemon.notice hostapd: wlan1-2: BSS-TM-RESP a4:39:b3:4f:80:b1 status_code=1 bss_termination_delay=0
Sun Jul 16 16:22:10 2023 daemon.notice hostapd: wlan1-2: BSS-TM-RESP a4:39:b3:4f:80:b1 status_code=1 bss_termination_delay=0
Sun Jul 16 16:22:10 2023 daemon.err dawn: ubus_call_umdns()=ubus.c@1385 Failed to look up test object for umdns
Sun Jul 16 16:22:11 2023 daemon.notice hostapd: wlan1-2: BEACON-REQ-TX-STATUS a4:39:b3:4f:80:b1 160 ack=1
Sun Jul 16 16:22:11 2023 daemon.notice hostapd: wlan1-2: BEACON-RESP-RX a4:39:b3:4f:80:b1 160 00 8124eeddaac900000000000000bbff02f9f85e5d7f0166deaac9

xiangfeidexiaohuo, 2023-08-05:
红米ax6000 也出现过这情况

yanxin152133, 2023-08-31:
你看看是不是买到减配的

yanxin152133, 2023-08-31:
我买的TP-LINK XDR3010 减配 会出现iPhone 隔一天就无法连接无线网，而且还有手机打开WiFi加入网络比其他路由器还慢，必须手动点击加入

ajr2l, 2024-05-31:
你好，请问你这个问题到最后是怎么解决的？

coolsnowwolf, 2026-09-05:
已处理：feeds/packages 中的 dawn 从 2022-07-24（9e8060e）更新到上游 2026-06-10（806166b），并启用主线的 no-LTO 构建选项。xdr6088 报告中的 DAWN 低速踢客户端和 hostapd 异常属于旧版本行为；新版本已完成本地 x86_64/musl 编译验证。提交 `16108e461eaead459eda777d6f93ace2a541ddd7` 已推送到 packages/master。请在 XDR6088 刷入后观察 1–2 天无线稳定性。

closed 2026-09-05 by coolsnowwolf