coolsnowwolf/lede #11394: 編譯 photonicat 出現 libiwinfo-data 相依性錯誤

shuenpin, 2023-07-18:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency libiwinfo-data for libiwinfo20230121
 * pkg_hash_fetch_best_installation_candidate: Packages for libiwinfo20230121 found, but incompatible with the architectures configured
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-cfg80211:
 *      libiwinfo-data
 * opkg_install_cmd: Cannot install package kmod-cfg80211.
make[2]: *** [package/Makefile:70: package/install] Error 255
make[2]: Leaving directory '/home/sbhsu/Downloads/lede'
make[1]: *** [package/Makefile:111: /home/sbhsu/Downloads/lede/staging_dir/target-aarch64_generic_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/sbhsu/Downloads/lede'
make: *** [/home/sbhsu/Downloads/lede/include/toplevel.mk:231: world] Error 2

coolsnowwolf, 2023-07-19:
rm -rf bin

shuenpin, 2023-07-19:
> rm -rf bin

跳出一樣的錯誤，需要 make dirclean or make clean 嗎？

shuenpin, 2023-07-20:
> 按大雕说的试下，不行就clean了重新看下可以不，如果还不行我明天给你看看

剛 make clean 後重新編譯還是一樣的錯誤 :'(

附件為 .config
[config.txt](https://github.com/coolsnowwolf/lede/files/12101692/config.txt)

謝謝

wuzhuang123, 2023-07-21:
删除bin和tmp文件夹，重新make menuconfig一下

WYC-2020, 2023-07-21:
> > 按大雕说的试下，不行就clean了重新看下可以不，如果还不行我明天给你看看
> 
> 剛 make clean 後重新編譯還是一樣的錯誤 :'(
> 
> 附件為 .config [config.txt](https://github.com/coolsnowwolf/lede/files/12101692/config.txt)
> 
> 謝謝
[onfig.zip](https://github.com/coolsnowwolf/lede/files/12115501/onfig.zip)
我把你那个config 拿过来 然后 make menuconfig  然后没有任何更改，再保存了，然后编译有一个地方有问题就是hostapd，其他都一切正常，你那个错误没遇到，hostapd错误的话 如此修复
[image]

shuenpin, 2023-07-21:
> 我把你那个config 拿过来 然后 make menuconfig  然后没有任何更改，再保存了，然后编译有一个地方有问题就是hostapd，其他都一切正常，你那个错误没遇到，hostapd错误的话 如此修复
> [image]

感謝，我重新來測試看看。

shuenpin, 2023-07-21:
順利編譯過去了，但是到了最後面出現 
<pre><code>
bash: line 6: 8: Bad file descriptor
Signing package index...
make[2]: Leaving directory '/home/sbhsu/Downloads/lede'
export MAKEFLAGS= ;make -w -r json_overview_image_info
make[2]: Entering directory '/home/sbhsu/Downloads/lede'
WORK_DIR=/home/sbhsu/Downloads/lede/build_dir/target-aarch64_generic_musl/json_info_files /home/sbhsu/Downloads/lede/scripts/json_overview_image_info.py /home/sbhsu/Downloads/lede/bin/targets/rockchip/armv8/profiles.json
make[2]: Leaving directory '/home/sbhsu/Downloads/lede'
export MAKEFLAGS= ;make -w -r checksum
make[2]: Entering directory '/home/sbhsu/Downloads/lede'
make[2]: Leaving directory '/home/sbhsu/Downloads/lede'
make[1]: Leaving directory '/home/sbhsu/Downloads/lede'
</code></pre>

shuenpin, 2023-07-21:
看來 image 正確出現，我來刷進去 Photonicat 看看

shuenpin, 2023-07-21:
@WYC-2020 我發現我用到 5.15 的內核編譯了，編譯完也可以正常 emmc 引導。我來測試看看 6.1 內核試試。

shuenpin, 2023-07-21:
@WYC-2020 把內核改成 6.1 編譯後，錯誤依舊一模一樣

[config61.txt](https://github.com/coolsnowwolf/lede/files/12130990/config61.txt)

再麻煩您幫我用這新版的 config file 編譯試試，謝謝

WYC-2020, 2023-07-22:
> @WYC-2020 把內核改成 6.1 編譯後，錯誤依舊一模一樣
> 
> [config61.txt](https://github.com/coolsnowwolf/lede/files/12130990/config61.txt)
> 
> 再麻煩您幫我用這新版的 config file 編譯試試，謝謝

没问题 
[config.zip](https://github.com/coolsnowwolf/lede/files/12135982/config.zip)

closed 2023-07-24 by shuenpin