coolsnowwolf/lede #11398: 加了busybox以后，编译无法启动

Jennal, 2023-07-22:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：不加busybox的时候，是好的，因为我希望能扩容磁盘，所以加了busybox的fdisk、losetup。以及加了Utilities/Filesystem的resize2fs。编译正常，安装使用的是openwrt-x86-64-generic-ext4-combined-efi.img.gz，用physicswrite刷写。不加busybox的时候，可以正常启动登录。加了以后会卡在

```
init: - preinit -
init: Failed to start preinit: No such file or directory
procd: - early -
procd: - ubus -
procd: - init -
Please press Enter to activate this console.
```

我敲了Enter，然后

```
Failed to execute /usr/libexec/login.sh
```

### (2) 路由器型号和固件版本

A：路由器是x86的软路由，固件版本（目前最新的commit）：279cda2a4ca5e948970e5f4cce59994ed5dd8881

WYC-2020, 2023-07-25:
刚试了下 没你这个问题呢 ，我也按你的 选择 吧 fdisk、losetup resize2fs 都选中了 没出现你这个问题
[image]

Jennal, 2023-07-25:
感谢回复，可能确实跟busybox没关系，我现在把busybox去掉了，也还是这样，实体机和虚拟机，都是一样的报错。怀疑是系统的软件包比较旧，我做了一下 `apt upgrade`，然后把首页建议安装的 `apt install`，现在正在重新打包测试

Jennal, 2023-07-25:
可能是我以前编译过，残留一些老的中间文件，我全部删除，重新clone，然后从头来一遍，就正常了

closed 2023-07-25 by Jennal