coolsnowwolf/lede #11401: 请教docker事件日志问题分析

pk49800, 2023-07-23:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
docker事件日志频繁打印容器创建启动销毁的循环
出现该问题的镜像：deluan/navidrome:latest

### (2) 路由器型号和固件版本
型号 R7800
固件版本 OpenWrt R23.7.7 / LuCI Master (git-23.198.59743-991daf5)

### (3) 详细日志

```bash
[2023-07-23 15:54:34] container exec_create: /bin/sh -c wget -O- http://localhost:${ND_PORT}/ping || exit 1 Container ID: 57305ed3fd4aec194051326374fa249c86fdd8e576c87855cb62022eafc3c421 Container Name: navidrome
[2023-07-23 15:54:34] container exec_start: /bin/sh -c wget -O- http://localhost:${ND_PORT}/ping || exit 1 Container ID: 57305ed3fd4aec194051326374fa249c86fdd8e576c87855cb62022eafc3c421 Container Name: navidrome
[2023-07-23 15:54:34] container exec_die Container ID: 57305ed3fd4aec194051326374fa249c86fdd8e576c87855cb62022eafc3c421 Container Name: navidrome
[2023-07-23 15:55:04] container exec_create: /bin/sh -c wget -O- http://localhost:${ND_PORT}/ping || exit 1 Container ID: 57305ed3fd4aec194051326374fa249c86fdd8e576c87855cb62022eafc3c421 Container Name: navidrome
[2023-07-23 15:55:04] container exec_start: /bin/sh -c wget -O- http://localhost:${ND_PORT}/ping || exit 1 Container ID: 57305ed3fd4aec194051326374fa249c86fdd8e576c87855cb62022eafc3c421 Container Name: navidrome
[2023-07-23 15:55:04] container exec_die Container ID: 57305ed3fd4aec194051326374fa249c86fdd8e576c87855cb62022eafc3c421 Container Name: navidrome
[2023-07-23 15:55:34] container exec_create: /bin/sh -c wget -O- http://localhost:${ND_PORT}/ping || exit 1 Container ID: 57305ed3fd4aec194051326374fa249c86fdd8e576c87855cb62022eafc3c421 Container Name: navidrome
[2023-07-23 15:55:34] container exec_start: /bin/sh -c wget -O- http://localhost:${ND_PORT}/ping || exit 1 Container ID: 57305ed3fd4aec194051326374fa249c86fdd8e576c87855cb62022eafc3c421 Container Name: navidrome
[2023-07-23 15:55:34] container exec_die Container ID: 57305ed3fd4aec194051326374fa249c86fdd8e576c87855cb62022eafc3c421 Container Name: navidrome
[2023-07-23 15:56:05] container exec_create: /bin/sh -c wget -O- http://localhost:${ND_PORT}/ping || exit 1 Container ID: 57305ed3fd4aec194051326374fa249c86fdd8e576c87855cb62022eafc3c421 Container Name: navidrome
[2023-07-23 15:56:05] container exec_start: /bin/sh -c wget -O- http://localhost:${ND_PORT}/ping || exit 1 Container ID: 57305ed3fd4aec194051326374fa249c86fdd8e576c87855cb62022eafc3c421 Container Name: navidrome
[2023-07-23 15:56:05] container exec_die Container ID: 57305ed3fd4aec194051326374fa249c86fdd8e576c87855cb62022eafc3c421 Container Name: navidrome
[2023-07-23 15:56:35] container exec_create: /bin/sh -c wget -O- http://localhost:${ND_PORT}/ping || exit 1 Container ID: 57305ed3fd4aec194051326374fa249c86fdd8e576c87855cb62022eafc3c421 Container Name: navidrome
[2023-07-23 15:56:35] container exec_start: /bin/sh -c wget -O- http://localhost:${ND_PORT}/ping || exit 1 Container ID: 57305ed3fd4aec194051326374fa249c86fdd8e576c87855cb62022eafc3c421 Container Name: navidrome
[2023-07-23 15:56:35] container exec_die Container ID: 57305ed3fd4aec194051326374fa249c86fdd8e576c87855cb62022eafc3c421 Container Name: navidrome
[2023-07-23 15:57:05] container exec_create: /bin/sh -c wget -O- http://localhost:${ND_PORT}/ping || exit 1 Container ID: 57305ed3fd4aec194051326374fa249c86fdd8e576c87855cb62022eafc3c421 Container Name: navidrome
[2023-07-23 15:57:05] container exec_start: /bin/sh -c wget -O- http://localhost:${ND_PORT}/ping || exit 1 Container ID: 57305ed3fd4aec194051326374fa249c86fdd8e576c87855cb62022eafc3c421 Container Name: navidrome
[2023-07-23 15:57:05] container exec_die Container ID: 57305ed3fd4aec194051326374fa249c86fdd8e576c87855cb62022eafc3c421 Container Name: navidrome
[2023-07-23 15:57:35] container exec_create: /bin/sh -c wget -O- http://localhost:${ND_PORT}/ping || exit 1 Container ID: 57305ed3fd4aec194051326374fa249c86fdd8e576c87855cb62022eafc3c421 Container Name: navidrome
[2023-07-23 15:57:35] container exec_start: /bin/sh -c wget -O- http://localhost:${ND_PORT}/ping || exit 1 Container ID: 57305ed3fd4aec194051326374fa249c86fdd8e576c87855cb62022eafc3c421 Container Name: navidrome
[2023-07-23 15:57:35] container exec_die Container ID: 57305ed3fd4aec194051326374fa249c86fdd8e576c87855cb62022eafc3c421 Container Name: navidrome
[2023-07-23 15:58:06] container exec_create: /bin/sh -c wget -O- http://localhost:${ND_PORT}/ping || exit 1 Container ID: 57305ed3fd4aec194051326374fa249c86fdd8e576c87855cb62022eafc3c421 Container Name: navidrome
[2023-07-23 15:58:06] container exec_start: /bin/sh -c wget -O- http://localhost:${ND_PORT}/ping || exit 1 Container ID: 57305ed3fd4aec194051326374fa249c86fdd8e576c87855cb62022eafc3c421 Container Name: navidrome
[2023-07-23 15:58:06] container exec_die Container ID: 57305ed3fd4aec194051326374fa249c86fdd8e576c87855cb62022eafc3c421 Container Name: navidrome
[2023-07-23 15:58:36] container exec_create: /bin/sh -c wget -O- http://localhost:${ND_PORT}/ping || exit 1 Container ID: 57305ed3fd4aec194051326374fa249c86fdd8e576c87855cb62022eafc3c421 Container Name: navidrome
[2023-07-23 15:58:36] container exec_start: /bin/sh -c wget -O- http://localhost:${ND_PORT}/ping || exit 1 Container ID: 57305ed3fd4aec194051326374fa249c86fdd8e576c87855cb62022eafc3c421 Container Name: navidrome
[2023-07-23 15:58:36] container exec_die Container ID: 57305ed3fd4aec194051326374fa249c86fdd8e576c87855cb62022eafc3c421 Container Name: navidrome
[2023-07-23 15:59:06] container exec_create: /bin/sh -c wget -O- http://localhost:${ND_PORT}/ping || exit 1 Container ID: 57305ed3fd4aec194051326374fa249c86fdd8e576c87855cb62022eafc3c421 Container Name: navidrome
[2023-07-23 15:59:06] container exec_start: /bin/sh -c wget -O- http://localhost:${ND_PORT}/ping || exit 1 Container ID: 57305ed3fd4aec194051326374fa249c86fdd8e576c87855cb62022eafc3c421 Container Name: navidrome
[2023-07-23 15:59:06] container exec_die Container ID: 57305ed3fd4aec194051326374fa249c86fdd8e576c87855cb62022eafc3c421 Container Name: navidrome
[2023-07-23 15:59:36] container exec_create: /bin/sh -c wget -O- http://localhost:${ND_PORT}/ping || exit 1 Container ID: 57305ed3fd4aec194051326374fa249c86fdd8e576c87855cb62022eafc3c421 Container Name: navidrome
[2023-07-23 15:59:36] container exec_start: /bin/sh -c wget -O- http://localhost:${ND_PORT}/ping || exit 1 Container ID: 57305ed3fd4aec194051326374fa249c86fdd8e576c87855cb62022eafc3c421 Container Name: navidrome
[2023-07-23 15:59:37] container exec_die Container ID: 57305ed3fd4aec194051326374fa249c86fdd8e576c87855cb62022eafc3c421 Container Name: navidrome
[2023-07-23 16:00:07] container exec_create: /bin/sh -c wget -O- http://localhost:${ND_PORT}/ping || exit 1 Container ID: 57305ed3fd4aec194051326374fa249c86fdd8e576c87855cb62022eafc3c421 Container Name: navidrome
[2023-07-23 16:00:07] container exec_start: /bin/sh -c wget -O- http://localhost:${ND_PORT}/ping || exit 1 Container ID: 57305ed3fd4aec194051326374fa249c86fdd8e576c87855cb62022eafc3c421 Container Name: navidrome
[2023-07-23 16:00:07] container exec_die Container ID: 57305ed3fd4aec194051326374fa249c86fdd8e576c87855cb62022eafc3c421 Container Name: navidrome
[2023-07-23 16:00:37] container exec_create: /bin/sh -c wget -O- http://localhost:${ND_PORT}/ping || exit 1 Container ID: 57305ed3fd4aec194051326374fa249c86fdd8e576c87855cb62022eafc3c421 Container Name: navidrome
[2023-07-23 16:00:37] container exec_start: /bin/sh -c wget -O- http://localhost:${ND_PORT}/ping || exit 1 Container ID: 57305ed3fd4aec194051326374fa249c86fdd8e576c87855cb62022eafc3c421 Container Name: navidrome
[2023-07-23 16:00:37] container exec_die Container ID: 57305ed3fd4aec194051326374fa249c86fdd8e576c87855cb62022eafc3c421 Container Name: navidrome
[2023-07-23 16:01:07] container exec_create: /bin/sh -c wget -O- http://localhost:${ND_PORT}/ping || exit 1 Container ID: 57305ed3fd4aec194051326374fa249c86fdd8e576c87855cb62022eafc3c421 Container Name: navidrome
[2023-07-23 16:01:07] container exec_start: /bin/sh -c wget -O- http://localhost:${ND_PORT}/ping || exit 1 Container ID: 57305ed3fd4aec194051326374fa249c86fdd8e576c87855cb62022eafc3c421 Container Name: navidrome
[2023-07-23 16:01:07] container exec_die Container ID: 57305ed3fd4aec194051326374fa249c86fdd8e576c87855cb62022eafc3c421 Container Name: navidrome
[2023-07-23 16:01:37] container exec_create: /bin/sh -c wget -O- http://localhost:${ND_PORT}/ping || exit 1 Container ID: 57305ed3fd4aec194051326374fa249c86fdd8e576c87855cb62022eafc3c421 Container Name: navidrome
[... cut ...]
[2023-07-23 16:36:37] container exec_create: /bin/sh -c wget -O- http://localhost:${ND_PORT}/ping || exit 1 Container ID: 57305ed3fd4aec194051326374fa249c86fdd8e576c87855cb62022eafc3c421 Container Name: navidrome
[2023-07-23 16:36:37] container exec_start: /bin/sh -c wget -O- http://localhost:${ND_PORT}/ping || exit 1 Container ID: 57305ed3fd4aec194051326374fa249c86fdd8e576c87855cb62022eafc3c421 Container Name: navidrome
[2023-07-23 16:36:37] container exec_die Container ID: 57305ed3fd4aec194051326374fa249c86fdd8e576c87855cb62022eafc3c421 Container Name: navidrome
```

WYC-2020, 2023-07-25:
自己加了什么保活玩意，退出不是很正常 wget -O- http://localhost:${ND_PORT}/ping 这玩意失败的时候 就执行后面的退出了 能怪谁，自己加了什么 自己排查

pk49800, 2023-07-25:
> 自己加了什么保活玩意，退出不是很正常 wget -O- [http://localhost:${ND_PORT}/ping](http://localhost:$%7BND_PORT%7D/ping) 这玩意失败的时候 就执行后面的退出了 能怪谁，自己加了什么 自己排查

完全不会docker命令，什么也没加，没用用命令行而是直接在界面正常的填写端口和绑定目录，还有就是勾选了`交互`和`TTY`这两个选项

WYC-2020, 2023-07-25:
[image]
这玩意 自己加的心跳，要么用老版本去找没有加心跳的，或者参考这个issue 试下
https://github.com/navidrome/navidrome/issues/773

> > 自己加了什么保活玩意，退出不是很正常 wget -O- [http://localhost:${ND_PORT}/ping](http://localhost:$%7BND_PORT%7D/ping) 这玩意失败的时候 就执行后面的退出了 能怪谁，自己加了什么 自己排查
> 
> 完全不会docker命令，什么也没加，没用用命令行而是直接在界面正常的填写端口和绑定目录，还有就是勾选了`交互`和`TTY`这两个选项

pk49800, 2023-07-25:
> [image] 这玩意 自己加的心跳，要么用老版本去找没有加心跳的，或者参考这个issue 试下 [navidrome/navidrome#773](https://github.com/navidrome/navidrome/issues/773)
> 
> > > 自己加了什么保活玩意，退出不是很正常 wget -O- [http://localhost:${ND_PORT}/ping](http://localhost:$%7BND_PORT%7D/ping) 这玩意失败的时候 就执行后面的退出了 能怪谁，自己加了什么 自己排查
> > 
> > 
> > 完全不会docker命令，什么也没加，没用用命令行而是直接在界面正常的填写端口和绑定目录，还有就是勾选了`交互`和`TTY`这两个选项
好的，谢谢，我试试

closed 2023-07-25 by pk49800