coolsnowwolf/lede #11408: make defconfig报错

smallprogram, 2023-07-27:
openwrt在make defconfig时报错，内容如下：
```
tmp/.config-package.in:77116:error: recursive dependency detected!
tmp/.config-package.in:77116:   symbol PACKAGE_nodogsplash depends on PACKAGE_opennds
tmp/.config-package.in:77138:   symbol PACKAGE_opennds depends on PACKAGE_nodogsplash
For a resolution refer to Documentation/kbuild/kconfig-language.rst
subsection "Kconfig recursive dependency limitations"

#
# configuration written to .config
#
```
貌似PACKAGE_nodogsplash 与 PACKAGE_opennds循环依赖了，但是我的config中并不存在什么PACKAGE_nodogsplash ，而PACKAGE_opennds也是 not set

aiamadeus, 2023-07-28:
你自己加了什么包吧

smallprogram, 2023-07-28:
问题出在route feed上，已经提交了pr，目前还没有被合并
https://github.com/openwrt/routing/pull/1001
https://github.com/openwrt/routing/pull/1002

closed 2023-07-30 by smallprogram