coolsnowwolf/lede #11409: lrzsz编译报错

smallprogram, 2023-07-27:
```
make[3]: Entering directory '/home/zhusir/openwrt/build_dir/target-x86_64_musl/lrzsz-990823'
cd . && aclocal
aclocal.real: warning: autoconf input should be named 'configure.ac', not 'configure.in'
acinclude.m4:2: warning: underquoted definition of AC_REPLACE_GNU_GETOPT
acinclude.m4:2:   run info Automake 'Extending aclocal'
acinclude.m4:2:   or see http://www.gnu.org/software/automake/manual/automake.html#Extending-aclocal
configure.in:106: error: automatic de-ANSI-fication support has been removed
/home/zhusir/openwrt/staging_dir/host/share/aclocal-1.15/obsolete.m4:26: AM_C_PROTOTYPES is expanded from...
configure.in:106: the top level
autom4te: /home/zhusir/openwrt/staging_dir/host/bin/m4 failed with exit status: 1
aclocal.real: error: echo failed with exit status: 1
make[3]: *** [Makefile:127: aclocal.m4] Error 1
make[3]: Leaving directory '/home/zhusir/openwrt/build_dir/target-x86_64_musl/lrzsz-990823'
make[2]: *** [Makefile:72: /home/zhusir/openwrt/build_dir/target-x86_64_musl/lrzsz-990823/.built] Error 2
make[2]: Leaving directory '/home/zhusir/openwrt/feeds/packages/utils/lrzsz'
time: package/feeds/packages/lrzsz/compile#0.35#0.00#0.35
    ERROR: package/feeds/packages/lrzsz failed to build.
make[1]: *** [package/Makefile:116: package/feeds/packages/lrzsz/compile] Error 1
make[1]: Leaving directory '/home/zhusir/openwrt'
make: *** [/home/zhusir/openwrt/include/toplevel.mk:231: package/compile] Error 2
```

WYC-2020, 2023-07-28:
一切正常 x86,没办法复现
[image]

smallprogram, 2023-07-28:
> 一切正常 x86,没办法复现 [image]

那估计就是我本地的问题了

closed 2023-07-28 by smallprogram

Beaverfffan, 2024-11-13:
ging_dir/target-aarch64_generic_musl/root-armvirt/'
touch /home/beaver/lede2/staging_dir/target-aarch64_generic_musl/root-armvirt/stamp/.lm-sensors_installed
echo "lm-sensors" >> /home/beaver/lede2/staging_dir/target-aarch64_generic_musl/pkginfo/lm-sensors.default.install
make[3]: Leaving directory '/home/beaver/lede2/feeds/packages/utils/lm-sensors'
time: package/feeds/packages/lm-sensors/compile#3.48#1.47#5.25
make[3]: Entering directory '/home/beaver/lede2/feeds/packages/utils/lrzsz'
rm -f /home/beaver/lede2/build_dir/target-aarch64_generic_musl/lrzsz-990823/.built
touch /home/beaver/lede2/build_dir/target-aarch64_generic_musl/lrzsz-990823/.built_check
CFLAGS="-Os -pipe -mcpu=generic -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/beaver/lede2/build_dir/target-aarch64_generic_musl/lrzsz-990823=lrzsz-990823 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/beaver/lede2/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/usr/include -I/home/beaver/lede2/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/include/fortify -I/home/beaver/lede2/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/include " CXXFLAGS="-Os -pipe -mcpu=generic -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/beaver/lede2/build_dir/target-aarch64_generic_musl/lrzsz-990823=lrzsz-990823 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/beaver/lede2/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/usr/include -I/home/beaver/lede2/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/include/fortify -I/home/beaver/lede2/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/include " LDFLAGS="-L/home/beaver/lede2/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/usr/lib -L/home/beaver/lede2/staging_dir/toolchain-aarch64_generic_gcc-11.3.0_musl/lib -znow -zrelro " make -j1 -C /home/beaver/lede2/build_dir/target-aarch64_generic_musl/lrzsz-990823/. AR="aarch64-openwrt-linux-musl-gcc-ar" AS="aarch64-openwrt-linux-musl-gcc -c -Os -pipe -mcpu=generic -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/beaver/lede2/build_dir/target-aarch64_generic_musl/lrzsz-990823=lrzsz-990823 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=aarch64-openwrt-linux-musl-ld NM="aarch64-openwrt-linux-musl-gcc-nm" CC="aarch64-openwrt-linux-musl-gcc" GCC="aarch64-openwrt-linux-musl-gcc" CXX="aarch64-openwrt-linux-musl-g++" RANLIB="aarch64-openwrt-linux-musl-gcc-ranlib" STRIP=aarch64-openwrt-linux-musl-strip OBJCOPY=aarch64-openwrt-linux-musl-objcopy OBJDUMP=aarch64-openwrt-linux-musl-objdump SIZE=aarch64-openwrt-linux-musl-size CROSS="aarch64-openwrt-linux-musl-" ARCH="aarch64" ;
make[4]: Entering directory '/home/beaver/lede2/build_dir/target-aarch64_generic_musl/lrzsz-990823'
cd . && automake --gnu Makefile
automake: warning: autoconf input should be named 'configure.ac', not 'configure.in'
configure.in:3: warning: AM_INIT_AUTOMAKE: two- and three-arguments forms are deprecated.  For more info, see:
configure.in:3: https://www.gnu.org/software/automake/manual/automake.html#Modernize-AM_005fINIT_005fAUTOMAKE-invocation
configure.in:3: error: your implementation of AM_INIT_AUTOMAKE comes from an
configure.in:3: old Automake version.  You should recreate aclocal.m4
configure.in:3: with aclocal and run automake again
Makefile.am:27: warning: '%'-style pattern rules are a GNU make extension
Makefile.am:28: warning: subst vcheck-,,$@: non-POSIX variable name
Makefile.am:28: (probably a GNU make extension)
Makefile.am:29: warning: '%'-style pattern rules are a GNU make extension
Makefile.am:30: warning: subst check-,,$@: non-POSIX variable name
Makefile.am:30: (probably a GNU make extension)
make[4]: *** [Makefile:120: Makefile.in] Error 63
make[4]: Leaving directory '/home/beaver/lede2/build_dir/target-aarch64_generic_musl/lrzsz-990823'
make[3]: *** [Makefile:72: /home/beaver/lede2/build_dir/target-aarch64_generic_musl/lrzsz-990823/.built] Error 2
make[3]: Leaving directory '/home/beaver/lede2/feeds/packages/utils/lrzsz'
time: package/feeds/packages/lrzsz/compile#0.41#0.49#0.82
    ERROR: package/feeds/packages/lrzsz failed to build.
make[2]: *** [package/Makefile:116: package/feeds/packages/lrzsz/compile] Error 1
make[2]: Leaving directory '/home/beaver/lede2'
make[1]: *** [package/Makefile:110: /home/beaver/lede2/staging_dir/target-aarch64_generic_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/beaver/lede2'
make: *** [/home/beaver/lede2/include/toplevel.mk:231: world] Error 2
(base) beaver@beaver-Virtual-Machine:~/lede2$

Beaverfffan, 2024-11-13:
armvirt下再次出现