coolsnowwolf/lede #11414: 编译X86 6.1.XX时：ath10k-ct出错

dbsjacky, 2023-07-31:
## 1.关于你要提交的问题
编译X86 6.1.XX时：ath10k-ct出错

Q：是否搜索了issue (使用 "x" 选择)
没有类似的issue

## 2. 详细叙述
编译X86 6.1.XX时：ath10k-ct出错

### (1) 具体问题
Applying ./patches/988-ath10k-always-use-mac80211-loss-detection.patch using plaintext: 
can't find file to patch at input line 21
Perhaps you used the wrong -p or --strip option?

### (2) 路由器型号和固件版本
A：X86  6.1.XX

### (3) 详细日志

Applying ./patches/960-0010-ath10k-limit-htt-rx-ring-size.patch using plaintext: 
patching file ath10k-5.15/htt.h

Applying ./patches/960-0011-ath10k-limit-pci-buffer-size.patch using plaintext: 
patching file ath10k-5.15/pci.c

Applying ./patches/988-ath10k-always-use-mac80211-loss-detection.patch using plaintext: 
can't find file to patch at input line 21
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|From f7d6edafe4358e3880a26775cfde4cd5c71ba063 Mon Sep 17 00:00:00 2001
|From: David Bauer <mail@david-bauer.net>
|Date: Wed, 5 Jul 2023 01:30:29 +0200
|Subject: [PATCH] ath10k: always use mac80211 loss detection
|
|ath10k does not report excessive loss in case of broken block-ack
|sessions. The loss is communicated to the host-os, but ath10k does not
|trigger a low-ack events by itself.
|
|The mac80211 framework for loss detection however detects this
|circumstance well in case of ath10k. So use it regardless of ath10k's
|own loss detection mechanism.
|
|Signed-off-by: David Bauer <mail@david-bauer.net>
|---
| ath10k-6.2/mac.c | 1 -
| 1 file changed, 1 deletion(-)
|
|--- a/ath10k-6.2/mac.c
|+++ b/ath10k-6.2/mac.c
--------------------------
No file to patch.  Skipping patch.
1 out of 1 hunk ignored
Patch failed!  Please fix ./patches/988-ath10k-always-use-mac80211-loss-detection.patch!
make[3]: *** [Makefile:128: /home/da/lede/build_dir/target-x86_64_musl/linux-x86_64/ath10k-ct-regular/ath10k-ct-2022-05-13-f808496f/.prepared_970248cf837600394704afe0e241c355_6664517399ebbbc92a37c5bb081b5c53] Error 1
make[3]: Leaving directory '/home/da/lede/package/kernel/ath10k-ct'
time: package/kernel/ath10k-ct/regular/compile#0.51#0.59#0.89
    ERROR: package/kernel/ath10k-ct failed to build (build variant: regular).
make[2]: *** [package/Makefile:116: package/kernel/ath10k-ct/compile] Error 1
make[2]: Leaving directory '/home/da/lede'
make[1]: *** [package/Makefile:110: /home/da/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/da/lede'
make: *** [/home/da/lede/include/toplevel.mk:231：world] 錯誤 2

WYC-2020, 2023-07-31:
胡搞瞎搞，乱拉官方patch,要么升级要么删除这个patch 自己看到办吧

qpzr, 2023-07-31:
星期六编译ipq4019的竞斗云也出错了
make[3]: *** [Makefile:128: /home/runner/work/LOW/LOW/openwrt/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx/ath10k-ct-regular/ath10k-ct-2022-05-13-f808496f/.prepared_7f86c705c6403a46d2281b6924828b53_6664517399ebbbc92a37c5bb081b5c53] Error 1
No file to patch.  Skipping patch.
1 out of 1 hunk ignored
Patch failed!  Please fix ./patches/988-ath10k-always-use-mac80211-loss-detection.patch!
make[3]: Leaving directory '/home/runner/work/LOW/LOW/openwrt/package/kernel/ath10k-ct'
time: package/kernel/ath10k-ct/regular/compile#0.25#0.16#0.34
    ERROR: package/kernel/ath10k-ct failed to build (build variant: regular).
make[2]: *** [package/Makefile:116: package/kernel/ath10k-ct/compile] Error 1
make[2]: Leaving directory '/home/runner/work/LOW/LOW/openwrt'
make[1]: Leaving directory '/home/runner/work/LOW/LOW/openwrt'
make[1]: *** [package/Makefile:110: /home/runner/work/LOW/LOW/openwrt/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2
make: *** [/home/runner/work/LOW/LOW/openwrt/include/toplevel.mk:231: world] Error 2
Error: Process completed with exit code 2.

dbsjacky, 2023-08-01:
我沒亂拉喔，這是“Lean”的原碼喔

dbsjacky, 2023-08-01:
> 
我沒亂拉喔，這是“Lean”的原碼喔

dbsjacky, 2023-08-01:
@WYC-2020 我沒有亂拉喔喔，這是“Lean”的原碼喔

WYC-2020, 2023-08-01:
> @WYC-2020 我沒有亂拉喔喔，這是“Lean”的原碼喔

我说有人提交了一个patch 乱搞，都没更新完毕，直接就官方那边拉过来合并，都不验证一下，我这边搞了一下升级到了6.2 你试试吧，我这边测试 5.4 5.10 6.1 5.15  内核编译都没问题
替换相同路径 
[ath10-ct.tar.gz](https://github.com/coolsnowwolf/lede/files/12223807/ath10-ct.tar.gz)

WYC-2020, 2023-08-01:
@dbsjacky  那个hash 我忘记计算了，你自己改下makefile 不改其实也没啥，看你自己 
[image]
afb7da1b57a5cdc78543eccf86fc5ad00fd7edcab76ba1986223b9afe11c9f1a

dbsjacky, 2023-08-01:
@WYC-2020 好的，謝謝大俠

dbsjacky, 2023-08-01:
@qpzr 同樣按“WYC－2020”大俠說的那樣去做吧

jaybingoodluck, 2023-08-03:
> > @WYC-2020我没有乱拉喔喔，这是“Lean”的原码喔
> 
> 我说有人提交了一个补丁乱搞，都更新完了，直接就官方那边拉过来合并，都不验证一下，我的香料搞了一下升级到了6.2 你试试吧，我的香料测试 5.4 5.10 6.1 5.15内核编译都没问题 替换相同路径 [ath10-ct.tar.gz](https://github.com/coolsnowwolf/lede/files/12223807/ath10-ct.tar.gz)
我也编译失败，请大神指教一下原因，R5S编译失败

closed 2026-09-05 by coolsnowwolf