coolsnowwolf/lede #11416: R7800最新源码编译失败

pk49800, 2023-07-31:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述
使用最新源码编译固件失败
### (1) 具体问题
如上
### (2) 路由器型号和固件版本

主机名 | OpenWrt
主机型号 | Netgear Nighthawk X4S R7800 (CpuMark : 13721.485039 Scores)
架构 | ARMv7 Processor rev 0 (v7l) x 2 (1000MHz, 59.4°C )
固件版本 | OpenWrt R23.7.7 / LuCI Master (git-23.198.59743-991daf5)
内核版本 | 5.4.247

### (3) 详细日志

```bash
No file to patch.  Skipping patch.
1 out of 1 hunk ignored
Patch failed!  Please fix ./patches/988-ath10k-always-use-mac80211-loss-detection.patch!
make[3]: *** [Makefile:128: /home/pk49800/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/linux-ipq806x_generic/ath10k-ct-regular/ath10k-ct-2022-05-13-f808496f/.prepared_3cb0797aa6ae1f3a8ff2a9d7e42471cc_6664517399ebbbc92a37c5bb081b5c53] Error 1
make[3]: Leaving directory '/home/pk49800/lede/package/kernel/ath10k-ct'
time: package/kernel/ath10k-ct/regular/compile#0.16#0.06#0.19
    ERROR: package/kernel/ath10k-ct failed to build (build variant: regular).
make[2]: *** [package/Makefile:116: package/kernel/ath10k-ct/compile] Error 1
make[2]: *** Waiting for unfinished jobs....
mkdir -p /home/pk49800/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/stamp
SHELL= flock /home/pk49800/lede/tmp/.root-copy.flock -c 'cp -fpR /home/pk49800/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/luci-app-firewall/.pkgdir/luci-app-firewall/. /home/pk49800/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/'
touch /home/pk49800/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/stamp/.luci-app-firewall_installed
echo "luci-app-firewall" >> /home/pk49800/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/pkginfo/luci-app-firewall.default.install
mkdir -p /home/pk49800/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/stamp
SHELL= flock /home/pk49800/lede/tmp/.root-copy.flock -c 'cp -fpR /home/pk49800/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/luci-app-firewall/.pkgdir/luci-i18n-firewall-zh-cn/. /home/pk49800/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/'
touch /home/pk49800/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/stamp/.luci-i18n-firewall-zh-cn_installed
echo "luci-i18n-firewall-zh-cn" >> /home/pk49800/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/pkginfo/luci-app-firewall.default.install
make[3]: Leaving directory '/home/pk49800/lede/feeds/luci/applications/luci-app-firewall'
time: package/feeds/luci/luci-app-firewall/compile#0.68#0.06#0.71
touch /home/pk49800/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/luci-base/.prepared_1189ab12061af7ebcd898cfe152924f2_6664517399ebbbc92a37c5bb081b5c53_check
for d in luasrc htdocs root src; do if [ -d ./$d ]; then mkdir -p /home/pk49800/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/luci-base/$d; cp -fpR ./$d/* /home/pk49800/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/luci-base/$d/; fi; done
. /home/pk49800/lede/include/shell.sh; bzcat /home/pk49800/lede/dl/LuaSrcDiet-0.12.1.tar.bz2 | tar -C /home/pk49800/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/luci-base/.. -xf -
mkdir -p /home/pk49800/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/stamp
SHELL= flock /home/pk49800/lede/tmp/.root-copy.flock -c 'cp -fpR /home/pk49800/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/luci-app-samba4/.pkgdir/luci-app-samba4/. /home/pk49800/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/'
touch /home/pk49800/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/stamp/.luci-app-samba4_installed
echo "luci-app-samba4" >> /home/pk49800/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/pkginfo/luci-app-samba4.default.install
mkdir -p /home/pk49800/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/stamp
SHELL= flock /home/pk49800/lede/tmp/.root-copy.flock -c 'cp -fpR /home/pk49800/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/luci-app-samba4/.pkgdir/luci-i18n-samba4-zh-cn/. /home/pk49800/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/'
touch /home/pk49800/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/stamp/.luci-i18n-samba4-zh-cn_installed
echo "luci-i18n-samba4-zh-cn" >> /home/pk49800/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/pkginfo/luci-app-samba4.default.install
make[3]: Leaving directory '/home/pk49800/lede/feeds/luci/applications/luci-app-samba4'
time: package/feeds/luci/luci-app-samba4/compile#0.62#0.03#0.61
[ ! -d ./src/ ] || cp -fpR ./src/. /home/pk49800/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/luci-base
touch /home/pk49800/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/luci-base/.prepared_1189ab12061af7ebcd898cfe152924f2_6664517399ebbbc92a37c5bb081b5c53
rm -f /home/pk49800/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/luci-base/.configured_*
rm -f /home/pk49800/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/stamp/.luci-base_installed
touch /home/pk49800/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/luci-base/.configured_98b31b11569cdb1701bb10c5b1d192c3
rm -f /home/pk49800/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/luci-base/.built
touch /home/pk49800/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/luci-base/.built_check
CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -mfloat-abi=hard -fmacro-prefix-map=/home/pk49800/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/luci-base=luci-base -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/pk49800/lede/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-11.3.0_musl_eabi/usr/include -I/home/pk49800/lede/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-11.3.0_musl_eabi/include/fortify -I/home/pk49800/lede/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-11.3.0_musl_eabi/include " CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -mfloat-abi=hard -fmacro-prefix-map=/home/pk49800/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/luci-base=luci-base -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/pk49800/lede/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-11.3.0_musl_eabi/usr/include -I/home/pk49800/lede/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-11.3.0_musl_eabi/include/fortify -I/home/pk49800/lede/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-11.3.0_musl_eabi/include " LDFLAGS="-L/home/pk49800/lede/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-11.3.0_musl_eabi/usr/lib -L/home/pk49800/lede/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-11.3.0_musl_eabi/lib -znow -zrelro " FPIC="-DPIC -fpic" LUCI_VERSION="git-23.198.59743-991daf5" LUCI_GITBRANCH="LuCI Master" make -j1 -C /home/pk49800/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/luci-base/src/ AR="arm-openwrt-linux-muslgnueabi-gcc-ar" AS="arm-openwrt-linux-muslgnueabi-gcc -c -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -mfloat-abi=hard -fmacro-prefix-map=/home/pk49800/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/luci-base=luci-base -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=arm-openwrt-linux-muslgnueabi-ld NM="arm-openwrt-linux-muslgnueabi-gcc-nm" CC="arm-openwrt-linux-muslgnueabi-gcc" GCC="arm-openwrt-linux-muslgnueabi-gcc" CXX="arm-openwrt-linux-muslgnueabi-g++" RANLIB="arm-openwrt-linux-muslgnueabi-gcc-ranlib" STRIP=arm-openwrt-linux-muslgnueabi-strip OBJCOPY=arm-openwrt-linux-muslgnueabi-objcopy OBJDUMP=arm-openwrt-linux-muslgnueabi-objdump SIZE=arm-openwrt-linux-muslgnueabi-size CROSS="arm-openwrt-linux-muslgnueabi-" ARCH="arm" clean compile;
make[4]: Entering directory '/home/pk49800/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/luci-base/src'
rm -f po2lmo parser.so version.lua *.o
arm-openwrt-linux-muslgnueabi-gcc  -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -mfloat-abi=hard -fmacro-prefix-map=/home/pk49800/lede/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/luci-base=luci-base -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro  -I/home/pk49800/lede/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-11.3.0_musl_eabi/usr/include -I/home/pk49800/lede/ [... cut ...]
echo "luci-i18n-base-zh-cn" >> /home/pk49800/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/pkginfo/luci-base.default.install
make[3]: Leaving directory '/home/pk49800/lede/feeds/luci/modules/luci-base'
time: package/feeds/luci/luci-base/compile#1.63#0.11#1.65
make[2]: Leaving directory '/home/pk49800/lede'
make[1]: *** [package/Makefile:110: /home/pk49800/lede/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/pk49800/lede'
make: *** [/home/pk49800/lede/include/toplevel.mk:231: world] Error 2
```

WYC-2020, 2023-08-01:
https://github.com/coolsnowwolf/lede/issues/11414  看这个帖子 @pk49800

pk49800, 2023-08-01:
> #11414 看这个帖子 @pk49800

测试过按帖子覆盖文件，编译通过但是刷机后路由器进入无限重启和刷机中的状态（橘色灯不变白灯），今天拉取最新源码编译不通过，错误日志已更新，请大佬看看。

hcym, 2023-08-01:
用官方的吧，插件用大雕ssrp或者pass1和2，昨天还是前天编译的master，完美，你要稳定性的话用2203，2305

pk49800, 2023-08-01:
> 用官方的吧，插件用大雕ssrp或者pass1和2，昨天还是前天编译的master，完美，你要稳定性的话用2203，2305

请问如何使用官方的源码加入大雕的插件？我需要大雕的ssrp和turbo acc, samba4 qbittorent这几个，有简单教程或示例连接吗？谢谢。

pk49800, 2023-08-01:
把整个ath10k-ct文件夹删掉重新下载最新源码里面的替换就可以了

closed 2023-08-01 by pk49800