coolsnowwolf/lede #11417: 诚心请教

Spring3month, 2023-08-01:
在编译X86时，全部未改动，使用默认设置，一直无法编译成功。
环境为4代CPU，5G内存，ubuntu 20.04.6 server。

during GIMPLE pass: fre
drivers/gpu/drm/amd/amdgpu/../display/dc/dcn321/dcn321_resource.c: In function 'dcn321_create_resource_pool':
drivers/gpu/drm/amd/amdgpu/../display/dc/dcn321/dcn321_resource.c:2015:23: internal compiler error: Segmentation fault
 2015 | struct resource_pool *dcn321_create_resource_pool(
      |                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~
0xb71213 crash_signal
        /home/xxx/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0/gcc/toplev.c:327
0xd0e5c0 do_rpo_vn
        /home/xxx/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0/gcc/tree-ssa-sccvn.c:7909
0xd0f374 execute
        /home/xxx/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0/gcc/tree-ssa-sccvn.c:8027
Please submit a full bug report,
with preprocessed source if appropriate.
Please include the complete backtrace with any bug report.
See <http://bugs.openwrt.org/> for instructions.
make[10]: *** [scripts/Makefile.build:250: drivers/gpu/drm/amd/amdgpu/../display/dc/dcn321/dcn321_resource.o] Error 1
make[10]: *** Waiting for unfinished jobs....
make[9]: *** [scripts/Makefile.build:500: drivers/gpu/drm/amd/amdgpu] Error 2
make[8]: *** [scripts/Makefile.build:500: drivers/gpu/drm] Error 2
make[7]: *** [scripts/Makefile.build:500: drivers/gpu] Error 2
make[6]: *** [scripts/Makefile.build:500: drivers] Error 2
make[5]: *** [Makefile:2012: .] Error 2
make[5]: Leaving directory '/home/xxx/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.42'
make[4]: *** [Makefile:29: /home/xxx/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.42/.modules] Error 2
make[4]: Leaving directory '/home/xxx/lede/target/linux/x86'

WYC-2020, 2023-08-01:
j1 日志

Spring3month, 2023-08-01:
在这前的环境下，执行make clean rm-rf ./tmp,再重新git pull.运行make V=s -j1,日志如下：

drivers/net/ethernet/mellanox/mlx4/main.c:4544:1: internal compiler error: Segmentation fault
 4544 | module_exit(mlx4_cleanup);
      | ^~~~~~~~~~~
0xb71213 crash_signal
        /home/xxx/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0/gcc/toplev.c:327
0x7df41d verify_marks_clear
        /home/xxx/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0/gcc/dwarf2out.c:29425
0x7df428 verify_marks_clear
        /home/xxx/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0/gcc/dwarf2out.c:29425
0x7fcb01 prune_unused_types
        /home/xxx/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0/gcc/dwarf2out.c:29863
0x82aa84 dwarf2out_early_finish
        /home/xxx/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0/gcc/dwarf2out.c:32456
0x799a73 symbol_table::finalize_compilation_unit()
        /home/xxx/lede/build_dir/toolchain-x86_64_gcc-11.3.0_musl/gcc-11.3.0/gcc/cgraphunit.c:2534
Please submit a full bug report,
with preprocessed source if appropriate.
Please include the complete backtrace with any bug report.
See <http://bugs.openwrt.org/> for instructions.
make[11]: *** [scripts/Makefile.build:250: drivers/net/ethernet/mellanox/mlx4/main.o] Error 1
make[10]: *** [scripts/Makefile.build:500: drivers/net/ethernet/mellanox/mlx4] Error 2
make[9]: *** [scripts/Makefile.build:500: drivers/net/ethernet/mellanox] Error 2
make[8]: *** [scripts/Makefile.build:500: drivers/net/ethernet] Error 2
make[7]: *** [scripts/Makefile.build:500: drivers/net] Error 2
make[6]: *** [scripts/Makefile.build:500: drivers] Error 2
make[5]: *** [Makefile:2012: .] Error 2
make[5]: Leaving directory '/home/xxx/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.42'
make[4]: *** [Makefile:29: /home/xxx/lede/build_dir/target-x86_64_musl/linux-x86_64/linux-6.1.42/.modules] Error 2
make[4]: Leaving directory '/home/xxx/lede/target/linux/x86'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/home/xxx/lede/target/linux'
time: target/linux/compile#1509.54#198.78#1710.54
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/compile] Error 1
make[2]: Leaving directory '/home/xxx/lede'
make[1]: *** [target/Makefile:23: /home/xxx/lede/staging_dir/target-x86_64_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/xxx/lede'
make: *** [/home/xxx/lede/include/toplevel.mk:231: world] Error 2

Spring3month, 2023-08-01:
> j1 日志

请再看看

WYC-2020, 2023-08-02:
看不到具体错误信息 ，是不是机器性能不够 磁盘大小多少

closed 2026-09-05 by coolsnowwolf